evcc-io/evcc #7499: 403 forbidden bei Fiat 500e nach Upgrade auf v0.116.2

fbhas, 2023-04-16:
### Describe the bug

Nach Upgrade auf v0.116.2 kommt folgende Meldung:

Apr 16 15:14:11 evcc evcc[458]: [lp-1  ] ERROR 2023/04/16 15:14:11 vehicle odometer: unexpected status: 403 (Forbidden)
Apr 16 15:14:11 evcc evcc[458]: [lp-1  ] INFO 2023/04/16 15:14:11 car connected
Apr 16 15:14:11 evcc evcc[458]: [lp-1  ] ERROR 2023/04/16 15:14:11 vehicle soc: unexpected status: 403 (Forbidden)

Mit v.0.114.1 lief alles ohne Probleme. Der Login in die myFiat App funktioniert ebenfalls weiterhin problemlos.

### Steps to reproduce

1. upgrade auf v0.116.2
2.
3.
...

### Configuration details

```yaml
nichts geändert seit v0.114.1
```

### Log details

```text
siehe oben
```

### What type of operating system are you running?

Linux

### Version

v0.116.2

premultiply, 2023-04-16:
Ohne betriebsbereite Glaskugel leider nicht zu lösen.

closed 2023-04-16 by premultiply

fbhas, 2023-04-16:
Tolle Antwort...

Was soll ich da noch zu schreiben, außer, dass dieselbe Einrichtung mit v.0.114.2 einwandfrei funktioniert hat und nach dem Upgrade dieser Fehler auftrat. Mehr steht dazu auch nicht in den Logs...

andig, 2023-04-23:
> Tolle Antwort...

@fbhas bitte beherrsch dich.

> NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.

> Mehr steht dazu auch nicht in den Logs...

Dann musst du mehr freischalten. Wie steht in der Doku.

fbhas, 2023-04-23:
Hat sich erledigt. Gibt es für so "hilfsbereite" Leute wie @premultiply eigentliche ne Ignoreliste ode sowas?

andig, 2023-04-23:
Nein, aber du bist hiermit für drei Tage aufgrund "abuse" gesperrt. 

Die Anforderungen an Issues sind klar beschrieben. Es obliegt Dir, die notwendigen Informationen bereit zu stellen, nicht uns sie mühsam zu suchen. Wenn das nicht funktioniert kannst Du leider keine Hilfe bekommen.

fbhas, 2023-04-27:
Persönliche Angriffe und Machtmissbrauch. Respekt...