evcc-io/evcc #7502: PLZ z.B. von Grünstromindex in Diagramm "Smartes Netzladen"

markus80d, 2023-04-16:
Hi,
mir ist aufgefallen, dass bei mir die Angaben der g CO2 pro kWh meiner Postleitzahl bei Grünstromindex so gar nicht zu den Diagrammwerten passen, die in der UI bei "Smartes Netzladen" angezeigt werden.
Wäre es möglich die Angaben hier auf die in der yaml angegebenen PLZ zu beziehen? Dann würde das zusammenpassen.
VG Markus

closed 2023-04-16 by andig

markus80d, 2023-04-16:
Hi andig, ist das schon umgesetzt oder ist der Vorschlag aus deiner Sicht nicht sinnvoll? Ich hatte gestern auch in einer Diskussion gefragt ob der Parameter smartCostLimit beim "Smarten Netzladen" auch eine Rolle spielt. Der schaut ja dann z.B. bei mir auf den Grünstromindex meiner PLZ und das passt eben nicht zu den Angaben/Diagramm in der UI beim "Smarten Netzladen".
Danke dir für deine Antwort!
VG Markus

markus80d, 2023-04-16:
Die Frage war gestern in der Diskussion auch, ob es Sinn machen würde den Parameter "smartCostLimit" in der UI anzuzeigen wenn er in der yaml entsprechend konfiguriert ist.

andig, 2023-04-16:
Dein Problem ist aufgrund fehlender Detail nicht bearbeitbar.

markus80d, 2023-04-16:
Ein Problem ist es keins, aber ein Verbesserungsvorschlag. Welche Details fehlen dir?

andig, 2023-04-16:
Alle die im Issuetemplate angefragt sind um überhaupt mal das Problem nachvollziehen zu können- die PLZ wird nämlich verwendet...

markus80d, 2023-04-16:
OK, dann ist es wohl doch ein Problem bei mir. Leider weiß ich immer noch nicht wie ich im HomeAssistant auf das vollständige LOG von EVCC komme. Ich vermute mal ohne das wird eine Fehlersuche schwierig sein, oder?

Welches Issuetemplate ist das wo ich das mal nachlesen kann?