evcc-io/evcc #7509: SQLite DB evcc.db wird befüllt, aber keine Anzeige im UI unter Ladevorgänge

KlausiW, 2023-04-17:
### Describe the bug

Ladevorgänge werden in die SQLITE DB evcc.db gespeichert, allerdings wird diese nicht unter Ladevorgänge im UI angezeigt. Des Weiteren sind die Werte meter_start_kwh, meter_end_kwh und meter_charged_kwh 0.0. Lediglich odometer ist befüllt.
Installation auf Raspberry PI 4 unter Libreelec

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
meters:
- name: grid1
  type: template
  template: sungrow-hybrid 
  id: 1  
  host: espressif  
  port: ???  
  usage: grid  
  modbus: tcpip  
- name: pv2
  type: template
  template: sungrow-hybrid 
  id: 1  
  host: espressif  
  port: ???  
  usage: pv  
  modbus: tcpip  
- name: battery3
  type: template
  template: sungrow-hybrid 
  id: 1  
  host: espressif  
  port: ???  
  usage: battery  
  modbus: tcpip  
# charger definitions
chargers:
- name: wallbox1
  type: template
  template: eebus 
  ski: ??????????????????????
  ip: 192.????????  

# vehicle definitions
vehicles:
- name: ID3
  type: template
  template: id 
  title: Auto ID3  
  user: ?????????  
  password: ???????  
  vin: WVW???????  
  capacity: 58
  phases: 3
  icon: car
  cache: 15m
  mode: MinPV
  minSoc: 20
  targetSoc: 80
  minCurrent: 6
  maxCurrent: 16

loadpoints:
- title: Garage
  charger: wallbox1
  vehicle: ID3
  mode: minpv # charge mode (off, now, minpv, pv)
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

# site describes the EVU connection, PV and home battery
site:
  title: Zuhause
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3
  residualPower: 600 #-2000
  prioritySoc: 30 # give home battery priority up to this soc (empty to disable)
  bufferSoc: 50 # ignore home battery discharge above soc (empty to disable)
  maxGridSupplyWhileBatteryCharging: # ignore battery charging if AC consumption is above this value
  smartCostLimit: # set cost limit for automatic charging in PV mode
```

### Log details

```text
● evcc.service - evcc
     Loaded: loaded (/storage/.config/system.d/evcc.service; enabled; vendor preset: disabled)
     Active: active (running) since Fri 2023-04-14 10:34:30 CEST; 7s ago
   Main PID: 12322 (evcc)
     CGroup: /system.slice/evcc.service
             └─12322 /storage/evcc/evcc -c /storage/evcc/evcc.yaml

Apr 14 10:34:30 LibreELEC evcc[12322]: [db    ] INFO 2023/04/14 10:34:30 using sqlite database: /storage/.evcc/evcc.db
Apr 14 10:34:30 LibreELEC evcc[12322]: [eebus ] INFO 2023/04/14 10:34:30 Local SKI:  ????
Apr 14 10:34:37 LibreELEC evcc[12322]: [site  ] INFO 2023/04/14 10:34:37 site config:
Apr 14 10:34:37 LibreELEC evcc[12322]: [site  ] INFO 2023/04/14 10:34:37   meters:      grid ✓ pv ✓ battery ✓
Apr 14 10:34:37 LibreELEC evcc[12322]: [site  ] INFO 2023/04/14 10:34:37     grid:      power ✓ energy ✓ currents
Apr 14 10:34:37 LibreELEC evcc[12322]: [site  ] INFO 2023/04/14 10:34:37     pv 1:      power ✓ energy ✓ currents
Apr 14 10:34:37 LibreELEC evcc[12322]: [site  ] INFO 2023/04/14 10:34:37     battery 1: power ✓ energy ✓ currents
Apr 14 10:34:37 LibreELEC evcc[12322]: [site  ] INFO 2023/04/14 10:34:37   vehicles:
Apr 14 10:34:37 LibreELEC evcc[12322]: [site  ] INFO 2023/04/14 10:34:37     vehicle 1: range ✓ finish ✓ status ✓
Apr 14 10:34:37 LibreELEC evcc[12322]: [lp-1  ] INFO 2023/04/14 10:34:37 loadpoint 1:
Apr 14 10:34:37 LibreELEC evcc[12322]: [lp-1  ] INFO 2023/04/14 10:34:37   mode:        minpv
Apr 14 10:34:37 LibreELEC evcc[12322]: [lp-1  ] INFO 2023/04/14 10:34:37   charger:     power ✓ energy ✗ currents
Apr 14 10:34:37 LibreELEC evcc[12322]: [lp-1  ] INFO 2023/04/14 10:34:37   meters:      charge ✓
Apr 14 10:34:37 LibreELEC evcc[12322]: [lp-1  ] INFO 2023/04/14 10:34:37     charge:    power ✓ energy ✗ currents
Apr 14 10:34:37 LibreELEC evcc[12322]: [lp-1  ] INFO 2023/04/14 10:34:37 vehicle updated: unknown -> Auto ID3
```

### What type of operating system are you running?

Linux

### Version

0.116.0

andig, 2023-04-17:
Ladevorgänge <0.1kWh werden nicht angezeigt.

closed 2023-04-17 by andig

KlausiW, 2023-04-17:
Dankeschön, macht auch Sinn.