evcc-io/evcc #7512: Planner: Verbessern des teuersten Slot Planung

Hofyyy, 2023-04-17:
**Is your feature request related to a problem? Please describe.**
Nach mehreren verschiedenen Logfiles und Usermeldungen würde ich gerne folgendes zur Diskussion stellen.

1. Wir haben immer noch Verbesserungspotential bei der Planung des teuersten) Slots in zwei Anwendungsfällen.
- Wenn er in der Mitte zwischen zwei Slots liegt, siehe Beispiel unten.
- Wenn er ganz am Anfang steht und z.B. bei 14:30 Uhr bis 15:00 Uhr geplant ist.

**Describe the solution you'd like**
Aktuell vereinbarte Kriterien
- Wir stoppen keinen aktiven Slots. (Buggy)
- Wir pausieren keine Ladung, wenn ein anderer Slot innerhalb der nächsten 60min startet.

1. Realer Ladeplan, der das Potential gut aufzeigt (aufbauend auf Tibber Preisen):
planned 2h32m35s until 2023-04-17 07:00:00 +0200 CEST at 4600W: total plan duration: 2h32m35s, avg cost: 0.289
slot from: 2023-04-17 02:00:00 +0200 CEST to 2023-04-17 03:00:00 +0200 CEST cost 0.288
slot from: 2023-04-17 03:00:00 +0200 CEST to 2023-04-17 03:32:35 +0200 CEST cost 0.290
-> Hier entsteht eine Lücke von 28min, die durch andere Sicherheitschecks gefangen wird.
-> Siehe Verbesserungsvorschlag 1.
slot from: 2023-04-17 04:00:00 +0200 CEST to 2023-04-17 05:00:00 +0200 CEST cost 0.289

2. Erster Zeitlicher Slot fängt z.b. bei 13:30 Uhr an und geht bis 14:00
Hier gibt es das Problem, dass der Estimator ja immer "minimal" defensiver Planen muss und soll als die Realität.
Wir laden also immer ein wenig schneller als geplant. Das führt in dem Fall aber dazu, dass der teuerste Slots immer kleiner wird,
und weil es der erste ist, der Startzeitpunkt schneller nach hinten wandert als wir laden. Wir rutschen also nach vorne aus unserem eigenen Slot raus. Wir sind also garnicht mehr in dem Slot, sondern laden vor ihm her.

**Describe alternatives you've considered**
Ich würde gerne folgendes ändern, wenn der teuerste Slot geplant wird.

1.
Wenn der teuerste Slot; einen Nachfolger und einen Vorgänger hat (wie im Ladeplan oben)
-> Blase ihn auf 60min auf und ziehe die Dauer vom Zeitlich letzten ab.
-> Da wir ihn Stundenplanen und sowieso nicht anhalten, würde es nichts teurer machen.

2.
Wenn der teuerste Slot; einen Nachfolger, aber keinen Vorgänger hat
Oder wir nur einen Slot haben der < 60 min ist.
-> Blase ihn auf 60min auf und ziehe die Dauer vom Zeitlich letzten ab.
-> Wir sollten denke ich generell den zeitlich ersten Slot immer ganz vorne Starten,
damit werden wir einiges an Problemen los.

Hofyyy, 2023-04-17:
@schenlap 
Magst du mal draufgucken was du von der Analyse hälst?

Hofyyy, 2023-04-18:
Anbei mal meine Überlegung im Sourcecode
```
// If we are the only one, cut to the right length at the end
// If we are the last slot in the plan, cut to the right lenght at the end
// If we are in the middle and we have no direct Successor, cut ...
if !SlotHasDirectSuccessor(slot, plan) {
	slot.End = slot.End.Add(requiredDuration)
} else {
	// If the costly slot has a direct successor do not cut him to avoid short gaps in the plan
	// TODO, cut latest slot by time?
}
requiredDuration = 0
```

Hofyyy, 2023-04-18:
Ich würde die Funktion SlotHasSuccessor() dann gerne in SlotHasDirectSuccessor() umbenennen.

Hofyyy, 2023-04-21:
@andig Magst du einmal rüber lesen?
Ich bin heute beim Testen schon wieder drüber gestolpert, das mir der Ladeplan abhaut, weil der erste Slot der teuerste war und er von vorne gekürzt wurde.

andig, 2023-04-21:
Ich habs auch die Schnelle nicht verstanden

Hofyyy, 2023-04-21:
Ok dann nur ein Punkt.
- Wenn wir Slots vorne kürzen. (Was wir aktuell tun) . Werden Sie in manchen Usecases vorne immer kürzer und laufen uns weg und wir sind vorne außerhalb des Slots. Das bringt probleme. Ich denke wir haben kein Problem jeden Slot von vorne zu Starten, da wir ja nur noch in Stunden planen.

andig, 2023-04-21:
Wir kürzen dann von vorne, wenn der zeitlich erste Slot der teuerste ist. Ist ein trade-off. Den ersten Slot voll mitzunehmen würden bedeuten den teuersten voll mitzunehmen. Ich verstehe dass es zeitlich defensiver wäre das nicht zu tun. Dann müssten wir aber- damit der Plan die richtige Länge bekommt- in dem Fall den zeitlich letzten Slot kürzen.

Was meinst Du @schenlap

Hofyyy, 2023-04-21:
Genau das schlage ich vor. Und zwar nicht um noch mehr Risiko zu sparen und defensiver zu werden, sondern weil es zu Problemen und Pausen führt. Und ja ich würde den Plan nicht verlängern, sondern die Differenz dann hinten abziehen.

Der Nachteil ist das wir "N" Minuten früher und ein wenig teurer starten.

andig, 2023-04-21:
> sondern weil es zu Problemen und Pausen führt.

Warum ist das so? Am Rest des Plans ändert sich ja nichts und der hat auch keine Garantie, am Stück zu sein?

Hofyyy, 2023-04-21:
Ok. Tue mal so er ist am Stück und wird nach hinten immer billiger. Also vorne teuer.
Dann laden wir langsamer als geplant (da wir ja defensiv planen). Heisst wir sind "IMMER" langsamer als der Plan. Was ja auch gut so ist.

Dann wird immer vorne etwas weggenommen und wir laufen aus dem aktiven slot raus.

andig, 2023-04-21:
So einen Plan kann es nicht geben ;)

Hofyyy, 2023-04-21:
Tibber ... 20 Cent 19 Cent 18 Cent 17 Cent ...?

andig, 2023-04-21:
Wenn die Slots ausserhalb des Planes billiger wären dann wären sie ja im Plan?

andig, 2023-04-21:
Ein teuerster erster Slot kann m.E. nicht weglaufen. Ein nicht erster wird such nicht von vorne gekürzt.

Hofyyy, 2023-04-21:
Also:

25 Cent (nicht geplant)
20 Cent (zu 50% geplant ab halber stunde)
19 Cent (100% geplant)
18 Cent (100% geplant)
25 Cent (nicht geplant

Hofyyy, 2023-04-21:
Beim regelmäßigem neuplanen schneidet er immer weiter von vorne weg.

Hofyyy, 2023-04-21:
Also angenommen wir haben eine Ladeeffizient von 50% (ich weiss es sind 90%) aber nurz zur verdeutlichung.
Wir laden 10 min.
Der Plan wird 20 min kürzer.

Frage: wo wird die 10 min extra abgeschnitten? Genau im ersten teuren slot vorne.

andig, 2023-04-21:
Korrekt. Aber dennoch würden wir das Laden für Pausen <1h ja nicht stoppen, also geht es dennoch weiter, auch wenn der Plan grad was Anderes sagt?

schenlap, 2023-04-22:
Außerdem wird ein begonnener Slot nicht mehr gestoppt. Die teuereste Stunde wurde weitergleaden bis sie vorbei ist. Da wir jetzt keine Pausen <1h mehr machen sollte sich nicht viel ändern. Ich bin mir aber nicht sicher ob wir nicht irgendwo einen Fehler drinnen haben dass die Verrigelung einen gestartet Slot nicht mehr zu stoppen versagt, ich bilde mir ein dies in einem Log gesehen zu haben.

Hofyyy, 2023-04-22:
Wie prüfen wir den ob wir in einem angefangenem slot sind? kann das sein das die logic nicht damit zurecht kommt, wenn der slot wegläuft?

andig, 2023-04-22:
In einem angefangenen Slot bist du wenn du lädst. Nur dann kann es ja neue Erkenntnisse über die Ladedauer geben. Und wenn wir begonnen haben bügeln wir ja Lücken aus. 

> aber die anderen algorithmen scheinen nicht zu greifen :-(

Dann sollten wir uns so einen konkreten Fall nochmal anschauen.

schenlap, 2023-04-22:
lp.planSlotEnd wird da gesetzt: https://github.com/evcc-io/evcc/blob/master/core/loadpoint_plan.go

sollte auch nicht rückgesetzt werden wenn der Slot komplett aus dem plan fällt. Wie gesagt, es war nur eine Vermutung.

Hofyyy, 2023-04-22:
Danke das gucke ich mir noch mal an.

Was mir noch in den Kopf kommt. Wir reduzieren ja wärend der Ladung Risiko, da wir ja immer schlauer werden, wie lange die Ladung noch dauert. Weil der Plan kürzer ist (weniger unsicherheiten) und weil wir wissen wie schnell wir laden.

Wenn wir jetzt den Plan vorne kürzen (in dem Usecase), behebt zwar die 1 Stunde ganz doofes verhalten, allerdings bildet der Plan nicht mehr die realität ab, sondern sogar das gegenteil. Also ist auch das Planende u.s.w. nicht richtig.

ich schlage die änderungen vor, das die 1 Stunde uns da nicht helfen muss und der Plan im idealfall auch das abbildet was in der Realität passiert. Wenn wir als beispiel vorne 30 min verschoben sind, heisst das auch das wir in der Realität 30min früher fertig sind.

schenlap, 2023-04-22:
jetzt wird es philosophisch :-)
Wir kürzen nicht zum Spass sondern weil wir keine ganz e Stunde brauchen. Ob hinen oder vorne ist für die Gesamtzeit mal egal.

Ich sage zum Ende hin steigt das Risiko (verhältnismäßig) weil zum Ende hin die Ladelesitung begrenzt wird.

Ich denke die Diskussion bringt uns so erstmal nicht weiter, über was diskutieren wir überhaupt?
Sauber aufbereitete Ladelogs mit Letztstand und dann können wir diskutieren ob das erwartete Verhaltem dem tatsächlichen entspricht, bringt da mehr.

Hofyyy, 2023-04-22:
Danke trotzdem für die Antworten :-) Ich tue mich schwer punkte die kein Crash sind zu positionieren, weil Sie für ärger sorgen und oft machen wir Sie dann 3 Monate später.

Wir hatten Logs von users, ist wo genau das passiert, Sie verwirrt sind und wir dann auf die 1 Stunde hochgegangen sind. Das sehe ich als Pflaster, aber ja das hilft.

Ich will aber auch nicht zuviel von eurer Zeit klauen, da ich merke ihr findet das verhalten so ok.

Ich hab die Diskussion gestartet weil:
- Wir haben einen sauberen Estimator, der das Ende mit berücksichtigt.
- Ich die eine Stunde für den Holzhammer halte.
- Wir User haben die sich das mit Trace=1 anschauen und nicht verstehen was das soll
- Der Plan im Trace nicht das anzeigen was in der Realität passiert.

Guck die mal die Logs hier an: trace2 von princemax... fast ganz am Ende
https://github.com/evcc-io/evcc/issues/6044#issuecomment-1510805209

closed 2023-04-22 by Hofyyy