evcc-io/evcc #7519: evcc.yaml Config Anpassungen über GUI / Homepage

horst789github, 2023-04-17:
Wäre es möglich das man die Config über die evcc Homepage ändert kann? also man hat ja schon Rechtsoben die 3 Striche und dann "Einstellungen" nur leider kann ich die Config aktuell nicht anpassen.

VolkerK62, 2023-04-17:
guckst du hier ;-)   https://github.com/evcc-io/evcc/issues/6029

horst789github, 2023-07-29:
ich habe evcc bei einem nachbarn installiert der keinerlei ahnung von it hat. er hat leider eine wallbox (heidelberg energy control) die keine automatische phasenumschaltung erlaubt, sondern nur durch die yaml config gesetzt werden kann. wie kann er als laie leicht zwischen 1P und 3P wechseln ? so wie ich das sehe MUSS er die config dazu jedes mal anpassen oder?

VolkerK62, 2023-07-29:
Das ist eigentlich ein Diskussions-Thema, kein Issue.
Der `phases` Eintrag in der yaml hat absolut nichts mit dem tatsächlichen physikalisch-technischem Anschluss der Wallbox zu tun.
D. h.  eine Änderung in der yaml (geht auch per API) z.B. von 3p auf 1p, ist nur erforderlich, wenn die Wallbox auch tatsächlich nur 1p angeschlossen ist.

horst789github, 2023-07-29:
> geht auch per API

ist das per API das gleiche wie das was man in der GUI einstellen kann (AUTO/1P/3P) ? oder ist das quasi wie der yaml eintrag bloß quasi in der aktuellen session (nach neustart weg, da config wert geladen wird) ?

VolkerK62, 2023-07-29:
ist definitiv nach Neustart weg. Keine ahnung ob die Einstellung im GUI einen Neustart überlebt.
Für API gibt es prima Smartphone Apps (z.B. HTTP Shortcuts), eigentlich auch für Nicht-Nerds geeignet ;)
Wenn die Wallbox einen Zähler hat, der auch die Phasenspannungen übermittelt, wird das automatisch eingestellt. Da braucht man gar nix ändern.

closed 2023-07-30 by andig