evcc-io/evcc #7520: [lp-3  ] WARN invalid phase wiring between charge meter and charger

horst789github, 2023-04-17:
### Describe the bug

error: [lp-3  ] WARN 2023/04/17 18:03:22 invalid phase wiring between charge meter and charger

ich habe keine probleme und alles funktioniert wie gewünscht... nur die meldung ist jetzt 2x aufgetauscht in einer nutzungszeit seit ~1-2 tagen

### Steps to reproduce

- weiß ich nicht

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: XXX

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
[... 204 lines cut ...]
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
[lp-3  ] WARN 2023/04/17 18:03:22 invalid phase wiring between charge meter and charger
```

### What type of operating system are you running?

Docker container

### Version

0.116.0

premultiply, 2023-04-17:
Kann durch zu niedrige (gemessene Phasenspannung) auftreten.
Hier müssen dauerhaft überall >= 208 anliegen bzw. gemessen werden.

Ohne aussagekräftiges Log allerdings nicht zu beurteilen.

Sollte dies mal der Fall sein ist entweder die Messeinrichtung in der Fallbox sehr ungenau oder nicht richtig kalibriert
oder es liegt womöglich sogar ein elektrischer Installationsfehler vor.

Eine ungenaue Messeinrichtung lässt sich durch eine zusätzliche Messung der Netzspannung z.B. mittels Multimeter während des Ladevorgangs überprüfen.

foto-andreas, 2023-04-17:
In der Easee-App kannst du einen Reset der Box auslösen. Das hat bei mir geholfen. Passierte, nachdem mein Hausanschluss erneuert wurde.

foto-andreas, 2023-04-18:
Ah, war durch die Reihenfolge der Chargers irritiert.

closed 2023-04-21 by andig

horst789github, 2023-04-21:
@andig  das Problem besteht weiterhin, warum wurde geclosed?

andig, 2023-04-21:
Seit https://github.com/evcc-io/evcc/issues/7520#issuecomment-1511915511 fehlen weiter führende Informationen. Ich bin davon ausgegangen dass es sich erledigt hat.

horst789github, 2023-04-21:
ich bin laie, die fehlermeldung sagt mir nicht was das problem ist. wie ich geschrieben habe, funktioniert alles, ich kann normal die autos laden. Ich vermute hier also ein bug in der software. Was genau braucht ihr von mir damit man diesen bug abstellen kann? ein Log auszug habe ich ja im ersten posting gemacht.

premultiply, 2023-04-22:
Was zeigt denn der goecharger für Phasenspannungen an?
Wenn davon eine niedriger als 208V ist, dann ist dies die Ursache für die Meldung.
Siehe oben.

horst789github, 2023-04-22:
> Wenn davon eine niedriger als 208V ist, dann ist dies die Ursache für die Meldung. 
habe kurz mal geschaut, 211V war das niedrigste was ich kurz gesehen habe... ich kann natürlich nicht ausschließen das es mal unter 208V war. Was könnte den die Ursache dafür sein und ist das bedenklich?

[image]

premultiply, 2023-04-22:
Das Gerät taugt wohl nix.
Beim Hersteller fragen ob sie es nachkalibrieren können, sonst zurückgeben.

horst789github, 2023-04-22:
@premultiply was wären den normale werte? wie soll ich das ggü dem Hersteller argumentieren, wenn doch alles geht? Also ein verweis auf zu niedrige Volt, brauch ich ja irgendwie ein verweis das zu wenig volt schlecht sind.

premultiply, 2023-04-22:
Vergleiche die tatsächlichen Messwerte der Phasen (z.B. Multimeter) mit der Ausgabe eines goechargers und rechne dir den Messfehler aus.

horst789github, 2023-04-24:
also der Hersteller hat geantwortet: ```
Grundsätzlich sind die Spannungsmessungen bei der Hardwareversion V2 etwas ungenauer als bei den neueren Versionen. Es handelt sich hier aber um keinen Defekt.

In diesem Fall denke ich, dass die Spannung des Stromnetzes während des Ladevorganges einbricht und das Fahrzeug ab einer gewissen Mindestspannung den Ladevorgang abbricht. Der Charger hat darauf leider keinen Einfluss.
```

premultiply, 2023-04-24:
Wenn ich mich nicht irre tauchten die Fehler ausserhalb von Ladevorgängen auf...
Von Messung ist das weit entfernt.
Ein Messgerät mit über 10% Messabweichung verdient diese Bezeichnung nicht.

horst789github, 2023-04-25:
> Wenn ich mich nicht irre tauchten die Fehler ausserhalb von Ladevorgängen auf...

ja und nein... gerade ist es auch wieder passiert, obwohl nicht geladen wurde. Aber wo ich das problem das erste mal gesehen hatte, war ich auch ma laden. Sprich mal so, mal so.

premultiply, 2023-04-25:
Und das spricht alles einfach nur dafür dass deine Wallbox grundsätzlich falsch misst und die Aussage des Herstellers auch basierend auf weiteren Schilderungen hier recht fragwürdig erscheint.

N-DEE1, 2025-02-08:
Hallo,

Ich habe genau das selbe phänomen.
Kann die "Fehlermeldung" denn nicht erst bei unter 200v angezeigt werden (zumindest wenn ein go-e in verwendung ist)?
Auch ist die Meldung irreführend - immerhin handelt es sich ja nicht um eine ungültige Phasenverdrahtung.

lg
Andy