evcc-io/evcc #7523: Kostenverechnung: Netzbezug, Einspeisevergütung

MaxV85, 2023-04-17:
Hallo,

ich werde das Gefühl nicht los, dass die Berechnung der Kostenwerte nicht stimmt. Ich dachte ich hab es verstanden, entweder ich stehe am Schlauch oder es stimmt was nicht. 

Mein Strompreis ist bei 42,9 Cent, Einspeisung wird mit 7,25 vergütet. Beide Werte habe ich korrekt in der evcc.yaml hinterlegt. 

Wie am Screenshot zu sehen, werden die Kosten pro kWh dynamisch angepasst. Wie aber kommen die zustande? Die Werte ändern sich extrem durcheinander. Im Screenshot kann ich mir das nicht erklären. Kann jemand helfen?

[image]

MaxV85, 2023-04-17:
Ergänzung: Hier ein weiterer Screenshot, ähnlicher Verbrauch, andere Verbrauchskosten die nicht stimmig sind mE. 

[image]

VolkerK62, 2023-04-17:
Bei dem ersten hast du Erzeugung. der Sonnenstrom macht deinen Gesamtverbrauch billiger.
Beim zweiten kein Sonnenstrom, dann kostet die der Verbrauch das Netzentgeld.

MaxV85, 2023-04-17:
Ok... ich glaub jetzt hab ich's. 

Es werden scheint nicht die aktuellen Kosten absolut berechnet sondern Cent/kWh... die Angabe führt in die Irre, da dort Cent als Einheit angegeben ist

MaxV85, 2023-04-17:
Ok... ich glaub ich komme dahinter. Es werden nicht absolute Cent Beträge angezeigt sondern Cent pro kWh... die Einheit ist irreführend

MaxV85, 2023-04-17:
> (30W * 7,25ct + 216W * 42,9ct) / 246 = 38,6ct

Vielen Dank! Ist ok jetzt... finde die Darstellung zwar mich irreführend, aber zumindest macht es Sinn. 

Danke

closed 2023-04-17 by MaxV85