evcc-io/evcc #7525: Planner Grünstromindex seems to be defect

thorstenwagner, 2023-04-17:
### Describe the bug

I updated from 1.15.x to 1.16.1 and noticed that that the co2 planning with grünstromindex does not work anymore. The planner does not seem to use the co2 data anymore:
[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: *******

meters:
- type: template
  template: powerfox-poweropti 
  usage: grid  
  user: *******  
  password: ******* 
  name: grid1
- type: template
  template: tasmota
  usage: pv
  host: 192.168.0.86
  name: mini-pv

chargers:
- type: template
  template: easee
  timeout: 20s 
  user: *******  
  password: *******
  charger: EHVCHBJJ  
  name: wallbox2

loadpoints:
- title: Garage
  charger: wallbox2
  mode: now
  vehicle: skoda
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

vehicles:
- name: skoda
  type: template
  timeout: 20s
  template: enyaq
  capacity: 60
  user: *******
  password: *******

tariffs:
  planner:
    type: grünstromindex
    zip: *****

site:
  title: MyHome
  meters:
    grid: grid1
    pvs:
      - mini-pv
```

### Log details

```text
evcc --log error gives me:

[gsi   ] ERROR 2023/04/17 19:38:59 json: cannot unmarshal object into Go struct field gsiForecast.Message of type string
[main  ] ERROR 2023/04/17 19:38:59 failed configuring planner tariff: cannot create tariff 'grünstromindex': json: cannot unmarshal object into Go struct field gsiForecast.Message of type string

```
```

### What type of operating system are you running?

Linux

### Version

0.116.1

VolkerK62, 2023-04-17:
Die HP funktioniert auch aktuell nicht.

andig, 2023-04-17:
Already fixed on master, only GSI _is_ broken atm.

closed 2023-04-17 by andig