evcc-io/evcc #7533: Fehlermeldungen auf der Homepage genauer zuordnen

horst789github, 2023-04-18:
Ich erhalte z.b. die Fehlermeldung "vehicle odometer: unexpected status: 403 (Forbidden)"
was mir jetzt fehlt, wäre um welchen LP oder Fahrzeug es den geht, da ich mehre habe. 

Könnte man nicht hier die Infos mitangeben?
z.b.:

"on {nameVomLoadpointLautConfig} vehicle odometer: unexpected status: 403 (Forbidden)"
oder
"vehicle {nameVomFahrzeugLautConfig} odometer: unexpected status: 403 (Forbidden)"

andig, 2023-04-21:
Siehe https://docs.evcc.io/docs/reference/configuration/messaging. Der Name des Ladepunkts ist `.title`.

closed 2023-04-21 by andig

horst789github, 2023-04-21:
> Siehe https://docs.evcc.io/docs/reference/configuration/messaging. Der Name des Ladepunkts ist `.title`.

ich meinte nicht die messaging funktion sondern die native Error Log anzeige in der GUI. Dort steht bei den Fehlermeldungen kein zuordnung zu welchem Gerät die Fehlermeldung gehört. Das kann man (soweit ich sehe) nicht mit der messaging anpassen.

z.b.  https://github.com/evcc-io/evcc/blob/261fab299677b1cc95af5d37bf936a29b7bfdd4f/util/request/functions.go

```
// NewStatusError create new StatusError for given response
func NewStatusError(resp *http.Response) StatusError {
	return StatusError{resp: resp}
}

func (e StatusError) Error() string {
	return fmt.Sprintf("unexpected status: %d (%s)", e.resp.StatusCode, http.StatusText(e.resp.StatusCode))
}

```

da fehlt dann die zuordnung welches gerät den diesen "unexpected status" hat. Ich habe z.b. 3 Wallboxen und 2 autos, aber die Fehlermeldung sagt mir in der GUI nicht welches gerät jetzt diesen fehler hat.

andig, 2023-04-21:
Das ist ne gute Idee- in der aktuellen Struktur aber leider nicht so einfach möglich :/