evcc-io/evcc #7537: My Renault funktioniert nicht mehr

kolossboss, 2023-04-18:
Hallo,
Seit heute funktioniert die My Renault integration nicht mehr. 

Anscheint hat sich der API Key geändert.

Im der Home Assistant Repo wird schon diskutiert, und dort steht auch schon der neue API Key

https://github.com/home-assistant/core/issues/91593

API Key: YjkKtHmGfaceeuExUDKGxrLZGGvtVS0J

Ich hoffe das bringt was.

Viele Grüße

Simon

CarstenPV, 2023-04-18:
Guten Tag, 
ich denke deswegen ist unsere Zoe beim Laden offline.
Bei meiner App auf dem Handy funktioniert die Ausgabe.
Nur wie bekomme ich den API key jetzt in EVCC auf Ubuntu?

Gruß Carsten

kolossboss, 2023-04-18:
Ich denke das muss in evcc direkt geändert werden.

Hofyyy, 2023-04-18:
https://github.com/evcc-io/evcc/pull/7541

Müsste aber noch wer testen, da ich kein Renault besitze.

sibbi77, 2023-04-18:
#7541 funktioniert mit meinem Zoe

kolossboss, 2023-04-18:
Kann ich leider nicht so einfach testen, da ich evcc als Home Assistant Addon nutze.

RaptorDE, 2023-04-18:
Da der API Key ja häufiger geändert wird, wäre es gut wenn man den API Key über die Config überschreiten/ändern könnte.

andig, 2023-04-18:
Das ist ne gute Idee! PR welcome.

closed 2023-04-18 by andig

Oasisfan1, 2025-01-10:
Hallo! Habe heute meinen Raspi und anschließend evcc eingerichtet. Habe alles hinbekommen, nur beim Versuch, den RENAULT ZOE einzubinden kommt auch bei mir 403 Forbidden. Meine Kontodaten stimmen, kann mich in der MYR-App und im Web damit einloggen. Was kann ich tun, um den API (sollte es daran liegen) zu ändern?! Da ich gerade die Ersteinrichtung nach offizieller Anleitung gemacht habe, sollte ja alles auf dem aktuellen Stand sein, oder? Danke + LG!

CarstenPV, 2025-01-12:
Zum Vergleich

vehicles:
- type: template
  template: renault
  title: Zoe 40
  icon: car
  user: xxxxxxx@gmx.xx
  password: jaja
  vin: VF1blabla
  capacity: 41
  phases: 3
  cache: 15m
  mode: minpv
  minCurrent: 7
  maxCurrent: 16
  alternativewakeup: true
  name: ev4

  Auf Leerstellen achten!!
  - 1 Leerstelle
  danach jede Zeile 2 Leerstellen dann, aber die Automatik macht das eigentlich. 
Mal versucht, die Abfrage über sudo zu machen?
Gruß

CarstenPV, 2025-01-12:
Und sind die Dienste, zum Abfragen, abgelaufen?