evcc-io/evcc #7539: Fahrzeugerkkung

harrykausl, 2023-04-18:
Ich habe 2 Fahrzeuge in evcc. Das eine ist ein Fiat 500e, das andere eine Mercedes B-Klasse hybrid. Der Fiat wird beim Anstecken erkannt, der Mercedes nicht, da kommt Gastfahrzeug und das Rädchen dreht sich. Ich habe deshalb (da ich den öfters laden muss) den Mercedes in loadpoints als Standard definiert, wenn ich den Fiat lade, dann schalte ich das Fahrzeug in evcc um. Wäre es möglich, anstatt dem harten Standardfahrzeug ein optionales zu definieren, welches dann genommen wird, wenn die anderen nicht erkannt wurden (vehicle_default). Beim Erkennen sollte dieses Fahzeug dann übersprungen werden (da es ja nicht erkannt wird)

andig, 2023-04-18:
Ich verstehe den Fall, das ist aber doch schon sehr spezifisch. Denke nicht, dass wir das in evcc lösen wollen- es sei denn wir würden das default vehicle generell in dieser Art interpretieren würden.

andig, 2023-04-18:
Dieses „eventual“ default müsste zumindest auch Verbindungsstatus A abdecken- anderenfalls funktioniert der poll mode „always“ nicht mehr.

harrykausl, 2023-04-19:
Ich denke, man bräuchte 2 verschiedene Typen von parameter. Einmal vehicle in loadpoints, damit bei nur einem Fahrzeug keine Erkennung gestartet wird (wie es heute funktioniert). Der andere, neue Fall wäre der, dass der default parameter genau an einem vehicle stehen würde. In diesem Fall, sollte kein vehicle in loadpoints stehen.
Der Fall ist natürlich speziell, könnte aber allen helfen, die mehr als 1 Auto ranhängen, wovon eins nicht erkannt wird.

RustyDust, 2023-04-26:
@harrykausl Kann es sein, dass du dasselbe Problem hast wie ich es gerade unter #7679 beschrieben habe? Evtl. hat ja Stellantis gleich mal alle ihre APIs verbockt ;)
@andig  Sorry für das neue Ticket, hatte nur nach PSA und Opel gesucht und nicht an die anderen Stellantis-Marken gedacht.

harrykausl, 2023-04-27:
In diesem Fall ist meiner Meinung nach der Mercedes das Problem.

igotsoul, 2023-05-03:
Würde ebenfalls für diese Lösung abstimmen. Habe einen ähnlichen Fall mit einem EV das nicht erkannt werden kann wegen fehlender Online Anbindung und ein weiteres, das unterstützt wird. Genau der oben beschriebene Prozess wäre wünschenswert. Nach dem Motto: "Wenn online Fahrzeug nicht erkannt, dann muss es wohl das andere sein"

premultiply, 2023-05-04:
"Das Andere" kann aber auch ein beliebiges Gastfahrzeug sein.

igotsoul, 2023-05-04:
True, ist halt abhängig vom Use Case. Ich persönlich habe bisher noch nie Gastfahrzeuge gehabt bzw. der von mir beschrieben Fall tritt deutlich häufiger ein. Und wenn ich mir die anderen Kommentare und Diskussionen ansehe, scheine ich damit nicht alleine zu sein. Vielleicht könnte man das einfach intelligent konfigurierbar machen. Wenn Gastfahrzeug erlaubt, dann Logik wie bisher. Falls nicht, dann ist Default bei Nichterkennung immer das Offline Auto.

andig, 2023-05-04:
Das ist doch ganz einfach: wenn ein Default gesetzt ist wirds letztendlich der Default. Das ist heute auch nicht anders, nur dass es sofort passiert. Ein Gastfahrzeug ist immer möglich.

igotsoul, 2023-05-04:
@andig  Vielleicht bekomme ich es auch nicht richtig konfiguriert, denn auch wenn ich 2 EVs eintrage, wird das zweite leider nur als "Gast" erkannt. Wenn das schon mit der bestehenden Funktionalität, wie in meiner Diskussion bzw. hier im Thread realisierbar ist, dann lass mich gerne wissen wie. Bisher war das für mich leider ein no-go für das kompletten Umstieg auf evcc ;-(

andig, 2023-05-04:
> dann lass mich gerne wissen wie

Das ist die Sache mit der Glaskugel. Ich kenne weder Deine Fahrzeuge, noch Deine Konfiguration, noch Dein Logfile das vllt. Ursachen zeigen könnte ;)

andig, 2023-05-04:
> denn auch wenn ich 2 EVs eintrage, wird das zweite leider nur als "Gast" erkannt. 

Wie soll da denn bitte was erkannt werden?

    template: offline
    title: Soul EV

igotsoul, 2023-05-04:
Sorry, you lost me. Es geht doch genau darum: "äre es möglich, anstatt dem harten Standardfahrzeug ein optionales zu definieren, welches dann genommen wird, wenn die anderen nicht erkannt wurden (vehicle_default). Beim Erkennen sollte dieses Fahzeug dann übersprungen werden (da es ja nicht erkannt wird)"

Der Anwendungsfall, dass ein Auto nicht erkannt wird, das es keine Online Anbindung hat. Ja, eines dieser älteren EV-Modelle, die das damals noch nicht hatten. Und diese Fahrzeuge können auch nicht über eine Typ-2 ID oder ähnliches erkannt werden. Daher war doch hier die Frage, wenn ein Auto immer sicher erkannt wird - in meinem Fall der Ioniq5 -, dass wenn es dieser nicht ist, wohl das "offline" Fahrzeug am Stecker hängen muss. Stichwort: Ausschlussverfahren und es sich dabei eben nicht um ein Gastfahrzeug handelt.

andig, 2023-05-04:
Genau. Aber da es das noch nicht gibt kannst du es natürlich auch jetzt nicht konfigurieren.

igotsoul, 2023-05-04:
Richtig und im Status dieses Tickets steht: "need discussion" Genau daran wollte ich mit beteiligen und dem eine weitere Stimme hinzufügen. Wenn das angekommen ist, umso besser. :)

andig, 2023-05-04:
Ist sie 👍🏻

harrykausl, 2023-05-14:
Es wäre natürlich nur eine Krücke für nicht erkannte Autos, bei mir der Mercedes. Würde aber zumindest bei den Symptomen helfen.

andig, 2023-05-14:
Es bleibt weiterhin unklar, wie das mit Defaultfahrzeuge und Abfragemodus "always" funktionieren soll.

harrykausl, 2023-05-14:
Man könnte den jetzigen Zustand so belassen. Wenn in Loadpoints definiert ist das Fahrzeug Standard. Wenn unter einem Fahrzeug bei vehicles definiert wäre es der Fallback.

harrykausl, 2023-05-22:
Besteht eine reelle Chance, dass da etwas kommt?

harrykausl, 2023-07-05:
Nochmals die Nachfrage, gibt es Überlegungen das zu implementieren?

Macrisu, 2023-10-29:
I also have the same hassle with my Mercedes each day!

mathoc77, 2024-05-21:
Schade, dass dieses Thema keine Priorität bekommt. Ich habe den gleichen Fall. Zweitfahrzeug ist e-Up für den ich keine mobilen Dienste zahlen möchte. Wäre schön wenn immer der e-Up ausgewählt wird, wenn mein Online Fahrzeug nicht erkannt wird. 
Die ganze Gastfahrzeug - Funktionalität würde ich nicht so hoch priorisieren, den Fall hatte ich noch nie.
Das sollte so selten sein, hier wäre manuelles Auswählen wirklich kein Problem. Ich würde mir wünschen, immer auf das Fallback-Fahrzeug zu gehen, statt auf ein Gastfahrzeug.

igotsoul, 2024-05-21:
Würde hier auch nochmal ein +1 als Vote hinterlassen. Dieser Use Case hält mich aktuell noch zurück die openWB Steuerung komplett evcc zu überlassen.

andig, 2024-05-21:
@igotsoul wie macht die openWB das denn?

igotsoul, 2024-05-21:
@andig  Gar nicht, die openWB ist diesbezüglich dumm und unterscheidet die Fahrzeug nicht. Zumindest nicht in der Version 1.x, die ich weiterhin verwende. Ich glaube ab Version 2.x gibt es Fahrzeug Profile - wie das dort geregelt ist, habe ich mir jedoch noch nicht angesehen.
Ich stimme @mathoc77 zu, dass der Use Case "Gastfahrzeug" eher selten vor kommt. Es gibt aber sicherlich einige Leute da draußen, speziell die EV Early Adaptors, die ein Fahrzeug besitzen, welches noch keine App Anbindung hat oder dessen ID anderweitig ausgelesen werden kann.

andig, 2024-05-21:
Dann verstehe ich nicht, wie Dich das noch vom Umstieg abhalten kann ;). Aber das ist hier dann OT...

neupertd, 2024-08-21:
Hallo,

ich will mich hier auch nochmal äußern.
Ich habe eine OpenWb mit RFID-Leser. evcc charger liefert auch die korrekten ID's der Chips.
Die Autoerkennung scheint hier aber regelmäßig zu gewinnen.
Als Fahrzeuge habe ich einen ID4 und einen E-Up. Meistens gewinnt der ID4.

`vehicles:
- type: template
  template: vw 
  title: ID4  
  icon: car  
  user: XXX
  password: XXX  
  vin: XXX  
  capacity: 77  
  phases: 3  
  cache: 15m  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 16  
  priority: 1  
  timeout: 10s  
  identifiers:
    - 0530883923
    - 1784945924
    - 3658618116
  name: ev4
- type: template
  template: vw 
  title: Up   
  icon: car  
  user: XXX  
  password: XXX  
  vin: XXX  
  capacity: 32  
  phases: 2  
  cache: 15m  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 16  
  priority: 1  
  identifiers:
    - 0530151075
    - 3121907299
  timeout: 10s  
  name: ev5
`

andig, 2024-08-21:
Ist OT. Hier gehts darum was passiert wenn kein Fahrzeug erkannt wird.

igotsoul, 2024-10-28:
@andig Bin jetzt umgestiegen und betreibe die openWB nur noch im Ladepunkt Modus. Das Thema hier würde ich trotzdem nochmal "up voten". Denn bisher wird mein Kia Soul - da offline ohne API vehicle - immer als Gastfahrzeug "identifizier". Ich wäre immer noch für eine Art Weiche in der Konfiguration, mit der man einstellen kann, dass das Gastfahrzeug automatisch das Offline Fahrzeug ist und somit auch die Zuordnung der geladen kwh dem passenden Fahrzeug zugeordnet wird

privat99, 2024-11-03:
Ich habe auch das Problem, dass ich zwei Fahrzeuge im evcc angelegt habe und die Fahrzeuge immer als Gastfahrzeug erkannt werden. Es handelt sich um einen BMW i3 und einem Audi e-tron.
Gibt es eine Lösung, dass die Fahrzeuge erkannt werden?
Weiter suche ich eine Einstellung, dass unbekannte Fahrzeuge auch nicht als Gastfahrzeug erkannt werden. Ich möchte ein Laden von mir unbekannten Fahrzeugen vermeiden.

Elchkopp, 2024-11-18:
Bei der openWB v2 kann jedem Fahrzeug über das Fahrzeugprofil eine ID zugewiesen werden. Wird diese ID erkannt, wird das Fahrzeug mitsamt dem konfiguriertem Fahrzeugprofil ausgewählt. Dies bleibt so lange ausgewählt, bis manuell oder durch eine neue ID ein neues Fahrzeugprofil ausgewählt wird. Im jeweiligen Fahrzeugprofil selbst kann ausgewählt werden, dass nach dem Abstecken auf das Standard-Fahrzeug(profil) umgeschaltet wird. 
Wird also kein Fahrzeug erkannt, bleibt entweder das zuletzt ausgewählte Profil aktiv oder es ist das Standard-Fahrzeug(profil) aktiv.

privat99, 2024-11-20:
Ich habe auf 0.131.6 upgedated.
Seither wird kein Fahrzeug mehr erkannt.
Ein Neustart von evcc und auch mehrere PI Neustarts haben keinen Erfolg gebracht.
[image]

Ein Downgrade auf 0.131.5 brachte auch keine Verbesserung. Nun bin ich wieder auf 0.131.6.
Hat jemand eine Idee?

Im Log ist folgender Hinweis enthalten
{"error": "invalid_client", "error_description": "Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method)"}

Mit der APP auf meinem Smartphone kann ich mit den Logindaten auf das Fahrzeug zugreifen.

[main ] INFO 2024/11/20 15:19:50 evcc 0.131.6
[main ] INFO 2024/11/20 15:19:50 using config file: /config/evcc.yaml
[db ] INFO 2024/11/20 15:19:50 using sqlite database: /data/evcc.db
[mqtt ] INFO 2024/11/20 15:19:50 connecting evcc-486284336 at tcp://192.168.0.12:1883
[mqtt ] DEBUG 2024/11/20 15:19:50 tcp://192.168.0.12:1883 connected
[main ] INFO 2024/11/20 15:19:50 listening at :7070
[main ] ERROR 2024/11/20 15:19:51 creating vehicle BMW_I3S_Martina failed: cannot create vehicle type 'template': cannot create vehicle type 'bmw': Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method): unexpected status: 401 (Unauthorized)
[lp-1 ] DEBUG 2024/11/20 15:19:52 charge total import: 640.380kWh
[lp-2 ] DEBUG 2024/11/20 15:19:52 charge total import: 17.756kWh
[site ] WARN 2024/11/20 15:19:54 vehicle 'BMW i3s Martina' does not support automatic detection

andig, 2024-11-20:
Können wir bitte beim Thema bleiben. Danke.

schpuntik, 2024-12-30:
So wie ich es sehe  es  gibt  noch  keine Fahrzeugerkennung   für   Autos  die  aus dem  Hersteller  cloud  raus  sind.  Ich besitze  zwei  Renault Zoe  die  10 Jahre alt  sind.  Mit  dem WiCAN  OBD  ich kann Akku SoC  und GoM  auslesen und  mit MQTT  am EVCC  benutzen.  Nur  Problem ist das ich 2x  habe und muss jedes mal  manuell Fahrzeug auswählen. 

Ist das möglich   unter  identifiers   als   custom    mit  MQTT   status   auslesen   und zuweisen  zum Fahrzeug Profil .  Bei  WiCAN  jeder  Adapter hat eigene  ID .  So  sieht es aus   wican/7c2c67b39bbd/can/status .

andig, 2024-12-30:
Klar geht das, hat hier aber nichts zu suchen. Bitte mach eine neue Diskussion dazu auf.

drahdiwaberl, 2025-03-03:
selbes Thema hier:
2 Fahrzeuge: eines davon offline/nicht identifizierbar
Gastfahrzeuge: nie

wäre schön wenn für einen charger statt "Gastfahrzeug" das nicht identifizierbare ausgewählt würde - wenn eben keines identifiziert wurde.

andig, 2025-03-04:
Mein Vorschlag wäre: wir ändern die Bedeutung des Defaultfahrzeugs in "eventual" default. Es wird zunächst die Erkennung durchgeführt. Wird kein Fahrzeug identifiziert, so wird am Ende das Defaultfahrzeug ausgewählt.
Nachteil: ein false Positive der Erkennung liefert dann ein anderes Fahrzeug als das heutige Default.

Wär das für alle ok?

drahdiwaberl, 2025-03-04:
ich kenne als Nutzer (und das noch nicht sehr lange) euer System noch viel zu wenig um da auch nur eine ansatzweise qualifizierte Antwort zu geben, aber klingt als würde es das Thema lösen.

Trotzdem noch ein Gedanke, ohne zu wissen wieviel Aufwand eine Umsetzung auf diesem Weg wäre:
Szenario 1: ich habe x Autos konfiguriert, eines davon ist "template: offline" -> wenn letzteres angesteckt wird (egal an welchem LP), wird statt "Gastfahrzeug" dieses ausgewählt.
Szenario 2: ich habe mehr als ein "template: offline" in der config -> Dann wird wie jetzt auch das "Gastfahrzeug"  ausgewählt, weil ja sowieso ein Umstellen im GUI (oder anderem Weg) notwendig wäre.

wie dabei jetzt ein eventueller default am LP reinspielt weiss ich nicht, weil noch nie so verwendet.

Aber so würde denk ich am aktuellen ("mapping"-)System wenig geändert werden müssen und es sollte auch niemand der aktuell mit externen Automatisierungen arbeitet negative Überraschungen erleben.

naltatis, 2025-03-05:
> Wir ändern die Bedeutung des Defaultfahrzeugs in "eventual" default.

Das ist eine sehr mutige Veränderung des aktuellen Verhalten. Wenn die Fahrzeugerkennung 100% zuverlässig wäre würde ich hier mitgehen. Das ist sie aber nicht. Zudem haben wir ja auch inzwischen einige andere Fälle, wo wir das "Default Vehicle" eigentlich für etwas anderes benutzen (Wärmeerzeuger und so).

Die aktuelle Funktionsweise, dass das setzen eines Standardfahrzeuges die Automatische Erkennung deaktiviert ist für ganz viele Anwendunsfälle (bspw. 2 LPs / 2 Autos) genau richtig. Hier jetzt "Unschärfe" einzuführen würde ich nicht machen. Zumindest nicht ohne Option.

Mein Vorschlag wäre dann eher die "Automatische Erkennung" über eine dedizierte Option (an/aus) am Ladepunkt konfigurierter zu machen. Wenn die angeschaltet ist und ein Default Fahrzeug gesetzt ist würde genau das Verhalten aus dem PR https://github.com/evcc-io/evcc/pull/19412 greifen.

andig, 2025-03-05:
Nachdem der Lösungsvorschlag jetzt seit 2 Jahren genau so hier steht dachte ich eigentlich, dass wir über diesen Punkt hinweg sind.

> Zudem haben wir ja auch inzwischen einige andere Fälle, wo wir das "Default Vehicle" eigentlich für etwas anderes benutzen (Wärmeerzeuger und so).

Die sind obsolet. Können wir als BC markieren.

andig, 2025-03-05:
@drahdiwaberl Szenario 1 ist genau die Lösung hier. Szenario 2 hat mit Default Vehicle nix zu tun.

andig, 2025-03-27:
Wir sind zu dem Schluss gekommen, dass das Verhalten ohne ein weiteres Setting nicht umsetzbar ist. Das wollen wir aktuell nicht angehen.

closed 2025-03-27 by andig