evcc-io/evcc #7540: Add Deye 3p inverters

premultiply, 2023-04-18:
> Im Zweifelsfall einfach mal beim Hersteller anfragen und dir das Protokoll schicken lassen.

Hallo premultiply,

Ich habe ein Modbus Protokoll gefunden:
https://drive.google.com/file/d/1dcDckwOo0D6AcKim6tq2j9WnpXdAHM_m/view

Ich kann da jetzt nichts mit anfangen, aber vielleicht könntet ihr da etwas draus lesen?

Habe mir auch einen DEYE SUN-12K-SG04LP3-EU bestellt.

Wäre richtig Klasse wenn ich ihn mit einem Modbus Konverter auch in EVCC einbinden könnte.

Oder hat jemand den Wechselrichter schon mit EVCC am laufen?

Gruß Monne

_Originally posted by @m0nn3 in https://github.com/evcc-io/evcc/discussions/4557#discussioncomment-5648455_

edvandreas, 2023-04-19:
Servus,

ich habe den genannten Wechselrichter mit evcc via MQTT verbunden. Benutzt habe ich dazu dieses Repository https://github.com/kbialek/deye-inverter-mqtt

Funktioniert bei mir absolut fehlerfrei.

Natürlich wäre es schön wenn evcc die Deye Wechselrichter nativ unterstützen würde.

Gruß,
Andy

Blowfly69, 2023-04-30:
Hallo,

ich habe den Deye SUN-12K per Modbus RTU über einen billigen RS485-USB Konverter am gleichen Raspi 2B dranhängen, auf dem auch evcc läuft.  Die Konfiguration war zwar mühsam, per custom Modbus Register jeden einzelnen Wert zu definieren, insbesondere wenn man sich erst mal in die evcc Logik einarbeiten muss... Aber es läuft - abgesehen von gelegentlichen timeouts beim Auslesen der Register - bis jetzt sehr ordentlich und ich möchte mich an der Stelle mal bei den Entwicklern bedanken: Ein Spitzen-Tool, sowohl von der Funktionalität als auch der Optik und Bedienung (kleine Abzüge in der B-Note für die Konfiguration)! 

Unten der entsprechende Auszug aus meiner config.yaml. Ist noch nicht vollständig, ggf. sind auch noch Fehler drin - so klar ist das Deye Modbus Dokument nicht zu interpretieren... Etwaige umständliche/falsche Definitionen bitte korrigieren, insbesondere wenn man die Abfragen perfomanter hinbekommen könnte. 
Ich habe übrigens am Deye neben zwei PV-Strings noch einen Mikrowechselrichter am GEN Eingang hängen, dessen Leistung ich zu den Strings dazuzähle. Wer das nicht nutzt, einfach die dritte Source unter pv-power-add löschen.

Viele Grüße
Hans

`meters:
- type: custom
  name: grid1
  power:  
    source: modbus
    id: 1 
    device: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0NKKQN-if00-port0
    baudrate: 9600
    comset: "8N1" 
    register: 
      address: 625
      type: holding
      decode: int16
    scale: 1.0
  currents:
    - source: modbus
      id: 1
      device: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0NKKQN-if00-port0
      baudrate: 9600
      comset: "8N1"
      register:
        address: 613
        type: holding
        decode: int16
      scale: 0.01
    - source: modbus
      id: 1
      device: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0NKKQN-if00-port0
      baudrate: 9600
      comset: "8N1"
      register:
        address: 614
        type: holding
        decode: int16
      scale: 0.01
    - source: modbus
      id: 1
      device: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0NKKQN-if00-port0
      baudrate: 9600
      comset: "8N1"
      register:
        address: 615
        type: holding
        decode: int16
      scale: 0.01
  energy:
    source: modbus
    id: 1
    device: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0NKKQN-if00-port0
    baudrate: 9600
    comset: "8N1"
    register:
      address: 522
      type: holding
      decode: int16
    scale: 0.1
- type: custom
  name: pv2
  energy:
    source: modbus
    id: 1
    device: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0NKKQN-if00-port0
    baudrate: 9600
    comset: "8N1"
    register:
      address: 529
      type: holding
      decode: int16
    scale: 0.1
  power:
    source: calc 
    add: 
    - source: modbus 
      id: 1 
      device: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0NKKQN-if00-port0 
      baudrate: 9600 
      comset: "8N1" 
      register:
        address: 672
        type: holding
        decode: int16
      scale: 1.0
    - source: modbus
      id: 1
      device: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0NKKQN-if00-port0 
      baudrate: 9600
      comset: "8N1"
      register:
        address: 673
        type: holding
        decode: int16
      scale: 1.0
    - source: modbus
      id: 1
      device: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0NKKQN-if00-port0
      baudrate: 9600
      comset: "8N1"
      register:
        address: 664 # Phase 1 GEN Port 
        type: holding
        decode: int16
      scale: 1.0
- type: custom
  name: battery3
  energy:
    source: modbus
    id: 1
    device: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0NKKQN-if00-port0
    baudrate: 9600
    comset: "8N1"
    register:
      address: 515
      type: holding
      decode: int16
    scale: 0.1
  power:
    source: modbus
    id: 1
    device: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0NKKQN-if00-port0
    baudrate: 9600
    comset: "8N1"
    register:
      address: 590
      type: holding
      decode: int16
    scale: 1.0
  soc:
    source: modbus
    id: 1
    device: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0NKKQN-if00-port0
    baudrate: 9600
    comset: "8N1"
    register:
      address: 588
      type: holding
      decode: int16
    scale: 1.0
`

m0nn3, 2023-04-30:
Hallo Hans, Klasse 👍🏻
Vielen Dank für's teilen.
Mein 12k ist gestern geliefert worden.
Werde ihn die nächsten Wochen aufbauen und berichten wie ich Ihn eingebunden habe.

Blowfly69, 2023-05-01:
Hier ist übrigens noch das Modbus Dokument in Englisch: [modbus-en.pdf](https://github.com/evcc-io/evcc/files/11362771/modbus-en.pdf) . Es ist damit zumindest lesbar, wenngleich die Interpretation manchmal trotzdem schwierig ist.

@premultiply  Wie aufwendig ist das in ein template zu gießen? Würde das ja übernehmen, aber bin noch ziemlicher Anfänger bei evcc ...

Blowfly69, 2023-05-12:
Ich hätte jetzt auch noch die templates dazu erstellt, auf Basis der growatt_hybrid Templates. Würde es gerne testen, scheitere aber momentan am Kompilieren von evcc in VSC, da muss ich mich noch weiter einarbeiten, genauso wie in Github. Bin da etwas eingerostet, meine Programmierzeit begann mit Basic auf einen CBM 4008 und hatte ihr Hoch zu Zeiten von Turbo Pascal...  
@premultiply: Würde einen PR machen, oder?

premultiply, 2023-05-12:
Ja, mach gerne mal einen PR.
Wir ruckeln das dann schon gemeinsam hin.

thierolm, 2023-05-13:
@premultiply gibt es eine Möglichkeit über die custom Modbus config eine Seriennummer mitzugeben? Ich würde gerne versuchen auch ein Config für die Deye/Bosswerk Microinverter zusammenstellen. Die brauchen aber eine Seriennummer (siehe https://github.com/kbialek/deye-inverter-mqtt/blob/cf7558c0ee218d2738b64712e8cca49b1e98ac7f/deye_modbus.py#LL91C11-L91C11)

Die Register sind in dem Projekt auch gelistet: https://github.com/kbialek/deye-inverter-mqtt/blob/cf7558c0ee218d2738b64712e8cca49b1e98ac7f/docs/metric_group_micro.md

premultiply, 2023-05-18:
Wie meinst du das?
Prinzipiell kann man ja mit configure bzw. den Templates beliebige Daten im Dialog abfragen.

closed 2023-05-22 by premultiply

trixing, 2023-07-05:
> Hallo,
> 
> ich habe den Deye SUN-12K per Modbus RTU über einen billigen RS485-USB Konverter am gleichen Raspi 2B dranhängen, auf dem auch evcc läuft. 

Hast du die Beschreibung irgendwo, an welche Pins von dem DEYE SUN-12K-SG04LP3-EU du die angeschlossen hast? RS485 pins gibt es ja einmal auf den BMS1/2 pins und dann nochmal auf dem "RS485" Stecker mit Modbus A/B ?

premultiply, 2023-07-05:
"BMS" ist definitiv falsch, also "Modbus"

Blowfly69, 2023-07-06:
Ich habe ein Standard Netzwerkkabel (T568B) genommen, auf der einen Seite am Deye in die „Modebus“ Buchse gesteckt, auf der anderen Seite den RJ45-Stecker abgeschnitten, und am RS485-USB Konverter das braune Kabel (8) auf RS485-, das weiß-braune Kabel (7) auf RS485+ und das grüne Kabel (6) auf GND gelegt.  

[image]

premultiply, 2023-07-06:
GND weglassen.

Blowfly69, 2023-07-06:
Guter Tipp, danke, werde ich gleich testen. Habe tatsächlich von Zeit zu Zeit serial read failures.

trixing, 2023-07-06:
Ich habe beim Modell SUN-12K-SG01HP3-EU-AM2 folgende RJ45 Ports zur Auswahl: 

Meter: for energy meter 
Parallel_1: Parallel communication
Parallel_2: Parallel communication
CAN: reserved.
DRM: Logic interface for AS/NZS 4777.2:2020.
BMS1: BMS port  1.
BMS2: BMS port 2
RS485: RS485 port.

RS485A/B pin gibt es quasi auf jedem Port. BMS2 hatte ich mit orange / weiss probiert - pin 7+8  - ich glaube bei einigen anderen WR fehlt der "RS485" Stecker.

Dann versuche ich nochmal RS485 port "Modbus_RS485_A + B". Ob pin1+2 und 7+8 gleich beschaltet sind - oder da ein Unterschied besteht?

Ueber den Port 8899 des LSW3 Wifi Sticks funktioniert es zumindest schonmal.

Quelle: https://www.deyeinverter.com/deyeinverter/2022/09/22/%E3%80%90b%E3%80%9130240301001258-%E4%BD%BF%E7%94%A8%E8%AF%B4%E6%98%8E%E4%B9%A6-%E5%82%A8%E8%83%BD%E4%B8%89%E7%9B%B8-sun-(5-20)k-sg01hp3-eu-am2-%E5%BE%B7%E4%B8%9A%E8%8B%B1%E6%96%87-%E5%B0%81%E9%9D%A2%E5%BD%A9%E5%8D%B0-%E5%B0%81%E9%9D%A2%E5%B0%81%E5%BA%95157g-%E5%86%85%E9%A1%B570g%E9%93%9C%E7%89%88%E7%BA%B8-142.5x210mm-ver2.2.pdf

Blowfly69, 2023-07-06:
Da lt. Anleitung nur der RS485 Port Modbus spricht, würde ich den nehmen. Pin 1 und 8 bzw. 2 und 7 sind wahrscheinlich parallel geschalten, musst du halt ausprobieren.

PR #7961 (2023-05-22): Add Deye hybrid inverters
Add templates for Deye Sun-(5-12)K-SG04LP3-EU and compatible Sunsynk hybrid inverters. Based on growatt-hybrid templates.

Fixes #7540