evcc-io/evcc #7547: Ocpp Bootnotification: invalid memory address or nil pointer dereference

sreichholf, 2023-04-19:
### Describe the bug

Absturz nach der OCPP Bootnotification einer ABB TerraAC.

### Steps to reproduce

EVCC wurde gestartet und stürzte mit besagter Meldung hart ab

### Configuration details

```yaml
chargers:
  - name: TerraAC
    type: ocpp
    connecttimeout: 30m
    timeout: 20m
    getconfiguration: true
    bootnotification: true
```

### Log details

```text
`[main  ] INFO 2023/04/19 08:12:12 evcc 0.116.2 (9b3e74e0)

[main  ] INFO 2023/04/19 08:12:12 using config file: /etc/evcc.yaml

...

[ocpp  ] WARN 2023/04/19 08:12:13 chargepoint connected, unknown: TerraAC

[ocpp  ] DEBUG 2023/04/19 08:12:13 waiting for chargepoint: 30m0s

[ocpp  ] TRACE 2023/04/19 08:12:14 received JSON message from TerraAC: [2, "4289088", "BootNotification", {"chargePointModel": "CDT_TACW11::NET_WIFI", "chargePointVendor": "ABB", "chargeBoxSerialNumber": "TerraAC", "firmwareVersion": "****::V1.6.5", "meterType": "V1"}]

[ocpp  ] TRACE 2023/04/19 08:12:14 handling incoming CALL [4289088, BootNotification] from TerraAC

panic: runtime error: invalid memory address or nil pointer dereference

[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0x112ce38]

goroutine 59 [running]:

github.com/evcc-io/evcc/charger/ocpp.(*CP).BootNotification(0x0?, 0x4000ebc005?)

	/build/charger/ocpp/cp_core.go:29 +0x18

github.com/evcc-io/evcc/charger/ocpp.(*CS).OnBootNotification(0x0?, {0x4000ebc005?, 0x0?}, 0x0?)

	/build/charger/ocpp/cs_core.go:70 +0x40

github.com/lorenzodonini/ocpp-go/ocpp1%2e6.(*centralSystem).handleIncomingRequest.func1()

	/go/pkg/mod/github.com/lorenzodonini/ocpp-go@v0.16.1-0.20230416102136-782920b66533/ocpp1.6/central_system.go:505 +0x594

created by github.com/lorenzodonini/ocpp-go/ocpp1%2e6.(*centralSystem).handleIncomingRequest

	/go/pkg/mod/github.com/lorenzodonini/ocpp-go@v0.16.1-0.20230416102136-782920b66533/ocpp1.6/central_system.go:502 +0x4f4`
```

### What type of operating system are you running?

Docker container

### Version

0.116.2 (9b3e74e0)

sreichholf, 2023-04-19:
Noch als kurzer Nachtrag: das passiert nicht immer, genau genommen ist es nur dieses eine Mal passiert.

andig, 2023-04-20:
Ich würd gerne sagen, dass das absolut unmöglich ist. An der Stelle wird zwar auf eine Variable zugegriffen, die ist aber *IMMER* initialisiert. Lass uns das mal beobachten. Falls es nochmal passieren sollte würde ich da Logging mit einbauen.

andig, 2023-04-21:
@sreichholf ich mache mal zu- wenns wieder auftritt und irgendwie reproduzierbar ist bitte melden.

closed 2023-04-21 by andig