evcc-io/evcc #7548: Shelly 3EM die gesamte Leistung abfragen

ausura79, 2023-04-19:
Hallo zusammen,
gibt es die Möglichkeit die gesamte Leistung von einem Shelly 3EM abzufragen?

simowNgithub, 2023-04-19:
Was meinst du mit "gesamte Leistung"? 

Leistung vom Shelly 3EM rufst du so ab:
`
meters:
  - name: my_grid
    type: template
    template: shelly-3em
    usage: grid
    host: 10.10.10.5 # IP-Adresse oder Hostname `

VolkerK62, 2023-04-19:
Ich habe meinen 3EM als custom-meter drin, um die "richtigen" Phasenströme zu bekommen.
Da siehst du auch wie Gesamt-Power abgerufen wird.

```
meters:
- name: grid
  type: custom
  power:
  - source: http
    uri: http://192.168.178.28/status
    jq: .emeters | map(.power) | add
  currents:
  - source: http
    uri: http://192.168.178.28/emeter/0
    jq: if .power < 0 then - .current else .current end
  - source: http
    uri: http://192.168.178.28/emeter/1
    jq: if .power < 0 then - .current else .current end
  - source: http
    uri: http://192.168.178.28/emeter/2
    jq: if .power < 0 then - .current else .current end

```