evcc-io/evcc #7551: Renault API Abruf funktioniert nicht mehr.

stprehn, 2023-04-19:
Describe the bug

Erkennung des Fahrzeugs und Ermittlung des SOC funktioniert seit gestern nicht mehr.
Fahrzeug wird als Offline angezeigt.

Da hat sich wohl der API-Key geändert.
Siehe auch:  https://github.com/db-EV/ZoePHP/issues/28

Die Renault App sowie ZoePHP (neueste Version) funktionieren.

Steps to reproduce

EVCC wurde neu gestartet

Log details

vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'renault': unexpected status: 403 (Forbidden)

What type of operating system are you running?

Docker container

Version

0.116.2

Hofyyy, 2023-04-19:
Ist im nightly schon drin. bitte close

stprehn, 2023-04-19:
Danke, funktioniert

closed 2023-04-19 by stprehn