evcc-io/evcc #7560: EEBUS charger overwrites vehicle minCurrent

DerAndereAndi, 2023-04-20:
### Describe the bug

The EEBUS charger code overwrites the loadpoints `MinCurrent` with the value that is reported by the vehicle via the charger. If the vehicle has a different value set in the configuration, e.g. a Zoe, this is overwritten.

### Steps to reproduce

1. Use an EEBUS wallbox
2. Configure a vehicle with a minCurrent != what the charger reports for the current connection (6A for IEC61851)

### Configuration details

```yaml
- EEBUS wallbox
- Vehicle defined minCurrent
```

### Log details

```text
See https://github.com/evcc-io/evcc/discussions/7241#discussioncomment-5666863
```

### What type of operating system are you running?

Linux

### Version

0.116.2

DerAndereAndi, 2023-04-20:
The problematic code is this in `charger/eebus.go`:

```
func (c *EEBus) setLoadpointMinMaxLimits() {
	if c.lp == nil {
		return
	}

	minLimits, maxLimits, _, err := c.emobility.EVCurrentLimits()
	if err != nil || len(minLimits) == 0 || len(maxLimits) == 0 {
		return
	}

	newMin := minLimits[0]
	newMax := maxLimits[0]

	if c.lp.GetMinCurrent() != newMin && newMin > 0 {
		c.lp.SetMinCurrent(newMin)
	}
	if c.lp.GetMaxCurrent() != newMax && newMax > 0 {
		c.lp.SetMaxCurrent(newMax)
	}
}
```

This logic needs to consider these cases:
- The EEBUS charger does report the min and max currents for the current connection, which depends on the vehicles connection type (IEC61851/ISO15118).
- The connection type may change back and forth while the EV is connected! And thus the min currents may change
- The loadpoint might have a min current and max current set
- A vehicle might have a min and max current set in the vehicle configuration

The best way I can think of to handle this would be:
- If the loadpoint or the EV have a min or max current defined in the configuration, then `setLoadpointMinMaxLimits` should not set the values reported by the EVSE

Question:
Is there a way to figure out if the current setting is defined in the loadpoint or the currently assigned EV configuration?

DerAndereAndi, 2023-04-20:
The current PR checks if the active vehicle has min or max current defined. If so, the charger provided values are ignored

hidminds, 2023-04-20:
Thanks @DerAndereAndi for creating the issue out of the discussion #7241 and providing a fix so quickly!
I guess I may test that fix with the new nightly build tomorrow?
I would install the evcc update from APT source `deb https://dl.cloudsmith.io/public/evcc/unstable/deb/raspbian buster main` .

DerAndereAndi, 2023-04-20:
@DerAndereAndi The change has to be reviewed by the evcc team and merged before it can be tested.

closed 2023-04-20 by andig

hidminds, 2023-04-21:
Yay! :tada: With the new nightly build, it slowly charges my Zoe's battery now in minpv mode and with an elli pro (eebus) wallbox. Thank you very much indeed! :beers:

```
Apr 21 10:55:24 openhabian evcc[9833]: [site  ] DEBUG 2023/04/21 10:55:24 ----
Apr 21 10:55:24 openhabian evcc[9833]: [lp-1  ] DEBUG 2023/04/21 10:55:24 charge power: 5543W
Apr 21 10:55:24 openhabian evcc[9833]: [script] DEBUG 2023/04/21 10:55:24 /bin/bash -c echo 0: 0
Apr 21 10:55:24 openhabian evcc[9833]: [site  ] DEBUG 2023/04/21 10:55:24 pv power: 0W
Apr 21 10:55:24 openhabian evcc[9833]: [site  ] DEBUG 2023/04/21 10:55:24 site power: 5543W
Apr 21 10:55:24 openhabian evcc[9833]: [lp-1  ] DEBUG 2023/04/21 10:55:24 charge currents: [8.11 8.03 7.96]A
Apr 21 10:55:24 openhabian evcc[9833]: [lp-1  ] DEBUG 2023/04/21 10:55:24 detected active phases: 3p
Apr 21 10:55:24 openhabian evcc[9833]: [lp-1  ] DEBUG 2023/04/21 10:55:24 charger status: C
Apr 21 10:55:24 openhabian evcc[9833]: [lp-1  ] DEBUG 2023/04/21 10:55:24 soc estimated: 65.89% (vehicle: 65.00%)
Apr 21 10:55:24 openhabian evcc[9833]: [lp-1  ] DEBUG 2023/04/21 10:55:24 vehicle soc: 66%
Apr 21 10:55:24 openhabian evcc[9833]: [lp-1  ] DEBUG 2023/04/21 10:55:24 vehicle range: 185km
Apr 21 10:55:24 openhabian evcc[9833]: [lp-1  ] DEBUG 2023/04/21 10:55:24 pv charge current: 0A = 8A + -8.03A (5543W @ 3p)
```