evcc-io/evcc #7568: Cupra Born odometer is not available

lex777777, 2023-04-20:
Hi,

in the charging session the odometer value is 0.
Is it just not implemented for this car or should it work?

with `evcc vehicle`:
[main  ] INFO 2023/04/20 13:33:00 evcc 0.116.2
[main  ] INFO 2023/04/20 13:33:00 using config file: /etc/evcc.yaml
[db    ] INFO 2023/04/20 13:33:00 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2023/04/20 13:33:00 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/04/20 13:33:00 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/04/20 13:33:00 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/04/20 13:33:00 SELECT * FROM `settings` 6 <nil>
[mqtt  ] INFO 2023/04/20 13:33:00 connecting evcc-1278863501 at tcp://192.168.1.100:1883
Soc:            62%
Capacity:       58.0kWh
Charge status:  A
Range:          210km
Finish time:    not available
Climate active: false
Target Soc:     80%
Features:       []

andig, 2023-04-20:
Cupra doesn't have odometer as as `evcc vehicle` shows.

lex777777, 2023-04-20:
Thank you for the reply.
Do you have an idea why e.g. [TRONITY ](https://app.tronity.io/) can access this value?