evcc-io/evcc #7576: PLanner/Estimator: add additional charging time for soc > 90%

VolkerK62, 2023-04-20:
### Describe the bug

Update auf 0.116.3 gemacht
In dem Moment, wo ich im UI auf "PLAN" klicke, kommt im Log dieser Eintrag
`[lp-1  ] DEBUG 2023/04/20 17:58:55 add additional charging time 1h44m0s for soc > 90%`

Ist da noch irgendwo ein Rest aus der alten Berechnungweise übriggeblieben? 

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

0.116.3

VolkerK62, 2023-04-20:
Eingestellt ist 20A@1p. Wenn ich maxcurrent auf 17A@1p ändere, erscheint dieser Hinweis nicht.
Da wirkt noch irgendeine 4kW Grenze.

Hofyyy, 2023-04-20:
Nein ... der commit kam mit Vorsatz erst nach dem Release rein.

closed 2023-04-20 by VolkerK62