evcc-io/evcc #7577: Add Amperified connect.home

andig, 2023-04-20:
https://www.amperfied.de/wp-content/uploads/2022/10/Modbus-TCP-Register-Layout-Amperfied-Wallbox-connect.home_20221019.pdf

andig, 2023-05-16:
Bedarf scheint nicht so hoch- bisher schient niemand die Box zu haben, von mir aus können wir uns das gerne sparen?

dnkh, 2023-08-05:
Hi, ich hab die Amperfied connect.business

andig, 2023-08-13:
@dnkh könntest Du Portweiterleitung für info@evcc.io einrichten?

dnkh, 2023-08-15:
@andig ich kann das immer mal wieder tun, wenn das Auto nicht dran hängt.
Der Hintergrund dazu ist, dass die Wallbox per OCCP an einem externen Backend hängt und Modbus TCP und OCCP geht gleichzeit nicht. So hab ich es zumindest verstanden.

D.h. ich muss die Wallbox umkonfigurieren wenn da jemand per Modbus TCP drauf möchte.
Wenn ich das weiß zu welchem Zeitpunkt das ist, kann ich das machen