evcc-io/evcc #7584: Wattpilot wird nicht mehr verbunden nachdem die Verbindungen verloren geht

kad49007, 2023-04-21:
### Describe the bug

Die Verbindung zum Wattpilot wird nicht mehr verbunden nachdem die Verbindungen verloren gegangen ist. Der Verbindung wird erst nach einem Restart von evcc neu aufgebaut.
Passiert relativ schnell, der letzte Neustart war gegen 7:00 Uhr

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error
  mini: debug
  site: debug
  lp-1: trace
  eMini: debug
  charger: debug

# unique installation id
plant: c3e77ab967105b1ccae8945ecccde2e3987556b17683360aa09957f8cdd84b27

interval: 10s # control cycle interval

meters:
- type: template
  template: fronius-solarapi-v1 
  usage: grid  
  host: 192.168.X.67  
  name: grid1
- type: template
  template: fronius-solarapi-v1
  usage: pv
  host: 192.168.X.15
  name: pv
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: 192.168.X.67  
  name: pv2
- type: template
  template: fronius-solarapi-v1 
  usage: battery  
  host: 192.168.X.67  
  name: battery3
  capacity: 19.3
- type:  custom
  name: Heizstab
  power: 
    source: http
    uri: http://192.168.X.67/solar_api/v1/GetOhmPilotRealtimeData.cgi?Scope=Device&DeviceId=0
    method: GET # default HTTP method
    headers:
      - content-type: application/json
    auth:
    insecure: false # set to true to trust self-signed certificates
    jq: .Body.Data.PowerReal_PAC_Sum # parse response json
    timeout: 10s # timeout in golang duration format, see https://golang.org/pkg/time/#ParseDuration
  
  
chargers:
- type: template
  template: fronius-wattpilot 
  host: 192.168.X.101  
  password: XXXX  
  name: wallbox2
- type: template
  template: fronius-wattpilot 
  host: 192.168.X.61  
  password: XXXX  
  name: wallbox1

vehicles:
- type: template
  template: mini 
  title: eMini  
  user: XXXX 
  password: XXXX  
  vin: XXXX 
  capacity: 29  
  phases: 3  
  icon: car  
  cache: 15m  
  mode: PV  
  minSoc: 30  
  targetSoc: 80  
  minCurrent: 6  
  maxCurrent: 16  
  name: emini

loadpoints:
- title: Wattpilot 2
  charger: wallbox2
  vehicle: emini
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
  soc:
    poll:
      mode: connected
      interval: 15m
    estimate: true
- title: Wattpilot 1
  charger: wallbox1
  mode: off
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Neufahrn
  meters:
    grid: grid1
    pvs:
    - pv
    - pv2
    batteries:
    - battery3
    aux:
    - Heizstab
  bufferSoc: 80
  prioritySoc: 75
  residualPower: -50

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.3652 # [currency]/kWh
   # type: awattar
   # cheap: 0.2 # [currency]/kWh
   # region: de

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0606 # [currency]/kWh
  planner:
    type: grünstromindex
    zip: XXX 

sponsortoken: XXX
```

### Log details

```text
> Apr 21 09:45:40 openhabian evcc[29254]: [lp-1  ] DEBUG 2023/04/21 09:45:40 charge currents: [6.9 0 0]A
Apr 21 09:45:40 openhabian evcc[29254]: [lp-1  ] DEBUG 2023/04/21 09:45:40 detected active phases: 1p
Apr 21 09:45:40 openhabian evcc[29254]: [lp-1  ] DEBUG 2023/04/21 09:45:40 charger status: C
Apr 21 09:45:40 openhabian evcc[29254]: [lp-1  ] DEBUG 2023/04/21 09:45:40 soc estimated: 36.10% (vehicle: 36.00%)
Apr 21 09:45:40 openhabian evcc[29254]: [lp-1  ] DEBUG 2023/04/21 09:45:40 vehicle soc: 36%
Apr 21 09:45:40 openhabian evcc[29254]: [lp-1  ] DEBUG 2023/04/21 09:45:40 vehicle range: 56km
Apr 21 09:45:40 openhabian evcc[29254]: [lp-1  ] DEBUG 2023/04/21 09:45:40 pv charge current: 7.17A = 7A + 0.169A (-39W @ 1p)
Apr 21 09:45:46 openhabian evcc[29254]: bye bye
Apr 21 09:45:46 openhabian evcc[29254]: Reconnect
Apr 21 09:45:46 openhabian evcc[29254]: bye bye
Apr 21 09:45:46 openhabian evcc[29254]: Reconnect
Apr 21 09:45:50 openhabian evcc[29254]: [site  ] DEBUG 2023/04/21 09:45:50 ----
Apr 21 09:45:51 openhabian evcc[29254]: [lp-1  ] ERROR 2023/04/21 09:45:51 charge meter: connection is not valid
Apr 21 09:45:51 openhabian evcc[29254]: [lp-2  ] ERROR 2023/04/21 09:45:51 charge meter: connection is not valid
Apr 21 09:45:51 openhabian evcc[29254]: [site  ] DEBUG 2023/04/21 09:45:51 pv power: 4084W
Apr 21 09:45:51 openhabian evcc[29254]: [site  ] DEBUG 2023/04/21 09:45:51 battery soc: 20%
Apr 21 09:45:51 openhabian evcc[29254]: [site  ] DEBUG 2023/04/21 09:45:51 battery power: -2105W
Apr 21 09:45:51 openhabian evcc[29254]: [site  ] DEBUG 2023/04/21 09:45:51 grid power: -2W
Apr 21 09:45:51 openhabian evcc[29254]: [site  ] DEBUG 2023/04/21 09:45:51 giving priority to battery charging at soc: 20%
Apr 21 09:45:51 openhabian evcc[29254]: [site  ] DEBUG 2023/04/21 09:45:51 aux power 1: 0W
Apr 21 09:45:51 openhabian evcc[29254]: [site  ] DEBUG 2023/04/21 09:45:51 aux power: 0W
Apr 21 09:45:51 openhabian evcc[29254]: [site  ] DEBUG 2023/04/21 09:45:51 site power: -52W
Apr 21 09:45:51 openhabian evcc[29254]: [lp-2  ] ERROR 2023/04/21 09:45:51 charge meter: connection is not valid
Apr 21 09:45:51 openhabian evcc[29254]: [lp-2  ] ERROR 2023/04/21 09:45:51 charge rater: connection is not valid
Apr 21 09:45:51 openhabian evcc[29254]: [lp-2  ] ERROR 2023/04/21 09:45:51 charger: connection is not valid
Apr 21 09:45:51 openhabian evcc[29254]: [site  ] DEBUG 2023/04/21 09:45:51 telemetry: charge: Δ14/14Wh @ 1620W
Apr 21 09:46:00 openhabian evcc[29254]: [site  ] DEBUG 2023/04/21 09:46:00 ----
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.116.3

andig, 2023-04-21:
@mabunixda

andig, 2023-04-21:
Du hast doch schon Logging auf die Konsole? Das ist nicht schön aber kommt auch an. Alternative ist ein Logger Interface oder ein `Errors() (chan <-error)` Api über das man sich auf einen Fehlerkanal subscriben kann.

kad49007, 2023-04-21:
Heute Nachmittag war der Fehler verschwunden, aber leider haben meine Einstellungen zur Vergrösserung der Logfiles nicht gefruchtet . Daher habe ich kein Log über den interessanten Zeitraum, es fehlen 2 Stunden. Ich kann also nicht sagen, was das passiert ist.
Bin leider kein Linux-Spezialist, habe in journald.conf die Einstellung SystemMaxUse von 50MB auf 250MB. Ist das die richtige Stelle?

m-milinovsky, 2023-04-21:
Genau dieses Verhalten beobachte ich auch sehr häufig: geht die Verbindung zum Wattpilot mal verloren hilft nur mehr ein Neustart von evcc. Alle anderen Methoden (Fahrzeug abstecken, Wattpilot neu starten bzw. vom Strom nehmen etc.) fruchten nicht.

andig, 2023-04-22:
Ich verschieb das Thema mal in die Discussions- als evcc Team können wir nichts beitragen.