evcc-io/evcc #7585: PV charging starts, ignoring prioritySoc

GrimmiMeloni, 2023-04-21:
### Describe the bug

Attached the car today. PrioritySoc of 70 was not reached.
PV Mode started charging nevertheless.

Observations:
- (09:07:24) attach car 
- (09:07:29) charger out of sync appeared (as usual)
- (09:07:29) evcc sends stop command to Easee
- (09:07:32) charging starts
- (09:07:33) evcc detects unexpected charge
- (09:07:33) evcc sends pauseCharge command to Easee
- (09:07:33) evcc then sends resume charge to Ease
- (09:07:38) Guard timer (5 mins) kicks in
- (09:07:44) charging... 
- (09:12:34) guard expires, evcc sends pauseCharge command to Easee

At first I expected the Easee API to have missed the stop command.
However, seeing that the last command evcc sends before the guard timer is actually resume charging, I think Easee is not at fault here. Indeed it seems, Easee does what it is being told. It could also be a race condition, between resume and stop charge commands.
Generally, I am wondering, why evcc decided to send a resume charging command in this situation. To me this looks like a bug.

### Steps to reproduce

Not sure.
I am pretty positive I have seen this work as expected multiple times. 
Nothing different - we plugged in the car, evcc takes care of the rest.
As stated above, I suspect a race condition between the stop/resume command sequence.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  easee: trace
  enyaq: debug

# unique installation id
plant: xxx
sponsortoken: xxx
interval: 10s # control cycle interval

meters:
- type: template
  template: tesla-powerwall 
  usage: grid  
  host: 192.168.3.38  
  password: xxx  
  name: grid1
- type: template
  template: tesla-powerwall 
  usage: pv  
  host: 192.168.3.38  
  password: xxx  
  name: pv2
- type: template
  template: tesla-powerwall 
  usage: battery  
  host: 192.168.3.38  
  password: xxx  
  name: battery3

site:
  title: Zu Hause
[... 541 lines cut ...]
[lp-1  ] DEBUG 2023/04/21 09:12:47 vehicle range: 248km
[lp-1  ] DEBUG 2023/04/21 09:12:47 pv charge current: 0A = 0A + -0.526A (121W @ 1p)
[lp-1  ] DEBUG 2023/04/21 09:12:47 pv enable timer reset
[lp-1  ] DEBUG 2023/04/21 09:12:47 pv timer inactive
[site  ] DEBUG 2023/04/21 09:12:54 ----
[lp-1  ] DEBUG 2023/04/21 09:12:54 charge power: 0W
[site  ] DEBUG 2023/04/21 09:12:54 pv power: 2606W
[site  ] DEBUG 2023/04/21 09:12:54 battery soc: 67%
[site  ] DEBUG 2023/04/21 09:12:54 battery power: -2300W
[site  ] DEBUG 2023/04/21 09:12:54 grid power: -18W
[site  ] DEBUG 2023/04/21 09:12:54 giving priority to battery charging at soc: 67%
[site  ] DEBUG 2023/04/21 09:12:54 site power: 82W
[lp-1  ] DEBUG 2023/04/21 09:12:54 charge currents: [0.009 0.007 0.01]A
[lp-1  ] DEBUG 2023/04/21 09:12:54 charge meter total import: 851.931kWh
[lp-1  ] DEBUG 2023/04/21 09:12:54 charger status: B
[lp-1  ] DEBUG 2023/04/21 09:12:54 pv charge current: 0A = 0A + -0.357A (82W @ 1p)
```

### What type of operating system are you running?

Linux

### Version

0.116.2

Crajs20, 2023-04-21:
Hallo,
Gleiches Version und verhalten.
Pv Modus und Akku wird langsam entladen. Prio soc ist bei 95%, aktuell bei der Akku auf 88% entladen.

andig, 2023-04-21:
`prioritySoc` bedeutet ausschließlich, dass die Batterieladeleistung *nicht* für den Loadpoint zur Verfügung steht. Wenns trotzdem genug PV gibt wird auch geladen.

Das eigentliche Problem hier ist doch, dass die Easee von sich aus vor sich hin lädt und sich anscheinend auch nicht stoppen lässt. Erstmal muss das Problem beseitigt werden.

@naltatis ist das bei Dir auch?

GrimmiMeloni, 2023-04-21:
ResidualPower habe ich auf 100 für das BMS der Batterie gemäß Anleitung.

andig, 2023-04-21:
Auch interessant ist, dass die Easee `pause_charging` anscheinend als Aufforderung nimmt mit dem Laden zu beginnen:

    [lp-1  ] WARN 2023/04/21 09:07:29 charger out of sync: expected disabled, got enabled
    [easee ] TRACE 2023/04/21 09:07:29 POST https://api.easee.cloud/api/chargers/EHWHL6VE/commands/pause_charging
    [easee ] TRACE 2023/04/21 09:07:29 {"device":"EHWHL6VE","commandId":48,"ticks":638176576496582470}
    [easee ] TRACE 2023/04/21 09:07:32 POST https://api.easee.cloud/api/chargers/EHWHL6VE/settings
    [easee ] TRACE 2023/04/21 09:07:32 {"smartCharging":true}
    --
    [lp-1  ] DEBUG 2023/04/21 09:07:32 charger status: C
    [lp-1  ] INFO 2023/04/21 09:07:32 start charging ->

@naltatis oder liegt das am `smartCharging` und das tut plötzlich mehr als nur LEDs zu schalten???

andig, 2023-04-21:
> ResidualPower habe ich auf 100 für das BMS der Batterie gemäß Anleitung.

Deine Config ist auch oben sichtbar ;)

naltatis, 2023-04-22:
Ich hab in #7597 jetzt erstmal "nur" das Logging für den CommandResponse Kanal hinzugefügt. Hoffe dass wir dadurch Race Conditions und ggf. Timing Issues besser sehen können.

andig, 2023-04-28:
Keine neuen Erkenntniss- können wir hier schließen? Temporäres Problem bei Easee?

GrimmiMeloni, 2023-04-29:
@andig wenn die Frage "Warum wurde hier ein Resume gesendet?" mit den vorliegenden Informationen nicht eindeutig beantwortet werden kann, dann können wir hier auch (erstmal) schließen. Der Effekt hier war auch nicht das Netzstrom genutzt wurde, sondern lediglich das PV in die "falsche" Batterie floss. Betrachte ich als harmlos.

Ich vermute der Auslöser und Fix dürfte in #7366 zu finden sein. Lasst uns hier dann beenden.

closed 2023-04-29 by GrimmiMeloni