evcc-io/evcc #7589: KiaConnect API / App offenbar unzuverlässig

kakulukia, 2023-04-21:
Es ist jetzt bereits des öfteren vorgekommen, dass EVCC einen falschen Ladestand angezeigt hat und ich habe mich jeweils immer gefragt, wie das wohl zustande kommt. Die letzten paar Mal konnte ich aber auch in meiner HandyApp die gleichen Informationen sehen und erst ein händischer Refresh führte dazu, dass sowohl in der App als auch via `evcc vehicle` der richtige SoC zu sehen ist.

Die letzten paar Tests haben ergeben, dass eine erneute Abfrage in der App den richtigen SoC ans Licht bringt.
Ein Aufruf von `evcc vehicle` sollte das gleiche Ergebnis bringen, allerdings gibt es noch keinen, einfach reproduzierbaren Weg, einen falschen SoC in der Kia App hervorzurufen.

Ich werde weiter berichten, sobald das Problem erneut auftritt, würde aber bisher darauf plädieren, dass bevor mit dem Laden begonnen wird, nochmal einen API-Abfrage eingeworfen wird. Die eine tunt nicht weh und sollte dazu führen, dass mein targetSoC nicht weiter überschritten wird.

kakulukia, 2023-04-21:
Offenbar kann man durch `evcc vehicle` doch kein "richtiges" Update anfordern. Mehrere Calls ins folge brachten gerade immer nur das gleiche, falsche Ergebnis.

Aber EVCC macht hier nichts falsch. Nachdem das Auto aufgeladen war und ich grad unterwegs war, sind zwei Sachen passiert.

1. EVCC registriert, dass ein Auto angesteckt wurde, zeigt mir aber "Gastfahrzeug", weil wohl die Antwort nicht schnell genug einging. Da gerade genügend Strom von Dach kommt, wurde das Auto direkt beladen, ohne zu warten, bis mein Kia erkannt und die mit ihm verbunden targetSoC von 90% gesetzt wurde.
Kann ich irgendwo, das Laden bei nicht erkanntem Fahrzeug ausschalten? Das hätte nicht passieren sollen. :(

2. Per API kam ein SoC von 0% und trotz erkanntem Auto wurde daher weitergeladen. Hier ist das Verhalten okay, aber irgendwas ist bei Kia faul. Können andere Kia-Besitzer das Verhalten bestätigen?
Ich schau also wieder in die App und muss leider auch dort sehen, dass mir 0% SoC angezeigt werden. 
Diesmal wollte ich zuerst schauen, ob ein weiterer Aufruf von `evcc vehicle` eine Besserung bringt. Leider nein. Es blieb bei 0%. Jetzt in der App ein Update angefordert und ich sehe wieder den richtigen Stand. Irgendwas macht die App also anders als die API-Anfrage. hmmmm
Danach kann ich per `evcc vehicle` auch den richtigen Stand sehen, ABER

3. Offenbar ist EVCC und diese Kommandozeilenabfrage getrennt, denn ich kann links etwas anders sehen als rechts.
[image]

Da EVCC keine Reload Button hat, hilft dann wohl nur ein kompletter restart der Anwendung. 
Wenn ich aufs Gastfahrzeug wechsle. Sehe ich das hier:

[image]

Es wird als voll angezeigt, aber trotzdem wird ein Ladeversuch counter gestartet.
Wechsle ich wieder zurück, wurde mir der Kia erst als leer angezeigt, obwohl über die Kommandozeile 95% SoC ausgegeben werden. Beim nächsten Wechsel auf Gastauto  und wieder zurück sehe ich dann die richtige Anzeige.

[image]

Aber obwohl hier erkannt wird, dass targetSoC bereits überschritten ist, bricht der timer nicht ab.
Immerhin: Nach Ablauf des Timers wurde kein Ladevorgang gestartet.

Ich konnte zwar grad durch hin und her schalten der virtuellen Autos wieder einen Kia mit 0% sehen, aber das bekomme ich nicht reproduziert bar hin. Irgendwo ist da der Wurm drin.

closed 2023-04-21 by andig

kakulukia, 2023-04-21:
@andig Kannst Du das kurz erklären? Es sind auch Sachen dabei, wo EVCC Quatsch macht.

andig, 2023-04-21:
> NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.