evcc-io/evcc #7596: Restart EVCC

thebrainkafka, 2023-04-21:
Hi.
Wäre es möglich über die UI "restart EVCC" einzubinden.

Oft hängt es oder wenn mal sonst was ist wäre es schön einfach kurz den Neustart zu triggern ohne sich an der Console anmelden zu müssen.

Danke.

StefanSchoof, 2023-04-21:
Ich muss sagen mein evcc hängt eigentlich nie. Ich muss eigentlich nur bei Updates oder config Änderungen neu starten

thebrainkafka, 2023-04-21:
Bei Stromausfall.
Manchmal (eigentlich öfters) zickt VW.
...
Kurzer Restart und es geht wieder.
Leider eben nur über die Console.

m-milinovsky, 2023-04-21:
Fände ich auch eine gute Idee: bei mir hängt evcc öfter mal bei Verbindungsproblemen zum Fronius Wattpilot.

NongJoWo, 2023-04-22:
Auch nutzbar um die Ladeanzeige für Shellys zu Nullen.

StefanSchoof, 2023-04-22:
Eine mögliche Lösung wäre per oliveTin https://docs.olivetin.app/action-service.html sich eine Restart Action anzulegen.

andig, 2023-04-22:
Das hat sein für und wieder (bei OCPP komme ich nicht mehr an den Ladevorgang ran etc) und *verlangt* natürlich dass evcc vom Service neu gestartet wird. Selbst kann es sich nur beenden.

@naltatis wäre das was fürs Menü? Mit kleiner Erläuterung und Warnung negativer Auswirkungen?

naltatis, 2023-04-26:
Ja, so richtig geil find ich das Feature nicht. Cool wäre wenn wir das nicht brauchen würden 😀.
Ich hab aber selbst aktuell auch den Fall das alle paar Tage Tibber Pulse bei mir hängen bleibt und ein GUI Neustart helfen würde #7314.

Ich überleg mir mal was.

PR #7671 (2023-05-01): Add 'Need help?' dialog with restart option
Introduced "Need help?" dialog that groups troubleshooting resources and actions.
- 🧹 removed blog, docs, GitHub from top navigation
- ✍️ renamed "About" to "evcc.io" (the only remaining external link)
- 🍱 added modal with links to docs, discussions and issues.
- 🔥 added button to restart server (with confirm dialog)

fixes #7596

https://user-images.githubusercontent.com/152287/234630480-005e870b-bc09-41bd-b25b-2c3f6a45909a.mp4

[image]
[image]
[image]

**TODOs**

- [ ] enable `/api/shutdown` endpoint for normal operating mode (already exist for fai [... cut]
closed 2023-05-01 by andig