evcc-io/evcc #7602: individuelle Fahrzeug-spezifische Estimator-Parameter

VolkerK62, 2023-04-22:
Durch den Estimator wird die voraussichtliche Ladedauer ermittelt. 
Die Ermittlung der voraussichtlichen Ladedauer ist beim PV-Laden eher von nachrangigem Interesse. Sie ist jedoch wichtig für die Ladeplanung/Zielladen.
In der aktuellen Version wird durch die unten aufgeführten Default-Parameter der abflachenden Ladekurve bei voller werdendem Akku Rechnung getragen und somit die Ladedauer realitätsnäher berechnet.
Da es hier nicht unerhebliche Abweichungen zwischen den Fahrzeugen gibt, sollten die entsprechenden Einstellungen durch die Nutzer fahrzeugspezifisch einstellbar sein:

-	minChargePower (default: 1000W)
-	maxChargeSoc (default: 50%) & maxChargePower (default: 50000W)
-	Effizienz (default: 0,9)

MinChargePower gibt die Ladeleistung an, mit der das Auto bis zum eigenständigen Abschalten bei 100% lädt. Das kann jeder Nutzer für sein(e) Fahrzeug(e) testen und entsprechend einstellen.

MaxChargeSoc gibt an, bis zu welchem Soc das Auto mit maximaler Leistung (maxChargePower) geladen werden kann. Danach reduziert das Auto die Ladeleistung. Auch das kann jeder Nutzer für sein(e) Fahrzeug(e) testen und entsprechend einstellen.

Effizienz spiegelt den Unterschied zwischen den Energiemengen, die die durch die Wallbox erfasst werden und die tatsächlich im Fahrzeug ankommen. Da diese auch von der Ladeleistung abhängt, sollte die Ermittlung dieses Wertes mit der für das Zielladen üblichen Ladeleistung stattfinden.  Die Wallboxdaten werden über den Zähler erfasst. Die Fahrzeugdaten können durch die Soc-Differenz zu Beginn und zum Ende der Ladung ermittelt werden. Damit das jeder Nutzer für sein(e) Fahrzeug(e) ermitteln kann, wäre es zusätzlich notwendig, die Soc-Daten im Ladelog mit aufzunehmen -> #6144 

Bei diesem Vorschlag geht es nur darum, diese Parameter händisch einstellbar zu machen. Es geht nicht darum, diese Einstellungen automatisiert zu erfassen und zu ändern.

andig, 2023-04-22:
Bitte nicht. Bevor wir drei (!) neue Parameter einführen möchte ich erstmal sehen welches echte Problem das lösen würde.

closed 2023-04-22 by andig

VolkerK62, 2023-04-22:
Sorry, evtl. mag ich es da ZU genau :-)  (Es sind sogar vier.)
Ob ein Unterschied von beispielsweise +/- 30 Minuten zwischen Plan und Realität ein echtes Problem ist, ist wahrscheinlich sehr individuell.

Hofyyy, 2023-04-22:
Ich vote für einen optionalen parameter. und zwar den Ladeverlust.

premultiply, 2023-04-22:
Der "Ladeverlust" ist von den 4 Parametern der unwichtigste und für Laien am schwierigsten messbare, der zudem bereits mit einem recht universell passenden internen Defaultwert belegt ist.
Er hat zudem auch nichts mit den anderen Parametern zu tun und kommt vom SoC-Estimator.

Hofyyy, 2023-04-22:
> Der "Ladeverlust" ist von den 4 Parametern der unwichtigste und für Laien am schwierigsten messbare, der zudem bereits mit einem recht universell passenden internen Defaultwert belegt ist. Er hat zudem auch nichts mit den anderen Parametern zu tun und kommt vom SoC-Estimator.

Wir können natürlich auch, alle, keinen oder einen anderen Wert nehmen. Ich finde den halt gut da es ein Prozentwert ist, der über die ganze Dauer geht. Man kann messen wie lange das Auto läd und wie lange der Plan initial geplant hat und dann anpassen. Die anderen Werte finde ich persönlich auch toll, glaube nur nicht das die normalen User die brauchen. Aber da mag es ja auch verschiedene Ansichten geben.

Das der Wert aus dem SoC Estimator kommt, ist reines Internes Design. Wichtig war für mich das er direkt auf die Ladekurve wirkt.

andig, 2023-04-23:
>Der "Ladeverlust" ist von den 4 Parametern der unwichtigste und für Laien am schwierigsten messbare,

Es ist ein sehr leicht verständlicher Parameter der sofort Einfluss auf "länger/kürzer" hat. Vermutung: Lieschen Müller wird sich nicht mit Maximalleistung bei Cutoff-Soc beschäftigen.

premultiply, 2023-04-23:
`virtualCapacity` wird nur initial aus `capacity * 0,9` berechnet. Anschließend wird dieser Wert aus der SoC-Änderung gewonnen.

andig, 2023-04-23:
Ahhh, good point. Stimmt- Effizienz ist ja jetzt aus dem Planner raus.