evcc-io/evcc #7622: Restzeit wird bei Änderung der Energiemenge angepasst

VolkerK62, 2023-04-23:
### Describe the bug

Mein eBike ist als Vehicle mit `capacity: 0.625` definiert und hängt an einem Shelly Plug S.
Eine Änderung der zu ladenden Energiemenge ändert jedoch nichts an der voraussichtlichen Ladedauer.

[image]

[image]

Log und/oder Konfig können bei Bedarf nachgeliefert werden

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

0.116.3 (b4e62a97)

consoleman, 2023-04-23:
Guten Abend, ich versuche genau diese Konfiguration hinzubekommen und scheitere.... könnte ich das evcc.yaml mal sehen?

andig, 2023-04-23:
Bitte neue Diskussion- hat hier nichts zu suchen.

andig, 2023-07-05:
@VolkerK62 funktioniert das evtl. mit der 118.6 wieder? Es gab ja grad einen Fix mit Energiemenge der ggf. auch hier zuschlagen könnte?

VolkerK62, 2023-07-05:
werde ich mal testen .... melde mich dann

VolkerK62, 2023-07-06:
hat sich insofern erledigt, dass keine Restzeit mehr angezeigt wird, sondern nur noch die vergangene Ladedauer.

closed 2023-07-06 by VolkerK62