evcc-io/evcc #7633: Websocket issue "[meter name] jq: empty result"

SirkoVZ, 2023-04-24:
### Describe the bug

When using Websocket with Volkszähler in meter configuration, example:

```
meters:
- name: gridmeter
  type: template
  template: volkszaehler-ws
  usage: grid
  host: 192.168.178.99
  port: 8082
  uuid: 93418420-92e6-11eb-82d3-redacted
```

then (obviously) since https://github.com/evcc-io/evcc/commit/6f1333946a7cb8adad78232b5831399239314b04 the meter values can't read properly and errors like:
`grid meter: jq: empty result`
occur.
Please also see https://github.com/evcc-io/evcc/discussions/7532

I trace logged the websocket and it seems to me whenever the last logged Volkszähler meter channel is not the needed one then the error occurs.

### Steps to reproduce

1. have several volkszaehler channels
2. set a meter using volkszaehler-ws
3. set log ws level to trace and site log level to debug

### Configuration details

```yaml
meters:
- name: gridmeter
  type: template
  template: volkszaehler-ws
  usage: grid
  host: 192.168.178.99
  port: 8082
  uuid: 93418420-92e6-11eb-82d3-redacted
```

### Log details

```text
Apr 24 08:59:27 raspi evcc[14802]: [ws    ] TRACE 2023/04/24 08:59:27 recv: {"version":"0.3","data":{"uuid":"8f32e780-a937-11ec-a5ac-redacted","tuples":[[1682319567986,-71,1]]}}
Apr 24 08:59:27 raspi evcc[14802]: [ws    ] TRACE 2023/04/24 08:59:27 recv: {"version":"0.3","data":{"uuid":"ad95eb20-a937-11ec-a757-redacted","tuples":[[1682319567986,-111,1]]}}
Apr 24 08:59:27 raspi evcc[14802]: [ws    ] TRACE 2023/04/24 08:59:27 recv: {"version":"0.3","data":{"uuid":"93418420-92e6-11eb-82d3-redacted","tuples":[[1682319567986,-366,1]]}}
Apr 24 08:59:27 raspi evcc[14802]: [ws    ] TRACE 2023/04/24 08:59:27 recv: {"version":"0.3","data":{"uuid":"fc035120-9a3d-11ec-ac4f-redacted","tuples":[[1682319567986,367.34693890380464,1]]}}
Apr 24 08:59:28 raspi evcc[14802]: [site  ] DEBUG 2023/04/24 08:59:28 ----
Apr 24 08:59:28 raspi evcc[14802]: [lp-1  ] DEBUG 2023/04/24 08:59:28 charge power: 0W
Apr 24 08:59:28 raspi evcc[14802]: [site  ] DEBUG 2023/04/24 08:59:28 pv power: 1076W
Apr 24 08:59:28 raspi evcc[14802]: [site  ] DEBUG 2023/04/24 08:59:28 battery soc: 85%
Apr 24 08:59:28 raspi evcc[14802]: [site  ] DEBUG 2023/04/24 08:59:28 battery power: -522W
Apr 24 08:59:28 raspi evcc[14802]: [site  ] ERROR 2023/04/24 08:59:28 grid meter: jq: empty result
Apr 24 08:59:28 raspi evcc[14802]: [ws    ] TRACE 2023/04/24 08:59:28 recv: {"version":"0.3","data":{"uuid":"9fe2f900-a937-11ec-b130-redacted","tuples":[[1682319568963,-183,1]]}}

As can be seen, the grid meter channel 93418420-92e6-11eb-82d3-redacted is not the last one before the [site  ] level  starts logging the values. (the last one here is fc035120-9a3d-11ec-ac4f-redacted)

here is a working example:
Apr 24 08:59:38 raspi evcc[14802]: [ws    ] TRACE 2023/04/24 08:59:38 recv: {"version":"0.3","data":{"uuid":"fc035120-9a3d-11ec-ac4f-redacted","tuples":[[1682319578001,180.4511277775343,1]]}}
Apr 24 08:59:38 raspi evcc[14802]: [ws    ] TRACE 2023/04/24 08:59:38 recv: {"version":"0.3","data":{"uuid":"ad95eb20-a937-11ec-a757-redacted","tuples":[[1682319578001,-104,1]]}}
Apr 24 08:59:38 raspi evcc[14802]: [ws    ] TRACE 2023/04/24 08:59:38 recv: {"version":"0.3","data":{"uuid":"93418420-92e6-11eb-82d3-redacted","tuples":[[1682319578001,-359,1]]}}
Apr 24 08:59:38 raspi evcc[14802]: [site  ] DEBUG 2023/04/24 08:59:38 ----
Apr 24 08:59:38 raspi evcc[14802]: [lp-1  ] DEBUG 2023/04/24 08:59:38 charge power: 0W
Apr 24 08:59:38 raspi evcc[14802]: [site  ] DEBUG 2023/04/24 08:59:38 pv power: 1087W
Apr 24 08:59:38 raspi evcc[14802]: [site  ] DEBUG 2023/04/24 08:59:38 battery soc: 85%
Apr 24 08:59:38 raspi evcc[14802]: [site  ] DEBUG 2023/04/24 08:59:38 battery power: -527W
Apr 24 08:59:38 raspi evcc[14802]: [site  ] DEBUG 2023/04/24 08:59:38 grid power: -359W
Apr 24 08:59:38 raspi evcc[14802]: [site  ] DEBUG 2023/04/24 08:59:38 site power: -846W
Apr 24 08:59:38 raspi evcc[14802]: [lp-1  ] DEBUG 2023/04/24 08:59:38 charge currents: [0 0 0]A
Apr 24 08:59:38 raspi evcc[14802]: [lp-1  ] DEBUG 2023/04/24 08:59:38 charge meter total import: 2448.033kWh
Apr 24 08:59:38 raspi evcc[14802]: [lp-1  ] DEBUG 2023/04/24 08:59:38 charger status: B
Apr 24 08:59:39 raspi evcc[14802]: [ws    ] TRACE 2023/04/24 08:59:39 recv: {"version":"0.3","data":{"uuid":"9fe2f900-a937-11ec-b130-redacted","tuples":[[1682319578986,-180,1]]}}

The last one is the correct one: 93418420-92e6-11eb-82d3-redacted
```

### What type of operating system are you running?

Linux

### Version

0.116.+

ecoCuyo, 2023-04-24:
Hi SirkoVZ,

bei mir konnte ich nachvollziehen, dass WS / vzlogger Push-Server ab der Version V0.116.1 nicht mehr geht - siehe Discussion #7532.
Aktuell läuft das System mit V0.116.0 problemlos.

Viele Grüße
Armin

andig, 2023-04-24:
Ich kann das Problem leider nicht nachvollziehen. In https://github.com/evcc-io/evcc/pull/7640 ist jetzt ein Testcase mit JQ der die erwarteten Daten liefert. Habt ihr eine Idee, woran es liegt?