evcc-io/evcc #7635: charger out of sync: expected disabled, got enabled // Juice Charger me Wallbox (Bender Controller)

ammann-ruchti, 2023-04-24:
### Describe the bug

Guten Tag,
gestern hatte ich wieder einmal das Problem, dass der Charger out of sync ging (letztes Mal passierte mir das am 27. März). Im Detail ging das wie folgt:
- Einstellung in EVCC war standardmässig PV-Überschussladen
- da bei wechselnder Bewölkung der Überschuss stark schwankte, habe ich im Web-Interface auf Min+PV umgeschaltet, während die Ladung gerade pausierte
- danach ist EVCC durcheinandergeraten. Es hat mir am Ladepunkt 11 kW Leistung angezeigt (im Web-Interface), die Produktion und Netzbezug zusammen waren jedoch viel geringer. Ich habe dann das Laden sofort gestoppt. Beim letzten Mal im März hatte ich festgestellt, dass damals trotz Anzeige am Ladepunkt von 11 kW und entsprechend hohem Eintrag in der Session DB jedoch effektiv nur mit 4.1 kW geladen wurde.
- das Logfile zeigt auch 3 Einträge mit charger out of sync
- Danach habe ich EVCC wieder auf pv-Laden gestellt, worauf alles ok war.

Mein **Vorschlag**: wenn EVCC bei verbundenem Fahrzeug ein Charger out of sync entdeckt, sollte die Ladung gestoppt und nach einer Pause (z.B. 1 Minute) neu gestartet werden. Dies könnte das Problem evtl. unschädlich machen.

Was meint ihr dazu?
Gruss, Hans-Jakob

### Steps to reproduce

Das Problem lässt sich nicht forcieren.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: ###

meters:
- type: template
  template: e3dc 
  usage: grid  
  host: 192.168.1.141  
  port: 502  
  name: Netz
- type: template
  template: e3dc 
  usage: pv  
  host: 192.168.1.141  
  port: 502  
  name: pv2
- type: template
  template: e3dc 
  usage: battery  
  host: 192.168.1.141  
  port: 502  
  name: battery3

chargers:
- type: template
  template: bender 
  host: 192.168.1.227  
  port: 502  
  name: wallbox5

vehicles:
- name: ev4
  type: template
  template: enyaq 
  title: Enyaq iV 80  
  user: ### 
  password: ###  
  vin: ###
  capacity: 80  
  phases: 3  
  mode: pv  
  minSoC: 15  
  targetSoC: 80  
  minCurrent: 6  
  maxCurrent: 16  

loadpoints:
- title: Carport
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Carport
  meters:
    grid: Netz
    pvs:
    - pv2
    batteries:
    - battery3

tariffs:
  currency: CHF
  grid: # Niedertarif plus Hochtarifzeiten
    type: fixed
    price: 0.300 # CHF/kWh
    zones:
      - days: Mo-Fr
        hours: 6-21
        price: 0.379
      - days: Sa
        hours: 6-12
        price: 0.379

  feedin: # Vergütung 24.5 Rp abzgl. Steuern
    type: fixed
    price: 0.176

messaging:
  events:
    start:
      title: Laden gestartet
      msg: Laden des Fahrzeugs ${vehicleTitle} im Modus ${mode} gestartet - aktuell geladen ${chargedEnergy:%.1fk} kWh - Ladestand = ${vehicleSoc:%.0f}%
    stop:
      title: Laden beendet
      msg: Laden des Fahrzeugs ${vehicleTitle} wurde beendet - neuer Ladestand ${chargedEnergy:%.1fk} kWh - ${vehicleOdometer} km - Ladestand = ${vehicleSoc:%.0f}%
  services:
  - type: telegram
    token: ###
    chats:
    - ###
```

### Log details

```text
-- Journal begins at Mon 2022-04-04 14:17:02 CEST, ends at Mon 2023-04-24 09:57:56 CEST. --
Apr 23 04:10:39 evcc-raspi evcc[5970]: [main  ] INFO 2023/04/23 04:10:39 new version available: 0.116.3
Apr 23 09:12:27 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 09:12:27 start charging ->
Apr 23 09:28:47 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 09:28:47 stop charging <-
Apr 23 09:54:07 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 09:54:07 start charging ->
Apr 23 10:09:07 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 10:09:07 stop charging <-
Apr 23 10:10:40 evcc-raspi evcc[5970]: [main  ] INFO 2023/04/23 10:10:40 new version available: 0.116.3
Apr 23 10:28:07 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 10:28:07 start charging ->
Apr 23 10:40:17 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 10:40:17 stop charging <-
Apr 23 10:46:57 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 10:46:57 start charging ->
Apr 23 11:30:07 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 11:30:07 stop charging <-
Apr 23 11:37:17 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 11:37:17 start charging ->
Apr 23 11:42:47 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 11:42:47 stop charging <-
Apr 23 11:47:47 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 11:47:47 start charging ->
Apr 23 11:49:47 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 11:49:47 stop charging <-
Apr 23 11:51:37 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 11:51:37 start charging ->
Apr 23 11:51:37 evcc-raspi evcc[5970]: [lp-1  ] WARN 2023/04/23 11:51:37 charger out of sync: expected disabled, got enabled
Apr 23 11:51:47 evcc-raspi evcc[5970]: [lp-1  ] WARN 2023/04/23 11:51:47 charger out of sync: expected disabled, got enabled
Apr 23 11:51:57 evcc-raspi evcc[5970]: [lp-1  ] WARN 2023/04/23 11:51:57 charger out of sync: expected disabled, got enabled
Apr 23 11:52:07 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 11:52:07 stop charging <-
Apr 23 12:20:27 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 12:20:27 start charging ->
Apr 23 14:42:27 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 14:42:27 stop charging <-
Apr 23 14:47:27 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 14:47:27 start charging ->
Apr 23 14:59:27 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 14:59:27 stop charging <-
Apr 23 15:19:07 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 15:19:07 start charging ->
Apr 23 15:24:07 evcc-raspi evcc[5970]: [lp-1  ] INFO 2023/04/23 15:24:07 stop charging <-
```

### What type of operating system are you running?

Linux

### Version

V0.116.2

Rainbow45-pixel, 2023-04-24:
Finde ich nicht gut. Im Fehlerfall lieber das Auto voll (evtl. mit Kosten verbunden) und fahrbereit als ein nicht geladenes Auto, mi dem ich nicht fahren kann. Man schaut ja nicht ständig nach, ob geladen wird.

ammann-ruchti, 2023-04-24:
Ich schlage ja einen Stopp und nach kurzer Pause einen Restart des Ladevorgangs vor. Und passieren tut es bei fast immer nur im Modus PV oder Min+PV, nicht beim Zielladen.

ammann-ruchti, 2023-07-05:
Hier das Log vom 24. Juni.

[evcc - Charger out of sync 20230624.log](https://github.com/evcc-io/evcc/files/11956356/evcc.-.Charger.out.of.sync.20230624.log)

andig, 2023-07-22:
> Soll man das einfach ignorieren oder gibt es da noch etwas zu tun?

When out of sync erkannt wird stellt evcc den aus seiner Sicht gewünschten Zustand wieder her. Das kann man ignorieren. Ggf. würde es bei Dir seltener auftreten wenn `interval` größer wäre. Ich sehe kein todo.

closed 2023-07-22 by andig