evcc-io/evcc #7636: Push notification receiver per vehicle

souhl, 2023-04-24:
**Is your feature request related to a problem? Please describe.**
We have two cars connected with evcc. One is used by my wife, the other one by me. The push messages are really useful for both of us, but only for our car. My wife gets spammed with messages from my car and vice versa.

**Describe the solution you'd like**
It would be great to be able to have the config of target device as part of the vehicle.

**Describe alternatives you've considered**
Instead of using evcc it would be possible to rebuild the notifications in Home Assistant.

andig, 2023-05-01:
Requirement understood. I'm wondering how we'd do that without hard-coding the decision. A type of filter seems appropriate, but filter on what? The message template applies to both cars, only the recipients don't.

PatriotIT, 2023-06-22:
This would also be a very nice feature for us. A vehicle based message is really helpful when you have more than one car! :)

StefanSchoof, 2024-08-28:
Possible workaround from @VolkerK62 in https://github.com/evcc-io/evcc/discussions/11908
> Doch, sollte funktionieren. Ich bekomme nur Nachrichten für ausgewählte Loadpoints
> 
> ```
> messaging:
>   services:
>   - type: ntfy
>     uri: https://ntfy.sh/***
>   events:
>     start:
>       title: Start {{.title}}
>       msg: |
>         {{ if eq .title "Carport" }} Start {{printf `(%02d:%02d)` now.Hour now.Minute}} Soc {{round (.vehicleSoc) 1}} {{- end }}
>         {{ if eq .title "Garage" }} Start {{printf `(%02d:%02d)` now.Hour now.Minute}} Soc {{round (.vehicleSoc) 1}} {{- end }} 
>         {{ if eq .title "WP -Anzeige-" }} Start {{printf `(%02d:%02d)` now.Hour now.Minute}} Temp {{.limitSoc}} {{- end }} 
>     stop:
>       title: Stop {{.title}}
>       msg: |
>         {{ if eq .title "Carport" }} Stop {{printf `(%02d:%02d)` now.Hour now.Minute}} Soc {{round (.vehicleSoc) 1}} {{- end }}
>         {{ if eq .title "Garage" }} Stop {{printf `(%02d:%02d)` now.Hour now.Minute}} Soc {{round (.vehicleSoc) 1}} {{- end }} 
>         {{ if eq .title "WP -Anzeige-" }} Stop {{printf `(%02d:%02d)` now.Hour now.Minute}} Temp {{.limitSoc}} {{- end }} 
> ```

closed 2025-01-06 by andig