evcc-io/evcc #7637: "Reduziere auf einphasig in s…" ob wohl nicht geladen wird/werden soll

horst789github, 2023-04-24:
### Describe the bug

Hi,

ich sehe ab und zu auf der GUI das hier:
[image]

der LP soll nicht laden und macht er auch nicht, da ja schon das ziel erreicht ist. Aber ich bekomme trotzdem ab und zu die Meldung das da ein Phasenwechsel ansteht.

### Steps to reproduce

weiß ich nicht

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: XXX

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true

# meter definitions
[... 207 lines cut ...]
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
-
```

### What type of operating system are you running?

Docker container

### Version

0.116.2

sash2kk, 2023-04-24:
Den Fehler habe ich auch, wenn ich das Auto spät abends anstecke und evcc auf PV steht und damit eigentlich keine Ladung starten sollte. (mit der 0.116.3) - ich vermute es ist ein Bug. Bei 0.115 ist mir das nicht aufgefallen...

TTXaser2, 2023-04-25:
ich hab den Fehler auch bei meinem VW ID.3, gerade vorher wieder gehabt: Auto eingesteckt, kein überschuss verfügbar, er hat auch nicht geladen, aber der Timer läuft runter exakt wie auf dem Bild oben.

andig, 2023-04-27:
> der LP soll nicht laden und macht er auch nicht, da ja schon das ziel erreicht ist. Aber ich bekomme trotzdem ab und zu die Meldung das da ein Phasenwechsel ansteht.

Das ist tricky. Die Frage hier ist ja: unter welchen Bedingungen sollen wir den Phasenwechsel machen? Wenn der soc schon erreicht ist sollten wir da gar nicht mehr rein laufen.

/cc @premultiply

premultiply, 2023-04-29:
Vom SoC sollte das nicht anhängig sein. Stichwort: Vorklima.

Umschaltung sollte IMHO erfolgen:
- Während des Ladens (Guard Timer)
- Bei bzw. vor Enable(false) -> Enable(true) (direkt, timerlos)

andig, 2023-05-08:
@horst789github @TTXaser2 @sash2kk  könntet Ihr mal ein trace log dazu anfertigen damit wir verstehen was im Hintergrund passiert wenn die Meldung kommt? Ich mache solange zu, weiter mit aussagekräftigem Logfile.

closed 2023-05-08 by andig