evcc-io/evcc #7641: Falsche Reichweite wird angezeigt

trombose009, 2023-04-24:
Hallo, 

neuerdings wird bei einem unserer Nissan leafs die Reichweite falsch angezeigt. Eigentlich sollte da was zwischen 300 und 400km stehen. Stattdessen nur ~80
[image]

Ich habe als Kapazität den Wert 62 in der config hinterlegt.  Beim anderen leaf funktioniert es auch. Bei diesem hier aber nicht. Wo könnte der Fehler von mir liegen?

VolkerK62, 2023-04-24:
Der aktuelle Wert kommt aus der Abfrage der Fahrzeug-API. Also müsste da der Fehler liegen.

andig, 2023-04-24:
Ohne Logfile ist keine Diagnose möglich. Gerne wieder melden wenns eins gibt und das einen Fehler in evcc zeigt.

closed 2023-04-24 by andig