evcc-io/evcc #7643: 'easee': unexpected status: 500 (Internal Server Error)

aBua-MUC, 2023-04-25:
### Describe the bug

Seit dem Update auf 116.3 und vermutlich dem damit verbundenen Neustart kommt der Fehler.
Auch beim Versuch einer neuen Konfiguartion wird diese mit Fehler bei der Wallbox abgebrochen.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: info  
levels:  
  cache: error  
plant: *****  
interval: 10s # control cycle interval  
  
sponsortoken: *****  
  
meters:  
- type: template  
  template: kostal-plenticore   
  id: 71    
  host: 192.168.1.xxx    
  port: 1502    
  usage: pv    
  modbus: tcpip    
  name: pv1  
- type: template  
  template: kostal-plenticore   
  id: 71    
  host: 192.168.1.xxx    
  port: 1502    
  usage: battery    
  modbus: tcpip    
  name: battery2  
- type: template  
  template: kostal-ksem-inverter   
  id: 71    
  host: 192.168.1.xxx    
  port: 1502    
  usage: grid    
  modbus: tcpip    
  name: grid3  
  
chargers:  
- type: template  
  template: easee   
  user: *****  
  password: *****  
  charger: EH******    
  name: wallbox4  
  
loadpoints:  
- title: Carport  
  charger: wallbox4  
  mode: pv  
  phases: 0 # Über den Wert kann die Automatik ein- bzw. ausgeschaltet werden. Mögliche Werte: 0|1|3  
  mincurrent: 6 # Definiert die minimal genutzte Stromstärke in Ampere (A) pro angeschlossener Phase von der Zuleitung zur Wallbox. Standardwert: 6  
  maxcurrent: 16 # Definiert die maximal zulässige Stromstärke in Ampere (A) pro angeschlossener Phase von der Zuleitung zur Wallbox. Standardwert: 16  
  vehicle: e-tron Q4  
  resetOnDisconnect: true  
  enable: # Definiert das Verhalten, wann im PV Modus das Laden begonnen wird.  
    threshold: -1400 # Ein positiver Wert für Netzbezug, ein negativer Wert für Export. Bei 0 muss der Export die minimale Ladeleistung erreicht haben.  
    delay: 1m  
  disable:  
    threshold: 400 # maximum import power (W)  
    delay: 5m  
  
vehicles:  
  - name: e-tron Q4  
    type: template  
    template: etron  
    title: e-tron Q4 # Wird in der Benutzeroberfläche angezeigt # Optional  
    capacity: 82 # Akku-Kapazität in kWh # Optional   
    user: *****  
    password: *****  
    vin: *****  
    mode: pv # enable PV-charging when vehicle is identified  
    minSoC: 40  # Midestladung der Batterie # immediately charge to 0% regardless of mode unless "off" (disabled)  
    targetSoC: 80  # Maximale Ladung der Batterie  
  
site:  
  title: Impulsgold  
  meters:  
    grid: grid3  
    pvs:  
    - pv1  
    batteries:  
    - battery2
```

### Log details

```text
today at 08:13:33[main  ] INFO 2023/04/25 08:13:33 evcc 0.116.3
today at 08:13:33[main  ] INFO 2023/04/25 08:13:33 using config file: /etc/evcc.yaml
today at 08:13:33[main  ] INFO 2023/04/25 08:13:33 starting ui and api at :7070
today at 08:13:43[db    ] INFO 2023/04/25 08:13:43 using sqlite database: /root/.evcc/evcc.db
today at 08:13:44[mqtt  ] INFO 2023/04/25 08:13:44 connecting evcc-1414739607 at tcp://192.168.1.10:1883
today at 08:13:45[main  ] FATAL 2023/04/25 08:13:45 cannot create charger 'wallbox4': cannot create charger 'template': cannot create charger 'easee': unexpected status: 500 (Internal Server Error)
today at 08:13:45[main  ] FATAL 2023/04/25 08:13:45 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

116.3

andig, 2023-04-25:
Das ist ein Fehler des Easee Backends.

closed 2023-04-25 by andig

aBua-MUC, 2023-04-25:
Da deren Cloud ja öfter mal Probleme hat, könnt ihr die Abfrage beim Start umgehen, damit evcc wenigstens läuft?
Oder gibt es eine lokale API für die Easee?

andig, 2023-04-25:
> damit evcc wenigstens läuft

Macht ja wenig sinn ohne Charger :(

aBua-MUC, 2023-04-25:
> > damit evcc wenigstens läuft
> 
> Macht ja wenig sinn ohne Charger :(

Hi Andi, deiner Sicht richtig, darüber hinaus hat evcc den Mehrwert, dass es alle Werte aus Wechselrichter. Stromzähler, Fahrzeug usw. ziehen kann und diese via MQTT an meinen Broker weiter gibt. Wenn evcc nicht startet = keine Werte.

jheinitz, 2023-04-26:
> Da deren Cloud ja öfter mal Probleme hat, könnt ihr die Abfrage beim Start umgehen, damit evcc wenigstens läuft? Oder gibt es eine lokale API für die Easee?

Lokale API für die Easee gibt's leider nicht. Habe ich auch schon beim Easee Support angefragt, aber da hieß es NEIN. Zumindest für die jetzt auf dem Markt befindliche Easee Home.