evcc-io/evcc #7647: Provide an API to support charging planning done in the EV and provided via the EVSE or online API

DerAndereAndi, 2023-04-25:
**Is your feature request related to a problem? Please describe.**
Some EVs (Audi Q8 e-tron, Taycan, Mercedes EQE/EQS, VW MEB EVs with software version >= 3.1) support communicating via ISO15118 over AC to the EVSE. These EVs also do support setting up charging plans in the EV itself or the app (which sends the data to the EV).

This includes:
- Minimum SoC: charging to that SoC level as quickly as possible
- Timers with dates/weekdays/time: Target SoC and/or climating
- Direct charging option: works like the "Now" mode in EVCC
- Maximum SoC: when no timer is set having an maximum SoC level

All this information can be communicated via compatible EVSEs and therefor be provided to evcc. Some EVSEs that support this are Porsche/Audi Mobile Charger Connect, OpenWB Pro (as far as I know technically, not sure if it provides an API for that yet), more coming.

Important note: the EV will not send SoC values for the targets but energy (Wh) demands.

**Describe the solution you'd like**
Right now some of this data, e.g. minimum SoC, is for some EVs fetched via the Online API. All the others values can only be set via the evcc API or the UI.

It would be great if there would be an Go API to provide all these either via the Online API or the EVSE. In that case the evcc UI might not allow switching modes or setting/changing the plans but mark the settings being fetched from the EV itself.

andig, 2023-04-25:
The EEbus charger already has access to loadpoint and it's api. Most functionality is available, schedule is not available at all. It's not clear what is missing?

DerAndereAndi, 2023-04-25:
Being able to:

1. Set minimum charge energy amount (which could be translated to an SoC for displaying purpose)
2. Set charging target time and energy amount via the charger implementation (which could be translated to an SoC for displaying purpose, but not 100% if the energy is required for climating as well which is unknown)
3. Get power tariffs for a duration
4. In these cases evcc should only limit the charging in the case of overload protection scenarios or PV surplus charging (depends on the EVs charging mode and demand)
5. A way to tell the UI that this data is from the EV and changes need to be done in the EV instead of the UI. To be clarified how this for all the possible cases could be done

So there are interfaces missing to do these and then handling of the different cases need to be defined wether what evcc will do and what not in adjusting the limits as well as how the UI would work and what it would allow.

DerAndereAndi, 2023-04-25:
No, these are not possible. But thanks for looking into it.