evcc-io/evcc #7665: Allready connected vehicle to elli pro wallbox is not recognized on evcc startup with v0.116.5 (works with 0.116.4)

hidminds, 2023-04-26:
### Describe the bug

The vehicle stays in status "not connected" on the loadpoint although it is already connected and ready to charge.
Also the connection state does not change in the web interface if I change the charge mode to minpv or now.

Workaround: Downgrade evcc to 0.116.4.

### Steps to reproduce

Stop evcc
2. Install evcc version 0.116.5
3. Connect a vehicle to a Elli Pro Wallbox which is connected to energy manager evcc.
4. Optional: Authenticate on the Elli Wallbox and enable the charging process. Via App or RFID card.
5. Start evcc

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: openhabian.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: info
  eebus: trace
  lp-1: debug
    #lp-2: info
  renault: info
  zoe: info

interval: 10s # control cycle interval

chargers:
- type: template
  template: ellipro
  ski: D3A7 1EA0 5AFD 9511 E352 18B7 5D8A 775B 6F8C 9678 # ski of charger from web interface
  ip: 192.168.150.70
  name: wallbox_garage # wallbox-2123A1ENSV
#- type: template
#  template: ellipro
#  ski: 99DF DA82 F7C8 894E 19EF 140B 54FF 6B8E 9B1C D0E1
#  ip: 192.168.150.121
#  name: wallbox_stellplatz # wallbox-2123A1EKJG

vehicles:
- type: template
  template: renault 
  title: Zoe  
  user: xxxxxxxxx@xxxxx.xx  
  password: xxxxxxxx  
  vin: VF1XXXXXXXXXXX  
  capacity: 50 # kWh 
  phases: 3 # uses a 3 wire charging cable 
  cache: 15m  
  mode: now # auto charging mode if this vehicle is connected to any charger
  minSoC: 40 # if under minSoC immediately charges to minSoC with mode 'now'. Ignores modes pv/minpv
  targetSoC: 80 # under targetSoc should charge with mode 'pv' up to targetSoC
  minCurrent: 8 # Zoe kann nur minimal 8A!
  maxCurrent: 16
  name: zoe

loadpoints:
- title: Garage
  charger: wallbox_garage
  vehicle: zoe # zoe = assigned default vehicle if any car connects
  #vehicles: # tries automatic identification of the vehicle via mac/rfid
  #  - zoe # does not work with eebus yet
  mode: minpv # off/pv/minpv/now
  phases: 3 # 0 = automatic (switch phases), 1|3
  #minCurrent: 6
  maxCurrent: 16
  resetOnDisconnect: true
  soc:
    poll:
      mode: charging # charging/connected
      interval: 60m
    estimate: true
#- title: Stellplatz
#  charger: wallbox_stellplatz
#  vehicle: # zoe = assigned default vehicle if any car connects
#  mode: pv
#  phases: 3
#  #mincurrent: 6
#  maxcurrent: 16
#  resetOnDisconnect: true
#  soc:
#    poll:
#      mode: charging # charging/connected
#      interval: 60m
#    estimate: true

meters:
#- name: wallbox_garage_meter
#  type: custom
#  power:
#    source: script
#    cmd: /bin/bash -c "evcc -c /home/openhabian/evcc.yaml charger wallbox_garage | grep 'Power' | sed 's/[^0-9]*\([0-9]\+\)W/\1/g'"
#    timeout: 10s
#- name: wallbox_stellplatz_meter
#  type: custom
#  power:
#    source: script
#    cmd: /bin/bash -c "evcc charger wallbox_stellplatz | grep 'Power' | sed 's/[^0-9]*\([0-9]\+\)W/\1/g'"
#    timeout: 10s
- name: dummyPV
  type: custom
  power:
    source: script
    cmd: /bin/bash -c "echo 0"
    timeout: 10s

site:
  title: Mein Zuhause
  meters:
    pvs:
    - dummyPV

eebus:
  interfaces:
    - enxb827ebfcf6fe
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          my very secret private key
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          MIIBhzCCAS2gAwIBAgIBATAKBggqhkjOPQQDAjAzMQswCQYDVQQGEwJERTENMAsG
          A1UEChMERVZDQzEVMBMGA1UEAwwMRVZDQ19IRU1TXzAxMB4XDTIyMTAxMjE1NTA0
          M1oXDTMyMTAwOTE1NTA0M1owMzELMAkGA1UEBhMCREUxDTALBgNVBAoTBEVWQ0Mx
          FTATBgNVBAMMDEVWQ0NfSEVNU18wMTBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IA
          BL9+vfj+Cnc88oQzcnUR7FwnVtjsKNiFrojnrN5+kvPayUN1FQ4R00QL0PIuprZp
          yx5Mo+HlOD6b9LHTj02Du0+jMjAwMA8GA1UdEwEB/wQFMAMBAf8wHQYDVR0OBBYE
          FDra1O1jilB+nJTItziOM+vHZTkbMAoGCCqGSM49BAMCA0gAMEUCIQDfSa+g1JGN
          ow/THIj8D08djmYuA+SD99VMUeWdbjsGWAIgX0yTa3SsF1QpiTFG/nqrF+PYOzsp
          xIUP4q4cNctTIro=
          -----END CERTIFICATE-----

tariffs:
  currency: EUR
  grid:
    # static grid price
    type: fixed
    price: 0.34 # EUR/kWh # ab 04/2023

  feedin: # Verguetung fuer eingespeisten Strom (nach EEG)
    type: fixed
    price: 0.0796 # EUR/kWh
```
```

### Log details

```text
See logfile.
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.116.5

jheinitz, 2023-04-26:
I think that this is not related to the Elli charger. I had the same situation today with my Easee Charger and the Skoda Enyaq. But to be honset, I did not investigate the full details yet.

thstubbe, 2023-04-26:
Found the solution for Easee wallbox - see #764:
Replace the mobilnummer if you use it for the login with your email. 

It is also working now for me again.

jheinitz, 2023-04-26:
> Found the solution for Easee wallbox - see #764:
> Replace the mobilnummer if you use it for the login with your email. 
> 
> It is also working now for me again.

Glad to hear that you found a solution for your problem. I think that this is related to the maintenance done by Easee recently. 
The original problem described here is definitely not related to the Easee maintenance. I have no errors or warnings from the Easee charger in my log when I faced the issue.

closed 2023-04-26 by andig

hidminds, 2023-04-27:
I can confirm this is fixed with the current nightly build version.
Thanks @DerAndereAndi and @andig ! :thumbsup: