evcc-io/evcc #7669: Vorschlag: Akku Min SoC Ladung in GUI ändern

horst789github, 2023-04-26:
Ich möchte gern wie bei den Autos/Wallboxen, mein Akku Ladung steuern. Vielleicht könnte man das so ähnlich in die GUI einbauen:

Beispiel:
[image]

und wenn man dann den Regler hin und her zieht, wäre das der Configwert: prioritySoc

oder alternativ das man den prioritySoc händisch irgendwo in der GUI einstellen könnte? Wäre das Möglich?

afuxi, 2023-04-26:
Das wäre schön.

lansester, 2023-04-29:
Eine Möglichkeit zur (vorübergehenden) Änderung des prioritySOC per GUI fände ich auch super.
Hintergrund:  Der prioritySOC hat für mich eine Jahreszeit-Abhängigkeit. Im Sommer kann er niedriger sein z.B. 50%, da ich mit dem Hausspeicher ohne Probleme über die Nacht komme. Je früher und je später im Frühjahr/Herbst, desto höher muss er sein.

thebrainkafka, 2023-04-29:
Gute Idee. Gerade zum jahreszeitenabhängigen anpassen.

DerBaerFreddy, 2023-05-02:
Das würde ich mir auch wünschen.

naltatis, 2023-05-03:
👆 ist gibts den passenden PR

PR #7802 (2023-05-08): Add battery settings UI
- 🔋 Add battery settings dialog
- 🪫 `prioritySoc` and `bufferSoc` can be changed
- 💽 changes are written to database and restored on restart (also added for smartCostLimit)

fixes #7669

In the future: Changing values is done via system select dropdowns (easy things first). Being able do vertically slide the limits could be an improvement later.

👉 try it here https://demo-next.evcc.io/

## Todos

- [x] proper en translation
- [x] animation glitches in dark mode
- [x] fix unreliable selects in desktop safari
- [x] capacity infos, multiple batteries
- [x] deal with initi [... cut]
closed 2023-05-08 by andig