evcc-io/evcc #7672: Timer until battery soc reached to start loadpoint charging

panzerdev, 2023-04-26:
**Is your feature request related to a problem? Please describe.**
When the battery is below the threshold `prioritySoc` and the battery is charging it's not easy to tell how long charging will still take.

**Describe the solution you'd like**
A countdown (like the one for phase change or enable disable timer) until when evcc thinks the soc for the battery is reached.

**Describe alternatives you've considered**
Alternative would be to show the calculated remaining charge time in the battery capacity popup.

andig, 2023-04-27:
Why would you want to do that? I understand the request technically, but adding timers beyond ev charging is probably nothing we'd want to do.

panzerdev, 2023-04-27:
It basically would be kind of the same thing like the enable timer. With the enable timer the user gets told when charging is going to start. With a timer for priority soc reached out would fulfill the same scope.

andig, 2023-04-29:
I don't think we want to add more timers, especially since the priority timer would _really_ require smoothing charging power over time to make projection meaningful.

closed 2023-04-29 by andig