evcc-io/evcc #7677: evcc geht nicht mehr seid Update

Speedster4736, 2023-04-26:
### Describe the bug

Ich erhalte habe EVCC aktualisiert und erhalte seit dem diese Fehler:

cannot create charger 'wallbox5'
cannot create charger 'template'
cannot create charger 'easee'
unexpected status: 500 (Internal Server Error)
Bitte überprüfe deine Konfigurationsdatei. Sollte dir die Fehlermeldung nicht weiterhelfen, suche in unseren [GitHub Diskussionen](https://github.com/evcc-io/evcc/discussions) nach einer Lösung.

Hinweis: Ein weiterer Grund könnte ein fehlerhaftes Gerät (Wechselrichter, Zähler, ...) sein. Überprüfe deine Netzwerkverbindungen.

Ich habe den Sponsortoken nachdem der Fehler kam, aktualisiert, dies brachte leider keine Verbesserung.

### Steps to reproduce

1. aktualisierte evcc auf latest
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: info  
levels:  
  cache: error  
  
interval: 10s # control cycle interval  
  
sponsortoken: *****  
  
meters:  
- type: template  
  template: senec-home   
  usage: grid    
  host: 192.168.178.114    
  name: grid1  
- type: template  
  template: senec-home   
  usage: pv    
  host: 192.168.178.114    
  name: pv2  
- type: template  
  template: senec-home   
  usage: battery    
  host: 192.168.178.114    
  name: battery3  
  
chargers:  
- type: template  
  template: easee   
  user: *****  
  password: *****  
  charger: EHH3Z55Z    
  name: wallbox5  
  
vehicles:  
- type: template  
  template: tesla   
  title: Markos Model 3 Performance    
  accessToken: *****  
  refreshToken: *****  
  vin: *****  
  capacity: 82    
  phases: 3    
  mode: pv    
  minCurrent: 6    
  maxCurrent: 16    
  name: ev4  
  
loadpoints:  
- title: Einfahrt  
  charger: wallbox5  
  mode: pv  
  phases: 3  
  mincurrent: 6  
  maxcurrent: 16  
  resetOnDisconnect: false  
  
site:  
  title: Mein Zuhause  
  meters:  
    grid: grid1  
    pvs:  
    - pv2  
    batteries:
```

### Log details

```text
Ich weiß nicht wie das geht.
```

### What type of operating system are you running?

Docker container

andig, 2023-04-26:
> Ich weiß nicht wie das geht.

Steht in der Doku ;). Hier ist die Lösung aber einfach: Email statt Telefonnummer für die Easee nehmen!

closed 2023-04-26 by andig

Speedster4736, 2023-04-26:
MEGA! Danke =)