evcc-io/evcc #7679: Erkennung angeschlossenes Fahrzeug funktioniert nicht (PSA-Bug?)

RustyDust, 2023-04-26:
### Describe the bug

### Situation:
- Ich habe in der Config zwei EV definiert, einen Corsa-e und einen Kia Niro EV. 
- die Wallbox hat nur einen (1) Anschluss

Wenn ich jetzt den Kia anschliesse, dann meldet mir evcc jedesmal `vehicle status: >1 matches, giving up`

### Ursache
Der Grund dafür ist, dass der Opel (via PSA-API) aus welchem Grund auch immer **immer** den Status `B` meldet, was dazu führt, dass evcc ZWEI angeschlossene Fahrzeuge erkennt und dann [hier](https://github.com/evcc-io/evcc/blob/0b7e6069274ed964ac391b7e5a792e076e294964/core/coordinator/coordinator.go#L72-L79) aufgibt. Siehe auch Log.

### Lösungsvorschlag ;)
#### Variante A 
Debug und ggf. fix der PSA-API (oder deren Antworten). Unterstütze ich gerne, wenn mir jemand sagt, wie ich evcc dazu kriege, die API-Antworten zu loggen.

#### Variante B
Nachdem ich nur einen Anschluss habe, ist es physikalisch gar nicht möglich, dass zwei Fahrzeuge gleichzeitig angeschlossen sind. Da der Coordinator - wenn ich den Code richtig lese - für jeden Loadpoint einzeln aufgerufen wird könnte er also bei Status `B` nicht einfach abbrechen, sondern mal weiter schauen, ob sich noch ein Fahrzeug mit Status `C` findet und das als das aktuelle verwenden. Das wirft zwar dann ein Problem auf, wenn evcc zwei Fahrzeuge mit `B` findet (oder auch zwei mit `C`) aber dann kann evcc von mir aus auch gerne so wie bisher aufgeben.
Insgesamt denke ich, dass das die bessere Variante wäre, da sie in den allermeisten Fällen funktionieren sollte und nicht jedesmal irgendwelchen, in der Regel von den Fahrzeug-Herstellern verschuldeten, Bugs in den APIs hinterhergelaufen werden muss.

### Steps to reproduce

Siehe oben, dürfte ohne Setup mit zwei Fahrzeugen und mindestens einem davon aus der PSA-Familie schwer reproduzierbar sein.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.whocares.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: redacted

interval: 10s # control cycle interval

meters:
- type: template
  template: sonnenbatterie
  usage: grid
  host: -redacted-
  port: 8080
  name: grid1
- type: template
  template: sonnenbatterie
  usage: pv
  host: -redacted-
  port: -redacted-
  name: pv2
- type: template
  template: sonnenbatterie
  usage: battery
  host: -redacted-
  port: -redacted-
  name: battery3

chargers:
- type: template
  template: etrel
  host: -redacted-
  port: -redacted-
  name: wallbox4

loadpoints:
- title: Garage
  charger: wallbox4
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

vehicles:
- name: opel
  type: template
  template: opel
  title: Corsa-e
  user: -redacted-
  password: -redacted-
  vin: -redacted-
  capacity: 50
  mode: pv
  targetSoC: 95
- name: niro
  type: template
  template: kia
  title: Kia Niro EV
  user: -redacted-
  password: -redacted-
  capacity: 64
  mode: minpv
  targetSoC: 80

site:
  title: WhoHome
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3
  prioritySoc: 50
  bufferSoc: 80
```

### Log details

```text
Apr 26 21:48:05 sru-pi evcc[279927]: [main  ] INFO 2023/04/26 21:48:05 evcc 0.116.5
Apr 26 21:48:05 sru-pi evcc[279927]: [main  ] INFO 2023/04/26 21:48:05 using config file: /opt/evcc/evcc.yaml
Apr 26 21:48:05 sru-pi evcc[279927]: [main  ] INFO 2023/04/26 21:48:05 starting ui and api at :7070
Apr 26 21:48:05 sru-pi evcc[279927]: [db    ] INFO 2023/04/26 21:48:05 using sqlite database: /var/lib/openhab/.evcc/evcc.db
Apr 26 21:48:07 sru-pi evcc[279927]: [site  ] INFO 2023/04/26 21:48:07 site config:
Apr 26 21:48:07 sru-pi evcc[279927]: [site  ] INFO 2023/04/26 21:48:07   meters:      grid ✓ pv ✓ battery ✓
Apr 26 21:48:07 sru-pi evcc[279927]: [site  ] INFO 2023/04/26 21:48:07     grid:      power ✓ energy ✗ currents ✗
Apr 26 21:48:07 sru-pi evcc[279927]: [site  ] INFO 2023/04/26 21:48:07     pv 1:      power ✓ energy ✗ currents ✗
Apr 26 21:48:07 sru-pi evcc[279927]: [site  ] INFO 2023/04/26 21:48:07     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
Apr 26 21:48:07 sru-pi evcc[279927]: [site  ] INFO 2023/04/26 21:48:07   vehicles:
Apr 26 21:48:07 sru-pi evcc[279927]: [site  ] INFO 2023/04/26 21:48:07     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
Apr 26 21:48:07 sru-pi evcc[279927]: [site  ] INFO 2023/04/26 21:48:07     vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] INFO 2023/04/26 21:48:07 loadpoint 1:
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] INFO 2023/04/26 21:48:07   mode:        minpv
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] INFO 2023/04/26 21:48:07   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] INFO 2023/04/26 21:48:07   meters:      charge ✓
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] INFO 2023/04/26 21:48:07     charge:    power ✓ energy ✗ currents ✓
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] DEBUG 2023/04/26 21:48:07 phase timer inactive
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] DEBUG 2023/04/26 21:48:07 pv timer inactive
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] DEBUG 2023/04/26 21:48:07 guard timer inactive
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] DEBUG 2023/04/26 21:48:07 max charge current: 6A
Apr 26 21:48:07 sru-pi evcc[279927]: [site  ] DEBUG 2023/04/26 21:48:07 ----
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] DEBUG 2023/04/26 21:48:07 charge power: 2560W
Apr 26 21:48:07 sru-pi evcc[279927]: [site  ] DEBUG 2023/04/26 21:48:07 pv power: 0W
Apr 26 21:48:07 sru-pi evcc[279927]: [site  ] DEBUG 2023/04/26 21:48:07 battery soc: 43%
Apr 26 21:48:07 sru-pi evcc[279927]: [site  ] DEBUG 2023/04/26 21:48:07 battery power: 3345W
Apr 26 21:48:07 sru-pi evcc[279927]: [site  ] DEBUG 2023/04/26 21:48:07 grid power: -5W
Apr 26 21:48:07 sru-pi evcc[279927]: [site  ] DEBUG 2023/04/26 21:48:07 site power: 3340W
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] DEBUG 2023/04/26 21:48:07 charge currents: [0.528 5.82 5.63]A
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] DEBUG 2023/04/26 21:48:07 charger status: C
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] INFO 2023/04/26 21:48:07 car connected
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] DEBUG 2023/04/26 21:48:07 vehicle api refresh
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] DEBUG 2023/04/26 21:48:07 pv timer elapse
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] DEBUG 2023/04/26 21:48:07 pv timer inactive
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] DEBUG 2023/04/26 21:48:07 charger: guard elapse
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] DEBUG 2023/04/26 21:48:07 guard timer inactive
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] INFO 2023/04/26 21:48:07 start charging ->
Apr 26 21:48:07 sru-pi evcc[279927]: [lp-1  ] DEBUG 2023/04/26 21:48:07 wake-up timer: stop
Apr 26 21:48:12 sru-pi evcc[279927]: [main  ] DEBUG 2023/04/26 21:48:12 vehicle status: C (Kia Niro EV)
Apr 26 21:48:13 sru-pi evcc[279927]: [main  ] DEBUG 2023/04/26 21:48:13 vehicle status: B (Corsa-e)
Apr 26 21:48:13 sru-pi evcc[279927]: [main  ] WARN 2023/04/26 21:48:13 vehicle status: >1 matches, giving up
Apr 26 21:48:13 sru-pi evcc[279927]: [lp-1  ] DEBUG 2023/04/26 21:48:13 pv charge current: 0A = 2.53A + -4.84A (3340W @ 3p)
Apr 26 21:48:13 sru-pi evcc[279927]: [site  ] DEBUG 2023/04/26 21:48:13 telemetry: charge: Δ4/4Wh @ 2560W
```

### What type of operating system are you running?

Linux

### Version

0.116.5

StefanSchoof, 2023-04-27:
> Debug und ggf. fix der PSA-API (oder deren Antworten). Unterstütze ich gerne, wenn mir jemand sagt, wie ich evcc dazu kriege, die API-Antworten zu loggen

Ich glaube die Antworten sind in der Log, wenn du das Log Level auf trace stellst.

premultiply, 2023-04-27:
Ein Fahrzeug kann ja auch zeitgleich extern z.B. öffentlich laden.

RustyDust, 2023-04-27:
> Ich glaube die Antworten sind in der Log, wenn du das Log Level auf trace stellst.

Indeed, danke. Dann schaue ich mal was die PSA-API sagt, sobald der Kia wieder da ist ;)

RustyDust, 2023-04-27:
> Ein Fahrzeug kann ja auch zeitgleich extern z.B. öffentlich laden.

Stimmt, hatte ich so nicht bedacht. Aber das würde dann auch bedeuten, dass im Szenario "eigenes Fahrzeug lädt extern, Gastfahrzeug lädt an meiner Wallbox" das Gastfahrzeug als eigenes Fahrzeug erkannt würde, oder?