evcc-io/evcc #7682: HomeAssistant Addon: "stealing" host name

fabsenet, 2023-04-27:
I am running home assistant with home assistant OS and in there, i am running evcc as an addon.

Usually, my home router (fritz box) asks devices on the network for their name and then they are accessible. This allows me to use my browser and to see the home assistant ui on http://homeassistant:8123 

But sometimes, it looks like evcc is responding to this host name query and my router is recording evcc as the name of the server. After this, http://homeassistant:8123 does no longer work but I have to use http://evcc:8123 to get to the home assistant ui! this is obviously not intended.

What i tried is to give a permanent host name in the router, but the active name resolution overrides it, so this is only temporary.

The solution could be, to allow me to disable (via addon config probably?) to disable the name resolution in evcc.

I am running other addons in homeassistant like vs code and nodered but did not observe the host name takeover there.

StefanSchoof, 2023-04-27:
What did you configure as host in your `evcc.yaml`? And since HA is using docker you should read this comment:

https://github.com/evcc-io/evcc/blob/9324f0c5187117d178727ffa75eace505a562aa6/evcc.dist.yaml#L7-L8

fabsenet, 2023-04-27:
this might be the fastest successful answer in a long time. I was only looking in the addon settings, because it was the right place for this setting in my mind. 

also, in my evcc config, I did not even had "network:" nor did I know, it existed.

I set the hostname there now to the same name `homeassistant` and I am reasonable sure, this will fix my issue.

Thank you @StefanSchoof !

closed 2023-04-27 by fabsenet