evcc-io/evcc #7683: Easee: Wieder mal: 500(Internal Server Error)

jpakusch, 2023-04-27:
### Describe the bug

Hallo,

Ich hatte gestern den Easee - "Bug" 500 (internal Server Error) mit EVCC Version 0.116.3.
Habe dann hier gelesen dass ich meine Mail-Adresse verwenden soll statt der Telefonnummer. Das hat auch Abhilfe geschaffen.
Danke für den tollen, schnellen Support und das tolle Forum an der Stelle!

Heute habe ich gesehen dass es nen Update gibt. Das also installiert (auf 0.116.5)

Und siehe da: der Bug ist wieder da. Mit beiden Usernamen, Mail-Adresse UND Telefonnummer.

### Steps to reproduce

1. Update to 0.116.5
2.
3.
...

### Configuration details

```yaml
chargers:  
- type: template  
  template: easee   
  user: *****  
  password: *****  
  charger: EHSCKS56    
  name: Easee  
  
loadpoints:  
- title: Stellplatz  
  charger: Easee  
  mode: minpv  
  phases: 0  
  mincurrent: 6  
  maxcurrent: 32  
  resetOnDisconnect: true
```

### Log details

```text
cannot create charger 'Easee'
cannot create charger 'template'
cannot create charger 'easee'
unexpected status: 500 (Internal Server Error)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.116.5

jpakusch, 2023-04-27:
Okay, war scheinbar ein Problem bei Easee.
EVCC startet nun immerhin wieder ohne dass ich was geändert habe.

jpakusch, 2023-04-27:
Modus "Min+PV" funktioniert aber weiterhin nicht. Laden wird nicht gestartet. Modus "Schnell" funktioniert.

closed 2023-04-27 by andig

jpakusch, 2023-04-27:
Danke!

andig, 2023-04-27:
> unexpected status: 500 (Internal Server Error)

Dann war dein Ticket sehr verwirrend!