evcc-io/evcc #7708: Nach 0.116.5 wird EV nicht mehr erkannt.

Gregor-de, 2023-04-28:
Hallo, 
habe eine Elli Pro und einen ID.4. 
Nach dem Update auf die 0.116.5 wird der ID.4 nicht mehr erkannt und lädt einfach mit 4kW als Hauslast, was eigentlich immer nur passiert, wenn die Elli die Verbindung zu EVCC verliert. Diese ist aber im Status verbunden. 
Ist das schon bekannt?

Gruß

klitzsparrow, 2023-04-28:
Moin, kann ich beim e-golf mit der Elli Pro nicht bestätigen, Fahrzeug wird erkannt... vielleicht ein Problem mit dem We Connect ID Zugang? ich nutze noch den alten We Connect Zugang.

Grüße

Corny2410, 2023-04-28:
Hallo,

Ich glaube nicht das es an EVCC liegt VW hat mal wieder mit Serverproblemen zu kämpfen so das EVCC keine Daten erhält aus der VW Cloud.

klitzsparrow, 2023-04-28:
was kommt denn beim Debug von Fahtzeug raus? `sudo evcc -l debug vehicle`

closed 2023-04-28 by andig