evcc-io/evcc #7711: Skoda Enyaq wird nicht identifiziert

stevoboy, 2023-04-28:
### Describe the bug

Guten Morgen zusammen,

ich habe hier zuhause zwei Wallboxen, eine OpenWB Pro und einen SMA EVCharger. 
An beiden wird unser Skoda Enyaq MJ 2023 nicht erkannt.

Bei der Pro bin ich mir sicher, dass diese MAC Adresse in der Wallbox ankommt.
Allerdings erkennt EVCC die ID nicht. Woran kann das liegen?
[image]

[image]

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: Enyaq
    type: template
    template: enyaq
    title: Enyaq # Wird in der Benutzeroberfläche angezeigt # Optional
    user: # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password:  # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 55 # Akkukapazität in kWh # Optional 
    identifiers:
      - "00:7D:FA:08:64:BD"
```

### Log details

```text
...
```

### What type of operating system are you running?

Linux

Corny2410, 2023-04-28:
Hallo,

Ich glaube nicht das es an EVCC liegt VW hat mal wieder mit Serverproblemen zu kämpfen so das EVCC keine Daten erhält aus der VW Cloud.

stevoboy, 2023-04-28:
> Hallo,
> 
> 
> 
> Ich glaube nicht das es an EVCC liegt VW hat mal wieder mit Serverproblemen zu kämpfen so das EVCC keine Daten erhält aus der VW Cloud.

Nach der Methode per MAC Adresse als Identifier brauche ich doch keine Verbindung zu Skoda Connect API, oder? Ich habe extra den Benutzernamen und Passwort wieder aus der YAML gelöscht, um zu vermeiden dass evcc über die Cloud geht, sondern nur den identifier verwendet, der über das ladekabel kommt.

andig, 2023-04-28:
Logfile fehlt.

closed 2023-04-28 by andig

stevoboy, 2023-04-28:
Log file liefere ich nach sobald ich herausgefunden habe wie ich das mit meiner Synology Docker Konstruktion auslesen kann.

Kann man die Identifikation über Skoda Connect und MAC identifier kombinieren?
D.h. SMA geht über skoda connect und openwb über mac identifier?

stevoboy, 2023-04-28:
Hier mal das LOG aus dem Container, das war gemeint richtig?
Ich habe gerade die User Daten wieder eingefügt, um zu Testen ob Variante 3 geht, über das Portal. Jedoch auch kein Ergebnis.
Wie soll eigentlich der Abgleich des SoC mit dem Portal erfolgen wenn der SoC dem EVCC gar nicht bekannt ist. Das muss ich doch über einen Datenpunkt erst hinzuspielen.

vehicles:
  - name: Enyaq
    type: template
    template: enyaq
    title: Enyaq # Wird in der Benutzeroberfläche angezeigt # Optional
    user: ...@gmail.com # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: ... # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 55 # Akkukapazität in kWh # Optional 
    #identifiers:
    #  - "00:7D:FA:08:64:BD"

```
[lp-1  ] DEBUG 2023/04/28 18:42:33 charge currents: [0 0 0]A                                                                                      
[lp-1  ] DEBUG 2023/04/28 18:42:33 charger status: B                                                                                              
[site  ] DEBUG 2023/04/28 18:42:42 ----                                                                                                           
[lp-1  ] DEBUG 2023/04/28 18:42:42 charge power: 0W                                                                                               
[lp-2  ] DEBUG 2023/04/28 18:42:43 charge power: 0W                                                                                               
[site  ] DEBUG 2023/04/28 18:42:43 pv power: 380W                                                                                                 
[site  ] DEBUG 2023/04/28 18:42:43 grid power: -294W                                                                                              
[site  ] DEBUG 2023/04/28 18:42:43 grid powers: [8 -54 -249]W                                                                                     
[site  ] DEBUG 2023/04/28 18:42:43 grid currents: [0.864 -1.01 -1.57]A                                                                            
[site  ] DEBUG 2023/04/28 18:42:43 site power: -294W                                                                                              
[lp-2  ] DEBUG 2023/04/28 18:42:43 charge currents: [-0 -0 -0]A                                                                                   
[lp-2  ] DEBUG 2023/04/28 18:42:43 charger status: A                                                                                              
[site  ] DEBUG 2023/04/28 18:42:52 ----                                                                                                           
[lp-1  ] DEBUG 2023/04/28 18:42:52 charge power: 0W                                                                                               
[lp-2  ] DEBUG 2023/04/28 18:42:53 charge power: 0W                                                                                               
[site  ] DEBUG 2023/04/28 18:42:53 pv power: 379W                                                                                                 
[site  ] DEBUG 2023/04/28 18:42:53 grid power: -282W                                                                                              
[site  ] DEBUG 2023/04/28 18:42:53 grid powers: [7 -50 -239]W                                                                                     
[site  ] DEBUG 2023/04/28 18:42:53 grid currents: [0.862 -1.01 -1.52]A                                                                            
[site  ] DEBUG 2023/04/28 18:42:53 site power: -282W                                                                                              
[lp-1  ] DEBUG 2023/04/28 18:42:53 charge currents: [0 0 0]A                                                                                      
[lp-1  ] DEBUG 2023/04/28 18:42:53 charger status: B                                                                                              
[lp-1  ] DEBUG 2023/04/28 18:42:53 vehicle api refresh                                                                                            
[site  ] DEBUG 2023/04/28 18:43:02 ----                                                                                                           
[lp-1  ] DEBUG 2023/04/28 18:43:02 charge power: 0W                                                                                               
[lp-2  ] DEBUG 2023/04/28 18:43:03 charge power: 0W                                                                                               
[site  ] DEBUG 2023/04/28 18:43:03 pv power: 366W                                                                                                 
[site  ] DEBUG 2023/04/28 18:43:03 grid power: -276W                                                                                              
[site  ] DEBUG 2023/04/28 18:43:03 grid powers: [9 -50 -235]W                                                                                     
[site  ] DEBUG 2023/04/28 18:43:03 grid currents: [0.862 -1.01 -1.53]A                                                                            
[site  ] DEBUG 2023/04/28 18:43:03 site power: -276W                                                                                              
[lp-2  ] DEBUG 2023/04/28 18:43:03 charge currents: [-0 -0 -0]A                                                                                   
[lp-2  ] DEBUG 2023/04/28 18:43:03 charger status: A            
```

andig, 2023-04-29:
> Bitte das Issue wieder öffnen, das Thema ist ja noch nicht erledigt.

Dann mach mal bitte ein trace log von der OpenWB Pro wo die Fahrzeugerkennung auch zu sehen ist.

stevoboy, 2023-04-29:
Also, ich habe gestern Abend das ganze noch mal als Windows 10 Version per Assistent aufgesetzt und eine YAML mit dem Fortgeschrittenen Assistenten erzeugt.
Auf Windows läuft es, und das Fahrzeug wird über die Skoda API identifiziert, über die MAC Addresse jedoch weiterhin nicht.
[image]

Ich habe dann einen komplett neuen Docker Container genau nach [dieser ](https://github.com/evcc-io/docs/files/10365841/Anleitung.EVCC.Synology.Docker.Elli.Charger.Connect-Pro.pdf )Anleitung aufgesetzt und die YAML von der Windows Version verwendet. Da geht die Fahrzeugerkennung per API dann wieder nicht mehr.
Ich vermute, das auch in einem Host Netzwerk die Anfragen des evcc an die Skoda API nur rausgehen, aber keine Rückmeldung durch die Firewall der Synology/Docker hereinkommt.
Dazu steht in @at4hawo1 Anleitung zur Installation auf dem Docker leider nichts.

LOGs im nächsten Post

stevoboy, 2023-04-29:
Siehe anderer thread, es lag am image.
Mit dem latest Image von evcc/ funktioniert es jetzt per api.
Leider immer noch nicht per mac Adresse

andig, 2023-04-29:
Es funktioniert per MAC, kannste mir glauben