evcc-io/evcc #7712: Elli Pro - lp-1 charge meter: data not available

Saharel001, 2023-04-28:
### Describe the bug

If i connect my car to the Wallbox, the Error: "charge meter: data not available" will print on the website. (see Logfile)
I must be activated the charging via app or via RFID before the car can charge. (my choice)
If i activate the charging via app or RFID the error is gone but so long as the car only are connecet the error still print eache 10 seconds on the Website

### Steps to reproduce

1. Connect the car but dont start charging via App oder RFID

### Configuration details

```yaml
chargers:
- type: template
  template: ellipro
  ski: xxxxxxx
  ip: xxxxxxx
  name: wallbox5

loadpoints:
- title: Garage
  charger: wallbox5
  vehicle: ev4
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
```

### Log details

```text
Apr 28 08:37:27 HW-PV-HEMS evcc[30378]: [lp-1  ] ERROR 2023/04/28 08:37:27 charge meter: data not available
Apr 28 08:37:28 HW-PV-HEMS evcc[30378]: [lp-1  ] ERROR 2023/04/28 08:37:28 charge meter: data not available
Apr 28 08:37:28 HW-PV-HEMS evcc[30378]: [lp-1  ] DEBUG 2023/04/28 08:37:28 charger status: B
Apr 28 08:37:28 HW-PV-HEMS evcc[30378]: [eebus ] DEBUG 2023/04/28 08:37:28 Send: d:_i:47859_Elli-Wallbox-2103A15FF5 notify 721 DeviceDiagnosisHeartbeatData
Apr 28 08:37:28 HW-PV-HEMS evcc[30378]: [eebus ] TRACE 2023/04/28 08:37:28 Send: d5e1516bcbda56dc288d8b9a9e5885efbf7b00dd {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.1.1"},{"addressSource":[{"device":"d:_i:EVCC_HEMS-EVCC-3931373562393063"},{"entity":[1]>
Apr 28 08:37:28 HW-PV-HEMS evcc[30378]: [eebus ] TRACE 2023/04/28 08:37:28 Recv: d5e1516bcbda56dc288d8b9a9e5885efbf7b00dd {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2103A15FF5"},{"entity":[1,1]>
Apr 28 08:37:28 HW-PV-HEMS evcc[30378]: [eebus ] DEBUG 2023/04/28 08:37:28 Recv: d:_i:47859_Elli-Wallbox-2103A15FF5: to DeviceConfiguration notify 807 DeviceConfigurationKeyValueListData
Apr 28 08:37:28 HW-PV-HEMS evcc[30378]: [eebus ] TRACE 2023/04/28 08:37:28 Recv: d5e1516bcbda56dc288d8b9a9e5885efbf7b00dd {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2103A15FF5"},{"entity":[1,1]>
Apr 28 08:37:28 HW-PV-HEMS evcc[30378]: [eebus ] DEBUG 2023/04/28 08:37:28 Recv: d:_i:47859_Elli-Wallbox-2103A15FF5: to DeviceConfiguration notify 809 DeviceConfigurationKeyValueListData
Apr 28 08:37:28 HW-PV-HEMS evcc[30378]: [eebus ] TRACE 2023/04/28 08:37:28 Recv: d5e1516bcbda56dc288d8b9a9e5885efbf7b00dd {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2103A15FF5"},{"entity":[1,1]>
Apr 28 08:37:28 HW-PV-HEMS evcc[30378]: [eebus ] DEBUG 2023/04/28 08:37:28 Recv: d:_i:47859_Elli-Wallbox-2103A15FF5: to ElectricalConnection notify 810 ElectricalConnectionPermittedValueSetListData
Apr 28 08:37:28 HW-PV-HEMS evcc[30378]: [eebus ] TRACE 2023/04/28 08:37:28 Recv: d5e1516bcbda56dc288d8b9a9e5885efbf7b00dd {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2103A15FF5"},{"entity":[1,1]>
Apr 28 08:37:28 HW-PV-HEMS evcc[30378]: [eebus ] DEBUG 2023/04/28 08:37:28 Recv: d:_i:47859_Elli-Wallbox-2103A15FF5: to ElectricalConnection notify 811 ElectricalConnectionPermittedValueSetListData
Apr 28 08:37:28 HW-PV-HEMS evcc[30378]: [eebus ] TRACE 2023/04/28 08:37:28 Recv: d5e1516bcbda56dc288d8b9a9e5885efbf7b00dd {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2103A15FF5"},{"entity":[1,1]>
Apr 28 08:37:28 HW-PV-HEMS evcc[30378]: [eebus ] DEBUG 2023/04/28 08:37:28 Recv: d:_i:47859_Elli-Wallbox-2103A15FF5: to ElectricalConnection notify 812 ElectricalConnectionPermittedValueSetListData
Apr 28 08:37:29 HW-PV-HEMS evcc[30378]: [eebus ] DEBUG 2023/04/28 08:37:29 Send: d:_i:47859_Elli-Wallbox-2103A15FF5 notify 722 DeviceDiagnosisHeartbeatData
Apr 28 08:37:29 HW-PV-HEMS evcc[30378]: [eebus ] TRACE 2023/04/28 08:37:29 Send: d5e1516bcbda56dc288d8b9a9e5885efbf7b00dd {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.1.1"},{"addressSource":[{"device":"d:_i:EVCC_HEMS-EVCC-3931373562393063"},{"entity":[1]>
Apr 28 08:37:30 HW-PV-HEMS evcc[30378]: [eebus ] DEBUG 2023/04/28 08:37:30 Send: d:_i:47859_Elli-Wallbox-2103A15FF5 notify 723 DeviceDiagnosisHeartbeatData
Apr 28 08:37:30 HW-PV-HEMS evcc[30378]: [eebus ] TRACE 2023/04/28 08:37:30 Send: d5e1516bcbda56dc288d8b9a9e5885efbf7b00dd {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.1.1"},{"addressSource":[{"device":"d:_i:EVCC_HEMS-EVCC-3931373562393063"},{"entity":[1]>
Apr 28 08:37:30 HW-PV-HEMS evcc[30378]: [eebus ] TRACE 2023/04/28 08:37:30 Recv: d5e1516bcbda56dc288d8b9a9e5885efbf7b00dd {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2103A15FF5"},{"entity":[1,1]>
Apr 28 08:37:30 HW-PV-HEMS evcc[30378]: [eebus ] DEBUG 2023/04/28 08:37:30 Recv: d:_i:47859_Elli-Wallbox-2103A15FF5: to DeviceConfiguration notify 815 DeviceConfigurationKeyValueListData
Apr 28 08:37:30 HW-PV-HEMS evcc[30378]: [eebus ] TRACE 2023/04/28 08:37:30 Recv: d5e1516bcbda56dc288d8b9a9e5885efbf7b00dd {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2103A15FF5"},{"entity":[1,1]>
Apr 28 08:37:30 HW-PV-HEMS evcc[30378]: [eebus ] DEBUG 2023/04/28 08:37:30 Recv: d:_i:47859_Elli-Wallbox-2103A15FF5: to DeviceConfiguration notify 816 DeviceConfigurationKeyValueListData
Apr 28 08:37:30 HW-PV-HEMS evcc[30378]: [eebus ] DEBUG 2023/04/28 08:37:30 Send: d:_i:47859_Elli-Wallbox-2103A15FF5 notify 724 DeviceDiagnosisHeartbeatData
Apr 28 08:37:30 HW-PV-HEMS evcc[30378]: [eebus ] TRACE 2023/04/28 08:37:30 Send: d5e1516bcbda56dc288d8b9a9e5885efbf7b00dd {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.1.1"},{"addressSource":[{"device":"d:_i:EVCC_HEMS-EVCC-3931373562393063"},{"entity":[1]>
Apr 28 08:37:30 HW-PV-HEMS evcc[30378]: [eebus ] TRACE 2023/04/28 08:37:30 Recv: d5e1516bcbda56dc288d8b9a9e5885efbf7b00dd {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2103A15FF5"},{"entity":[1,1]>
Apr 28 08:37:30 HW-PV-HEMS evcc[30378]: [eebus ] DEBUG 2023/04/28 08:37:30 Recv: d:_i:47859_Elli-Wallbox-2103A15FF5: to ElectricalConnection notify 817 ElectricalConnectionPermittedValueSetListData
Apr 28 08:37:30 HW-PV-HEMS evcc[30378]: [eebus ] TRACE 2023/04/28 08:37:30 Recv: d5e1516bcbda56dc288d8b9a9e5885efbf7b00dd {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2103A15FF5"},{"entity":[1,1]>
Apr 28 08:37:30 HW-PV-HEMS evcc[30378]: [eebus ] DEBUG 2023/04/28 08:37:30 Recv: d:_i:47859_Elli-Wallbox-2103A15FF5: to ElectricalConnection notify 819 ElectricalConnectionPermittedValueSetListData
Apr 28 08:37:30 HW-PV-HEMS evcc[30378]: [eebus ] TRACE 2023/04/28 08:37:30 Recv: d5e1516bcbda56dc288d8b9a9e5885efbf7b00dd {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2103A15FF5"},{"entity":[1,1]>
Apr 28 08:37:30 HW-PV-HEMS evcc[30378]: [eebus ] DEBUG 2023/04/28 08:37:30 Recv: d:_i:47859_Elli-Wallbox-2103A15FF5: to ElectricalConnection notify 820 ElectricalConnectionPermittedValueSetListData
Apr 28 08:37:31 HW-PV-HEMS evcc[30378]: [eebus ] TRACE 2023/04/28 08:37:31 Recv: d5e1516bcbda56dc288d8b9a9e5885efbf7b00dd {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.3.0"},{"addressSource":[{"device":"d:_i:47859_Elli-Wallbox-2103A15FF5"},{"entity":[1,1]>
[... cut ...]
Apr 28 08:37:47 HW-PV-HEMS evcc[30378]: [lp-1  ] DEBUG 2023/04/28 08:37:47 charger status: C
Apr 28 08:37:47 HW-PV-HEMS evcc[30378]: [lp-1  ] DEBUG 2023/04/28 08:37:47 soc estimated: 34.08% (vehicle: 34.00%)
Apr 28 08:37:47 HW-PV-HEMS evcc[30378]: [lp-1  ] DEBUG 2023/04/28 08:37:47 vehicle soc: 34%
Apr 28 08:37:47 HW-PV-HEMS evcc[30378]: [lp-1  ] DEBUG 2023/04/28 08:37:47 vehicle range: 19km
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.116.5 (d7d1e0bf)

premultiply, 2023-04-28:
@DerAndereAndi dein Fachgebiet ;)

DerAndereAndi, 2023-04-28:
This message can not be completely avoided the way evcc works and with the Software issues the Elli software has.

andig, 2023-04-28:
Du könntest `api.ErrNotAvailable` zurück geben und wir den Fehler ignorieren. Spätestens mit Lastmanagement wird das allerdings doof, aber das geht mit der Elli eh nicht mangels Phasenströmen.

DerAndereAndi, 2023-04-28:
Ob und was man tun könnte, würde ich gerne im PR besprechen. Sonst kommen wir durcheinander. Das Thema um das es hier geht, ist nicht zum ersten Mal hier.

andig, 2023-05-03:
Diskussion zur Lösung weiter im PR.

closed 2023-05-03 by andig