evcc-io/evcc #7724: Tesla API changes

peezy45, 2023-04-28:
Hi together,

I think Tesla has changed some things on their end as starting this morning my vehicleSoC doesn't update anymore on any of our three cars. 
Tried the nightly as well as the stable version.
I'm getting vehicleSoC and vehicle odometer 404 errors.

[image]

Sapippo, 2023-04-28:
Wollte auch gerade deswegen ein Ticket aufmachen. Selbes Problem hier.
Heute morgen hab ich auf meinem Model Y das 2023.12.1.1 Update installiert.
Seit dem Kein SOC Status mehr. 
"vehicle soc: 404 Not Found"

peezy45, 2023-04-28:
> Wollte auch gerade deswegen ein Ticket aufmachen. Selbes Problem hier.
> 
> Heute morgen hab ich auf meinem Model Y das 2023.12.1.1 Update installiert.
> 
> Seit dem Kein SOC Status mehr. 
> 
> "vehicle soc: 404 Not Found"

Das haben wir auf unserem Model Y vor zwei Tagen auch bekommen , danach funktionierte es aber noch. Hab das Problem erst seit heute Nacht .. vermute mal dass Tesla serverseitig irgendwas verändert hat. Bin schon mal froh dass ich nicht alleine damit bin 😅

Sapippo, 2023-04-28:
> Das haben wir auf unserem Model Y vor zwei Tagen auch bekommen , danach funktionierte es aber noch. Hab das Problem erst seit heute Nacht .. vermute mal dass Tesla serverseitig irgendwas verändert [...]

Dann evtl. ein Zufall mit dem Update heute morgen. Jedenfalls keine SOC und damit keine Ladesteuerung mehr :-/.
Hoffe das ist ein schneller fix.

NIREKI, 2023-04-28:
Kann das Problem so auch bestätigen. Bin auf EVCC Version 0.116.3, Fehler 404 vehicle not found.

peezy45, 2023-04-28:
Okay, dann scheint es tatsächlich etwas bei Tesla zu sein

andig, 2023-04-28:
Das ist wohl https://github.com/bogosj/tesla/issues/97

peezy45, 2023-04-28:
> Das ist wohl [bogosj/tesla#97](https://github.com/bogosj/tesla/issues/97)

Klingt plausibel... ist es sehr viel Arbeit den neuen Endpoint zu integrieren ? 
Ernstgemeinte Frage .. bin da nicht so im Thema 😅

romerod, 2023-04-29:
Ist vielleicht ein wenig off topic, aber was sind die Nachteile Teslas über teslamate (mqtt) einzubinden?

Hier wurde das schon mal ein wenig besprochen #5909. Ich sehe auch noch den zusätzlichen Vorteil, dass der SoC im evcc auch dann aktuell wäre wenn das Auto nicht angeschlossen ist.

Nutzt evcc die tesla API auch zum Befehle senden?

peezy45, 2023-04-29:
> Ist vielleicht ein wenig off topic, aber was sind die Nachteile Teslas über teslamate (mqtt) einzubinden?
>
>
>
> Hier wurde das schon mal ein wenig besprochen #5909. Ich sehe auch noch den zusätzlichen [...]

Habe leider keine Erfahrungswerte mit Teslamate aber ja , die Tesla API wird für Start / Stop sowie die Charging Amps Steuerung genutzt.

romerod, 2023-04-29:
> 

Was hat den Vorrang betreffend charging amps? Bei mir steuerts ja auch die easee Wallbox.

mettibuh, 2023-04-29:
Grundsätzlich scheint es aber noch offen zu sein. 
Via iobroker Adapter "tesla-motors" kann ich zumindest noch per 
tesla-motors.1.LRW3E7FSXNC5xxxxxx.remote.set_charging_amps-charging_amps die Stromstärke sowie über
tesla-motors.1.LRW3E7FSXNC5xxxxxx.remote.charge_start und _stop die Ladevorgänge fernsteuern. Soweit ich Laie das verstehe müsste das ählich kommunizieren, da der Adapter auch die Tokens benögit.
Das lässt zumindest hoffen, dass es für den evcc auch eine Lösung geben wird =)
Schönes WE und VG

peezy45, 2023-04-29:
> Grundsätzlich scheint das API aber noch offen zu sein. 
> 
> Via iobroker Adapter "tesla-motors" kann ich zumindest noch per 
> 
> tesla-motors.1.LRW3E7FSXNC5xxxxxx.remote.set_charging_amps-charging_amps die Stromstärke sowie über
> 
> tesla-motors.1.LRW3E7FSXNC5xxxxxx.remote.charge_start und _stop die Ladevorgänge fernsteuern. Soweit ich Laie das verstehe müsste das ähnlich kommunizieren, da der Adapter auch die Tokens benötigt.
> 
> Das lässt zumindest hoffen, dass es für den evcc auch eine Lösung geben wird =)
> 
> Schönes WE und VG
> 
> 

Ja stimmt , die Ampere und Start / Stop funktioniert bei mir auch noch. Aber der vehicleSoC und vehicleOdometer stehen nicht mehr zur Verfügung

jingej, 2023-04-29:
wäre ja nun interessant, woher dann teslamate den SOC un die km nimmt, denn das geht hier noch

peezy45, 2023-04-29:
Ich vermute mal dass Teslamate schon früher auf die neue API umgestellt hat.. die sind ja wohl schon eine Zeit lang parallel gelaufen wenn ich das richtig gelesen habe.

Superexpertman, 2023-04-29:
Same problem here with Tesla MY, firmware version 2023.12.1.1, EVCC version 0.116.6, error "vehicle soc: 404 Not Found".
Any workaround here with Tesla API how to proceed?

cschlipf, 2023-04-29:
Dasselbe hier wie bei @jingej . Teslamate funktioniert noch. Seit ich heute Nacht das Tesla Update auf 2023.12.1.1 bekommen habe, hat EVCC ein Problem.

wiggal, 2023-04-29:
natürlich auch bei unserem Model3 404 not found.😭

wernerb, 2023-04-29:
Brand new here and was confused why my setup was not working. I am guessing this is related:

This is the log at startup
```
[main  ] ERROR 2023/04/29 22:00:19 creating vehicle Tesla failed: cannot create vehicle 'template': cannot create vehicle 'tesla': cannot get vehicles: Get "https://owner-api.teslamotors.com/api/1/vehicles": oauth2: cannot fetch token: 400 Bad Request
Response: {"error":"server_error","error_description":"Internal server error","error_uri":"https://auth.tesla.com/error/reference/0c3f68ed-ccc1-4c3b-8e43-9751672647a5-1682798419232"}
````

andig, 2023-04-29:
Not related…

jingej, 2023-04-29:
man muss nun aus https://owner-api.teslamotors.com/api/1/vehicles/{#VID}/vehicle_data die werte rausparsen, sollte zu machen sein, ist halt unpraktisch und auch langsamer, weil zuviele Daten auf einmal kommen, aber wird schon gehn, ich bin da zuversichtlich

Sudden24, 2023-04-30:
same here. Tesla M3 and Y can only be charged via "Guest Car" since SoC is not found.

kbman22, 2023-04-30:
Gleiches Problem hier...Model Y mit 2023.12.1.1 und evcc v0.116.6...alle paar Sekunden die Meldung "vehicle soc: 404 Not Found" :(

tofue78, 2023-04-30:
Moin, gleiches Problem leider auch bei mir. Hab die 0.116.6 mit der 2023.12.1.1

Hab schon ein Downgrade auf die 0.116.1 gemacht, die Tesla Token neu erstellt und trotzdem kommt beim erstellen der YAML "404 Not Found".
Da ich als Hauptladestation den Go-E Gemini benutze, ist es nicht so tragisch, da ich trotzdem weiter Sonne laden kann.
Ich nutze aber ab und zu den TWC3, da brauch ich natürlich wieder die Steuerung durch die Tesla API.
:(

bicaluv, 2023-04-30:
Das Problem hat nichts mit der Tesla Version (ich habe noch 2023.6.11) und auch nichts mit der evcc Version zu tun. Ein Downgrade macht also keinen unterschied. Tesla hat die API geändert und das muss halt nachgebessert werden.
Wenn's fertig ist, steht es hier bzw. der Bug wird dann geschlossen.

naltatis, 2023-04-30:
Hier @andig's Draft PR für `bogosj/tesla`
https://github.com/bogosj/tesla/pull/98

horst789github, 2023-04-30:
gibt es irgendwas was wir als tesla fahrer liefern sollen damit das problem gefixt wird? bei teslamate gibt es ja keine probleme, daher scheint es ja eine lösung zu geben.

cschlipf, 2023-04-30:
> gibt es irgendwas was wir als tesla fahrer liefern sollen damit das problem gefixt wird? bei teslamate gibt es ja keine probleme, daher scheint es ja eine lösung zu geben.

Siehe Kommentar von @jingej: https://github.com/evcc-io/evcc/issues/7724#issuecomment-1528988380

Teslamate macht das genau so - macht hier ja auch Sinn, da hier jedesmal deutlich mehr Daten als nur Vehicle SOC ausgelesen werden.

stebo1970, 2023-04-30:
Ich bin da guter Hoffnung, dass die Experten das Problem lösen werden. 
Viel Erfolg und allen ein schönes langes Wochenende😄

user1272349087234, 2023-05-01:
Kann mir vielleicht einer helfen und sagen wie ich die neue tesla.go auf dem raspberry pi installiert bekomme?

closed 2023-05-01 by andig

cschlipf, 2023-05-01:
Yay! Es tut wieder mit 0.116.7

Vielen Dank für den schnellen Fix.

LuRhe, 2023-05-01:
Super, habt vielen lieben Dank für Eure schnellen Anpassungen, wenn Tesla wieder was geändert hat!

tofue78, 2023-05-01:
Super, danke👍🏻 Läuft wieder. Und deshalb bin ich gerne Sponsor😎

kbman22, 2023-05-01:
Top Arbeit - vielen Dank für die schnelle Lösung!

andig, 2023-05-01:
Es ist wirklich nett, dass ihr Euch bedankt- ich schließe den Thread trotzdem um das "noise" zu reduzieren. Danke!