evcc-io/evcc #7728: Skoda Enyaq not created

stevoboy, 2023-04-28:
### Describe the bug

Skoda Enyaq is not created as vehicle. Please check log.

### Steps to reproduce

1. Create YAML file
2. Start Docker on Synology DSM
3. In UI only Guest Car shows up and unidentified

### Configuration details

```yaml
---
sponsortoken: ...
network:
  schema: http
  host: evcc.local
  port: 7070
log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
interval: 10s
meters:
  - name: SHM
    type: template
    template: sma-home-manager
    usage: grid
    host: ...
  - name: SMA10
    type: template
    template: sma-inverter
    usage: pv
    host: ...
    password: ...
chargers:
  - name: OpenWB
    type: template
    template: openwb-pro
    host: ...
  - name: EVcharger
    type: template
    template: smaevcharger
    host: ...
    user: My User Name
    password: My Password
vehicles:
  - name: enyaq
    type: template
    template: enyaq
    title: enyaq
    user: My User name
    password: My Password
    capacity: 50
    identifiers:
      - "00:7D:FA:06:3E:6A"
loadpoints:
  - title: Küchenladepunkt
    charger: OpenWB
    mode: pv
    vehicles:
      - enyaq
  - title: Garagenladepunkt
    charger: EVcharger
    mode: pv
site:
  - title: Zuhause
    meters:
      grid: SHM
      pv: SMA10
tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.4
    zones:
      - days: Mo-Fr
        hours: 2-5
        price: 0.4
      - days: Sa,So
        price: 0.4
  feedin:
    type: fixed
    price: 0.08
```

### Log details

```text
[main  ] INFO 2023/04/28 22:35:09 evcc 0.111.1                                                                                                    
[main  ] INFO 2023/04/28 22:35:09 using config file: /etc/evcc.yaml                                                                               
[main  ] INFO 2023/04/28 22:35:09 starting ui and api at :7070                                                                                    
[main  ] ERROR 2023/04/28 22:35:14 creating vehicle enyaq failed: cannot create vehicle 'template': cannot create vehicle 'enyaq': Get "https://id
entity.vwgroup.io/oidc/v1/authorize?client_id=f9a2359a-b776-46d9-bd0c-db1904343117%40apps_vw-dilab_com&code_challenge=Y0gsmCf0VRQeLCSIGGjtkoLiuFuO
FheMEKxbvFs88cs&code_challenge_method=S256&redirect_uri=skodaconnect%3A%2F%2Foidc.login%2F&response_type=code+id_token&scope=openid+mbb+profile&no
nce=GlFSrQcNKDIWriOeOMExanhzZMUlnWaRrekDRBFlmwq&state=22900cb2-d193-4ae2-9efb-1d1ea2ef5c7d": x509: certificate signed by unknown authority        
[lp-1  ] WARN 2023/04/28 22:35:14 vehicles option is deprecated                                                                                   
[lp-2  ] WARN 2023/04/28 22:35:14 phases not configured, assuming 3p                                                                              
[site  ] INFO 2023/04/28 22:35:14 site config:                                                                                                    
[site  ] INFO 2023/04/28 22:35:14   meters:      grid ✓ pv ✓ battery ✗                                                                         
[site  ] INFO 2023/04/28 22:35:14     grid:      power ✓ energy ✓ currents ✓                                                                   
[site  ] INFO 2023/04/28 22:35:14     pv 1:      power ✓ energy ✓ currents ✓                                                                   
[site  ] INFO 2023/04/28 22:35:14   vehicles:                                                                                                     
[site  ] INFO 2023/04/28 22:35:14     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗                                                
[lp-1  ] INFO 2023/04/28 22:35:14 loadpoint 1:                                                                                                    
[lp-1  ] INFO 2023/04/28 22:35:14   mode:        pv                                                                                               
[lp-1  ] INFO 2023/04/28 22:35:14   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗                                               
[lp-1  ] INFO 2023/04/28 22:35:14   meters:      charge ✓                                                                                        
[lp-1  ] INFO 2023/04/28 22:35:14     charge:    power ✓ energy ✓ currents ✓                                                                   
[lp-2  ] INFO 2023/04/28 22:35:14 loadpoint 2:                                                                                                    
[lp-2  ] INFO 2023/04/28 22:35:14   mode:        pv                                                                                               
[lp-2  ] INFO 2023/04/28 22:35:14   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗                                               
[lp-2  ] INFO 2023/04/28 22:35:14   meters:      charge ✓                                                                                        
[lp-2  ] INFO 2023/04/28 22:35:14     charge:    power ✓ energy ✗ currents ✓                                                                   
[lp-1  ] DEBUG 2023/04/28 22:35:14 phase timer inactive                                                                                           
[lp-1  ] DEBUG 2023/04/28 22:35:14 pv timer inactive                                                                                              
[lp-2  ] DEBUG 2023/04/28 22:35:14 phase timer inactive                                                                                           
[lp-2  ] DEBUG 2023/04/28 22:35:14 pv timer inactive                                                                                              
[main  ] FATAL 2023/04/28 22:35:14 listen tcp :7070: bind: address already in use
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.111.1

stevoboy, 2023-04-28:
Update: Ich habe gerade mal eine evcc Instanz auf Windows 10 installiert und mit dem Assistenten eine YAML erzeugt. Hier funktioniert die Fahrzeugerkennung des Enyaqs über die Cloud API problemslos.
Ich vermute, ich habe ein Problem mit dem Internetzugang des Containers. 
Ich melde mich, nachdem ich das mal eruiert habe.

premultiply, 2023-04-29:
x509: certificate signed by unknown authority

Doku nicht aufmerksam gelesen und /etc gemountet.

closed 2023-04-29 by premultiply

stevoboy, 2023-04-29:
D

> `x509: certificate signed by unknown authority`
> 
> Doku nicht aufmerksam gelesen und `/etc` gemountet.

Das stimmt leider nicht. Ich bin streng nach dieser Anleitung [Anleitung.EVCC.Synology.Docker.Elli.Charger.Connect-Pro.pdf](https://github.com/evcc-io/docs/files/10365841/Anleitung.EVCC.Synology.Docker.Elli.Charger.Connect-Pro.pdf) vorgegangen. 
Am mounten von Ordner liegt es nicht, ich vermute auch im Host Netzwerk geht die Abfrage an Skoda Connect nur raus, es kommt durch die Firewall aber keine Rückmeldung zurück.

stevoboy, 2023-04-29:
Also, es funktioniert jetzt die Identifikation des Fahrzeuges sowohl an openwb als auch an der SMA Wallbox.
Ich habe das docker Image von evcc  und nicht von der Quelle, die im PDF beschrieben ist, heruntergeladen.
Dann hat es mit der gleichen YAML funktioniert.
Da sollte man mal die PDF updaten.
Die MAC Identifikation geht weiterhin nicht.