evcc-io/evcc #7737: Token expired

IAmKonni, 2023-04-29:
### Describe the bug

Hi, i am using not evcc but the evcc module within OpenWB to read SoC values of my EVs (currently Cupra Born and Skoda Enyaq).

Since some days OpenWB does not read the SoC values any more. I get this log message:

Error from EVCC: 2023/04/29 19:20:33 rpc error: code = Unknown desc = token is expired by 116h25m26s (LV0)

I did not change anything. Do I need a new token and if so, where do I get a new one? I am a monthly sponsor of this project.

Thanks in advance

### Steps to reproduce

see description

### Configuration details

```yaml
OpenWB 1.9.295
```

### Log details

```text
Error from EVCC: 2023/04/29 19:20:33 rpc error: code = Unknown desc = token is expired by 116h25m26s (LV0)
```

### What type of operating system are you running?

Linux

VolkerK62, 2023-04-30:
Have a look here  https://docs.evcc.io/docs/reference/configuration/sponsortoken

andig, 2023-04-30:
@IAmKonni du kannst die einfach unter https://sponsor.evcc.io ein neues Token holen!

closed 2023-04-30 by andig

IAmKonni, 2023-04-30:
Danke, hab's auf die schnelle nicht finden können. 👍

tommy25091974, 2023-10-22:
Hallo andig, 

habe das gleiche Problem. Der neue Token funktioniert aber auch nicht bei mir.

andig, 2023-10-22:
Das Thema ist seit 6 Monaten erledigt. Bitte Mail an info@evcc.io