evcc-io/evcc #7741: Add FritzPowerline

Nitron01, 2023-04-30:
Add template Support for AVM Fritz Powerline. 

Similar to FritzDect the FritzPowerLine Adapter can be read and managed. 
For example FritzPowerLine 546E. 

I traced the API Calls. 
Switch on:
HTTP Post:
http://192.168.xx.xx/net/home_auto_overview.lua?sid=a31111fc9a8bf18c&device=20001&switch=1&xhr=1&useajax=1
sid=a31111fc9a8bf18c <- Authentication like in FritzBox
device=20001 <- Device List Available (see below)
**switch=1** 
xhr=1
useajax=1

Switch off:
HTTP Post:
http://192.168.xx.xx/net/home_auto_overview.lua?sid=a31111fc9a8bf18c&device=20001&switch=0&xhr=1&useajax=1
sid=a31111fc9a8bf18c <- Authentication like in FritzBox
device=20001 <- Device List Available (see below)
**switch=0** 
xhr=1
useajax=1

Information regarding the current Power Data:
HTTP Get:
http://192.168.xx.xx/net/home_auto_query.lua?sid=a31111fc9a8bf18c&no_sidrenew=1&command=EnergyStats_10&id=20001&useajax=1&xhr=1&t1682835386978=nocache

Information regarding Device IDs:
http://192.168.xx.xx/net/home_auto_overview.lua?sid=a31111fc9a8bf18c&update=uiSmarthomeTables&view=groups&useajax=1&xhr=1&t1682835536172=nocache

https://boxmatrix.info/wiki/Property:home_auto_overview.lua

I put the Output of PowerData and Device Overview in Attachment. 
[FritzPowerline546E Device Overiew.txt](https://github.com/evcc-io/evcc/files/11360736/FritzPowerline546E.Device.Overiew.txt)
[FritzPowerline546E Energie Overiew.txt](https://github.com/evcc-io/evcc/files/11360737/FritzPowerline546E.Energie.Overiew.txt)

Let me know, when you need more/something else

premultiply, 2023-04-30:
/cc @thierolm

thierolm, 2023-04-30:
Hi Nitron, schick mal einen Screenshot, wie die 546E in der Fritzbox Oberfläche dargestellt wird.

Nitron01, 2023-04-30:
Hi thierolm,

so sehen die Steckdosen in der FB und in der Conifg Oberfläche des Apdaters selbst aus:
[image]
[image]

thierolm, 2023-04-30:
Das evcc fritzdect Modul nutzt die AHA API der Fritzbox: https://avm.de/fileadmin/user_upload/Global/Service/Schnittstellen/AHA-HTTP-Interface.pdf
Ich würde erwarten, dass du damit auch die 546E schalten kannst.
Du brauchst nur deine AIN!  
Soweit ich in deinem Screenshot oben sehe ==> 24:65:1 ...

Nitron01, 2023-04-30:
Ja, das habe ich probiert. Aber z. B. der Name des LUA-Scriptes passt nicht...
FritzDect: 
Zeile 99: 	uri := fmt.Sprintf("%s/webservices/homeautoswitch.lua", c.URI)

Und im Call der 546e "home**_**auto**_**overview.lua"

Deshalb kriege ich immer 400 Bad Request.

thierolm, 2023-04-30:
Kannst du mir bitte mal einen kompletten Trace schicken?

`evcc -l trace charger`

thierolm, 2023-04-30:
Und hast du die AIN in doppelte Hochkommata gesetzt, da sie Doppelpunkte enthält?

```yaml
charger:
- name: wallbox1
  type: template
  template: fritzdect
  uri: https://fritz.box
  user: <deinUser>
  password: <deinPasswort>
  ain: "24:65:1" # switch actor identification number without blanks (see AIN number on switch sticker)
  standbypower: 15
```

Nitron01, 2023-04-30:
Moment, trace mache ich sofort.

Ich habe die ":" bei der AIN immer entfernt, sodass sich eine HEX Nummer ergeben hat.
Ich probiere mit "

thierolm, 2023-04-30:
Ich würde auch mal mit : probieren ...

thierolm, 2023-04-30:
Es handelt sich um eine AIN nicht um eine MAC Adresse ...

thierolm, 2023-04-30:
Du kannst auch mal auf die Dose selbst schauen. Auf den Aufklebern hinten müsste die AIN auch draufstehen.

Nitron01, 2023-04-30:
Ja mit der AIN, mit ":" scheint es zu gehen. 
Keine 400er Fehlermeldung mehr. :-)
Allerdings ist die Leistungsanzeige bei 1 W, während die FB 50 W anzeigt

Nitron01, 2023-04-30:
Ah, das lag an der Konfig bzgl. StandbyPower. 

Jetzt geht es!

Also hier für interessierte Leser die Config:

- type: template
  template: fritzdect
  uri: http://fritz.box # HTTP(S) Adresse (Optional)
  user: SmartHome # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
  password: SmartHome # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
  ain: "24:65:11:C6NN2:NN" # Die AIN ist auf dem Typenschild auf der Geräterückseite aufgedruckt. Bei führenden Nullen bitte in doppelte Hochkommata setzen. **und bei Sonderzeichen**
  standbypower: 15 # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet (Optional) 
  name: PowerLinePC

thierolm, 2023-04-30:
Könntest du den Issue bitte als "gelöst" markieren? :-)

Nitron01, 2023-04-30:
> Könntest du den Issue bitte als "gelöst" markieren? :-)

Hab ich getan. 

Vielen Dank für Deine Unterstützung. Da war es jetzt doch einfacher, als vermutet.

closed 2023-04-30 by Nitron01