evcc-io/evcc #7754: ID4 wird nicht mehr erkannt

Gregor-de, 2023-04-30:
Hallo, 
Gibt es ein Problem mit VW? 
Ich bekomme folgende Fehlermeldung: 
vehicle not available: cannot create vehicle 'template': cannot create vehicle 'id': Get "https://login.apps.emea.vwapps.io/authorize?client_id=a24fba63-34b3-4d43-b181-942111e6bda8%40apps_vw-dilab_com&code_challenge=Epxb2fAtajVhosHpuYx2TjbTcilzzYayETs3XTmHlqE&code_challenge_method=S256&redirect_uri=weconnect%3A%2F%2Fauthenticated&response_type=code+id_token+token&scope=openid+profile+badge+cars+vin&nonce=uhropA1lxhqqRoAaJKqaQvX7sqzZAP6dE4J3SHpPzCt&state=21d27d4d-a6a1-474e-a96e-f11117ab7865": dial tcp: lookup login.apps.emea.vwapps.io on 192.168.178.1:53: no such host

Sieht irgendwie so aus, als ob die Site nicht mehr existiert. 
Oder gibt es andere mögliche Ursachen?

jheinitz, 2023-04-30:
Den Host `login.apps.emea.vwapps.io` scheint es tatsächlich nicht mehr zu geben. Mehr kann ich dazu auch nicht sagen. Liegt also nicht an Deinem DNS Server.

StefanSchoof, 2023-04-30:
Welche Version von evcc hast du im Einsatz?

Ich glaube das ist eine alte Login URL die nicht mehr benutzt wird. Bei mir funktioniert die Erkennung mit dem ID.3.

https://github.com/evcc-io/evcc/blob/5bdc578187b9cfe5a197cddff242906a03515db9/vehicle/vag/vwidentity/endpoint.go#L27

Gregor-de, 2023-04-30:
Ich bin wieder zurück auf die 0.116.3, da es mit der .6 Probleme mit meiner ElliPro gibt.

andig, 2023-04-30:
Sieht aus wie eines der DNS Probleme. Evtl. Pihole?

closed 2023-04-30 by andig

StefanSchoof, 2023-04-30:
> Ich bin wieder zurück auf die 0.116.3, da es mit der .6 Probleme mit meiner ElliPro gibt.

AFAIK sollten die letzten bekannten Probleme für die Elli mit der mit 0.116.6 gefixt sein. Wenn du mit der Version noch Probleme hast, würde ich empfehlen eine Diskussion mit Problembeschreibung und Config zu erstellen.

Gregor-de, 2023-04-30:
Schau ich morgen nochmal an.

Gregor-de, 2023-05-01:
Hab heute komplett neu installiert mit der 0.116.7. 
funktioniert wieder alles.