evcc-io/evcc #7755: chargers -> type: custom fehlt Syntax, z.B. integrateddevice und standbypower

TobiasSt78, 2023-04-30:
Hallo,
z.B. bei " template: shelly" kann mit den 2 Befehlen:
```
  standbypower: 3
  integrateddevice: true
```
fein justiert werden.

Die gleiche Syntax führt bei "type: custom" jedoch zu Fehlermeldungen und verhindert den Start.

Wäre klasse, wenn insbesondere integrateddevice auch bei custom integriert werden könnte.

andig, 2023-04-30:
Das geht bereits, siehe charger.go

closed 2023-04-30 by andig

TobiasSt78, 2023-04-30:
Ja, eben. In Shelly auch - nicht jedoch bei custom. Ist das nicht inkonsistent?

andig, 2023-04-30:
Charger.go ist das custom Gerät. Es geht, kannste mir glauben.

nicx, 2023-09-26:
@TobiasSt78 ich habe selbes problem, mit "type: custom" kann ich diverse parameter nicht setzen und bekomme beim ivcc start eine Fehlermeldung.

@TobiasSt78 Hast du eine Lösung gefunden?
@andig oder verwende ich das einfach falsch?

```
chargers:
  - name: WP_SG
    type: custom
    standbypower: -1 # um charger als static-switch zu nutzen
    integrateddevice: true
    icon: waterheater
    enable:
      source: mqtt
      topic: espaltherma/sg/set
      payload: 2
```

Fehlermeldung:
```
* '' has invalid keys: integrateddevice, standbypower
[main  ] FATAL 2023/09/26 09:50:34 will attempt restart in: 5m0s
```

andig, 2023-09-26:
Ja, tust du da kein Template. Das muss eine Liste sein:

    feastures:
       - integrateddevice

nicx, 2023-09-26:
@andig sorry aber deinen Beitrag verstehe ich nicht ganz?! :)

auch mit folgender config funktioniert es nicht:
```
chargers:
  - name: WP_SG
    type: custom
    features:
      - standbypower: -1 # um charger als static-switch zu nutzen
      - integrateddevice: true
      - icon: waterheater
    enable:
      source: mqtt
      topic: espaltherma/sg/set
      payload: 2
```

Fehler:
```
* 'features[0]' expected type 'api.Feature', got unconvertible type 'map[string]interface {}', value: 'map[standbypower:-1]'
* 'features[1]' expected type 'api.Feature', got unconvertible type 'map[string]interface {}', value: 'map[integrateddevice:true]'
* 'features[2]' expected type 'api.Feature', got unconvertible type 'map[string]interface {}', value: 'map[icon:waterheater]'
```