evcc-io/evcc #7760: Aufploppen "Keine Verbindung zum Server" Meldung nach Update, Bedienung ohne Einschränkung möglich

StevieC121176, 2023-05-01:
### Describe the bug

Nach dem Update auf Version `0.116.6 (d327fe71)` ploppt im Sekundentakt unten markierte Meldung auf. Die Bedienung ist aber ohne Einschränkung möglich und auch die Anzeigen funktionieren. Das Verhalten hatte ich mit der vorigen Version nicht. Geändert wurde ausser dem Update nichts.

[image]

### Steps to reproduce

1.
2.
3.
...
EVCC stoppen
Update auf neue Version `0.116.6 (d327fe71)`
EVCC starten
Fehlermeldung

### Configuration details

```yaml
<details><summary>Konfiguration (/home/pi/evcc.yaml)</summary>

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.117  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.118  
  port: 1502  
  usage: pv 
  modbus: tcpip 
  name: pv2
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.118  
  port: 1502  
  usage: battery  
  modbus: tcpip 
  capacity: 12.8 
  name: battery2
- type: template
[... 261 lines cut ...]
May 01 14:10:09 raspberrypi evcc[3340]: [site  ] DEBUG 2023/05/01 14:10:09 grid powers: [-250 590 -340]W
May 01 14:10:09 raspberrypi evcc[3340]: [site  ] DEBUG 2023/05/01 14:10:09 grid currents: [-1.64 3.29 -1.56]A
May 01 14:10:09 raspberrypi evcc[3340]: [site  ] DEBUG 2023/05/01 14:10:09 giving priority to battery charging at soc: 0%
May 01 14:10:09 raspberrypi evcc[3340]: [site  ] DEBUG 2023/05/01 14:10:09 site power: 100W
May 01 14:10:09 raspberrypi evcc[3340]: [lp-1  ] DEBUG 2023/05/01 14:10:09 charge voltages: [239 237 237]V
May 01 14:10:09 raspberrypi evcc[3340]: [lp-1  ] DEBUG 2023/05/01 14:10:09 detected connected phases: 3p
May 01 14:10:09 raspberrypi evcc[3340]: [lp-1  ] DEBUG 2023/05/01 14:10:09 charge currents: [0 0 0]A
May 01 14:10:09 raspberrypi evcc[3340]: [lp-1  ] DEBUG 2023/05/01 14:10:09 charge meter total import: 2803.800kWh
May 01 14:10:09 raspberrypi evcc[3340]: [lp-1  ] DEBUG 2023/05/01 14:10:09 charger status: A
May 01 14:10:09 raspberrypi evcc[3340]: [lp-1  ] INFO 2023/05/01 14:10:09 car disconnected
May 01 14:10:09 raspberrypi evcc[3340]: [lp-1  ] DEBUG 2023/05/01 14:10:09 set charge mode: off
May 01 14:10:09 raspberrypi evcc[3340]: [lp-1  ] DEBUG 2023/05/01 14:10:09 set min current: 6
May 01 14:10:09 raspberrypi evcc[3340]: [lp-1  ] DEBUG 2023/05/01 14:10:09 set max current: 16
May 01 14:10:09 raspberrypi evcc[3340]: [lp-1  ] DEBUG 2023/05/01 14:10:09 set min soc: 0
May 01 14:10:09 raspberrypi evcc[3340]: [lp-1  ] DEBUG 2023/05/01 14:10:09 set target soc: 100
May 01 14:10:10 raspberrypi evcc[3340]: [site  ] DEBUG 2023/05/01 14:10:10 ----
```

### What type of operating system are you running?

Linux

### Version

0.116.6 (d327fe71)

StevieC121176, 2023-05-01:
Ich konnte es etwas eingrenzen. Es scheint nur neuere IOS Geräte zu betreffen. Das Verhalten kann ich auf dem iPhone mit IOS 16.5 und dem iMac mit der neuesten Version beobachten. Auf einem alten iPad mit iOS 12.3 funktioniert es ohne Probleme genau wie auf Android Geräten.

andig, 2023-05-01:
Ist schon behoben.

closed 2023-05-01 by andig