evcc-io/evcc #7765: "Reduziere auf einphasig" / "erhöhe auf 3 phasig" button zum sofortigen wechsel ohne wartezeit

horst789github, 2023-05-01:
hi,

könnte man vielleicht wenn der text kommt irgendwo ein button /hyperlink machen, wo sofort gewechselt wird und nicht erst die zeit abläuft? ich habe aktuell oft den fall das ich diese meldung sehe und dann immer den countdown abwarten muss.

[image]

jheinitz, 2023-05-01:
Moin!

Du kannst das auf jeden Fall verkürzen, wenn Du in Deiner Konfiguration die Werte `delay` für `enable` bzw. `disable` anpasst. Die Defaults sind 1m bzw. 3m.

Viel Erfolg

horst789github, 2023-05-01:
das man das in der config ändern kann, weiß ich natürlich, allerdings ist eine config änderung ja immer mit einem neustart verbunden und etwas permanentes. In meinem fall möchte ich keine dauerhafte Config änderung sondern "spontan" einfach "jetzt" umschalten.

premultiply, 2023-05-01:
API benutzen?

horst789github, 2023-05-01:
> API benutzen?

wäre für mich als workaround auch in ordnung, ist das schon funktional?

naltatis, 2023-05-02:
Einen API Endpunkt der PV/Phase Timer überspringt haben wir mMn noch nicht oder an was hast du gedacht @premultiply ?

premultiply, 2023-05-02:
Eher an die Timer selbst.

naltatis, 2023-05-03:
@andig was meinst du? Neuen Endpunkt `POST /api/loadpoints/<id>/skiptimer` (o.ä. der laufende PV/Phase/Guard Timer überspringt?

In der UI würde ich bei den passenden Meldungen einfach einen kleinen Textlink hinter dem Statustext integrieren.

andig, 2023-05-04:
Ich fände es blöd wenn das UI sich mit den Timern in der Form auseinander setzen müsste. Prinzipiell soll der Anwender ja nicht selbst rumschalten, sondern die Automatik greifen. Ich sehe zwei Optionen:
- Delays kurzer einstellen (Anwender)
- Settingsmenü nutzen (api=sofort)

Wir (anderer PR) suchen prinzipiell nochmal Stellen an denen das eigentlich sofort passieren sollte. Das gilt aber nicht für den PV Betrieb im Steady state.

Aus meiner Sicht: wontfix, hier kein Todo.

andig, 2023-05-04:
Anforderung verstanden, aktuell aber kein Thema für uns- config ui ist wichtiger 😀

closed 2023-05-04 by andig